MichaIng/DietPi #2362: DietPi-Software | myMPD: Use latest release via api.github.com

MichaIng, 2018-12-23:
... instead of master or fixed history point: https://api.github.com/repos/jcorporation/myMPD/releases/latest

sChUhBiDu, 2019-02-23:
How can I do that? Sorry Linux noob here? having problems with playing music. Can see all files but when I press play nothing happens ;_(

MichaIng, 2019-02-23:
@sChUhBiDu 
High likely this is due to a current permissions issue with mpd. Please try the following:
```
sed -Ei '/^(user|group)[[:blank:]]/d' /etc/mpd/mpd.conf
G_CONFIG_INJECT 'User=' 'User=mpd' /lib/systemd/system/mpd.service '\[Service\]'
G_CONFIG_INJECT 'PermissionsStartOnly=' 'PermissionsStartOnly=true' /lib/systemd/system/mpd.service '^User=mpd'
usermod -a -G audio,dietpi mpd
systemctl daemon-reload
systemctl restart mpd
```

sChUhBiDu, 2019-02-23:
Thanks but this [https://github.com/Fourdee/DietPi/issues/2522](url) help me already.

1 problem left: when I want to save a **playlist** it says "**Invalid filename**"

How can I install latest version of MyMPD?

MichaIng, 2019-02-23:
@sChUhBiDu 
We currently froze the version due to a database update issue with the latest release some time ago: https://github.com/Fourdee/DietPi/issues/2156

But you can try out the most current release (actually from today 😄) and if it works, we can finish this topic and implement auto install of latest release into DietPi-Software. To try it out:
```
cd /tmp
wget https://github.com/jcorporation/myMPD/archive/v5.1.0.tar.gz
tar xf v5.1.0.tar.gz
cd myMPD*
./mkrelease.sh
cd /tmp
rm -R myMPD*
```

MichaIng, 2020-01-01:
Latest upstream version is already installed since DietPi v6.25. We simply download the master branch archive, which matches the latest release/is release branch.

closed 2020-01-01 by MichaIng