MichaIng/DietPi #2363: Software request - acme.sh

Kreeblah, 2018-12-23:
## Creating a software request:

#### Give us some formal software information:
- Software title | acme.sh
- Short description | Let's Encrypt client written in shell scripts
- GitHub URL (if available) | https://github.com/Neilpang/acme.sh
- External URL (if available) | https://acme.sh
- (Official) install documentation | https://github.com/Neilpang/acme.sh/wiki/How-to-install

#### Are there similar/alternative software titles available with DietPi-Software?

- Certbot

#### What makes your requested software better than the above solutions, if available?

- acme.sh supports a much, much wider list of DNS services (which is frequently expanding) for automated domain control validation, in addition to all of the validation methods and DNS services that Certbot supports, domain-specific certificates, wildcard certificates, etc.

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?

- The same reasons for supporting Certbot apply.
- It would also be useful for generating certificates for things other than Apache or Nginx installs, if they're needed.
- It looks like dietpi-letsencrypt doesn't currently support DNS validation, so that could be useful for generating certificates for internal-only hosts.

#### Can you provide the installation steps that you would suggest DietPi-Software to do?

MichaIng, 2018-12-23:
@Kreeblah 
Thanks for your request.

Jep we had this suggestion in the past. Actually my plan is to create a new `DietPi-TLS` script. This should allow to:
- Create self-singed certificate
- Get LetsEncrypt cert via certbot
- Get LetsEncrypt cert via acme.sh
- Enable/Disable HTTPS/TLS encryption for all webservers and other software titles individually

It makes sense then to allow DNS validation method, e.g. if no webserver is installed, the cert is required for something else.

Other issues/feature requests related to this:
- https://github.com/Fourdee/DietPi/issues/1622
- https://github.com/Fourdee/DietPi/issues/1840
- https://github.com/Fourdee/DietPi/issues/1910

MichaIng, 2018-12-26:
Added to FeatHub, feel free to vote for it: https://feathub.com/Fourdee/DietPi/+23

Kreeblah, 2018-12-26:
Um, did that comment get left on the wrong issue?  That link points to an entry for Etherpad.

MichaIng, 2018-12-26:
@Kreeblah 
Whoopsie, fixed!

keithellis74, 2019-06-20:
Hi guys, I'm in the process of setting up HAproxy to handle a single SSL cert for multiple sub domains directing to multiple backends.  Is there a timeline for wildcard lets encrypt certificates?

MichaIng, 2019-06-20:
@keithellis74 
Larger tasks, will take at least 2 DietPi subversions longer. However easy to apply manually, which webserver do you use?

ovz93br43v7, 2021-02-17:
Probably useful, I read [here](https://www.kuketz-blog.de/dietpi-raspberry-pi-auf-diaet-raspberry-pi-teil2/) from DietPi. In the comment section was a discussion about Certbot which ended in two proposals: 
https://github.com/diafygi/acme-tiny
and the already mentioned https://github.com/acmesh-official/acme.sh

So probably [acme-tiny](https://github.com/diafygi/acme-tiny) is worth a look.

MichaIng, 2021-02-17:
Here is how I use acme.sh currently: https://github.com/MichaIng/hacks/blob/main/install_acme.sh

acme-tiny is a Python script (hence not so "tiny" when taking into account the dependency) and not developed for more than a year. acme.sh is very actively developed and has a large set DNS authentication plugins, full support for as well ECC certs and all such, so I think between those two the match is more than clear 😉.

sannidhyaroy, 2023-06-28:
Adding acme.sh to dietpi-software would be really useful.

cujomalainey, 2023-07-02:
Came across this after going through the guide on digitalocean and then tricking the dietpi script into setting up renewal services. Definitely would be great to get better support for use cases like HTTPS over VPN.

Joulinar, 2024-08-31:
You would need to configure your `acme.sh` container in away to store certificates on local storage outside the container. This way you are able to access them and can do whatever you want.

MichaIng, 2024-08-31:
Note that acme.sh is a well documented single standalone shell script. I see zero reason why one would want to wrap it into a Docker container, where you just replace one CLI with another + complexity + overhead + points of failure etc. Nothing against containers, but a shell script is not not a use cass for them.