MichaIng/DietPi #2367: DietPi-Software | ownCloud/Nextcloud: PHP7.0 support will be dropped soon

MichaIng, 2018-12-24:
ownCloud, as well as Nextcloud, plan to drop PHP7.0 support soon.

On ownCloud this will be upcoming v10.1: https://doc.owncloud.org/server/administration_manual/installation/system_requirements.html
Nextcloud v15 already shows a warning on admin panel, v16, expected until mid of 2019 most likely will drop support.

Debian Stretch most likely will stay on PHP7.0 and Buster release is most likely afterwards. And it takes much time until Raspbian Buster is released then and much more until we (can) drop Stretch support of course.

So either we need to limit max version for those clouds, similar to Jessie => NC13, or we upgrade PHP version on Stretch to at least PHP7.1, better PHP7.2 or even PHP7.3 directly. This brings of course the risk that other software titles do not work anymore then... Perhaps allow multiple PHP versions then, if not possible without, which should generally work well since PHP7.

Fourdee, 2018-12-24:
@MichaIng 

Interesting, NC/OC must require features from php 7.1+.

In theory, a sub version of PHP should not break software. However, if we were to offer 7.2 installs on Stretch, we need to do some extensive testing.

Appears all archs are supported via this repo:
https://packages.sury.org/php/pool/main/p/php7.2/
https://linuxhostsupport.com/blog/how-to-install-php-7-2-on-debian-9/

@MichaIng 

I'd suggest we go with PHP7.3 directly on all distros? But as above, extensive testing required if we do.

MichaIng, 2018-12-25:
@Fourdee 
Jep Ondrej is official PHP package maintainer anyway, so the repo/packages themselves are trustworthy and should work fine themselves.

But on Debian repo, it is guaranteed that all software is compatible with the offered PHP version, which is not the case for any newer version from outside of Debian repo of course. So jep, requires intense testing, and I already know that ownCloud currently (10.0.10) does not yet support PHP7.3, why I disabled it on Buster for now: https://github.com/Fourdee/DietPi/pull/2361/commits/a454034c0060f815e88deee8048bb89404b43ac2

So for the first, I would wait until PHP7.0 support is really dropped for a certain software title, e.g. next ownCloud or Nextcloud release most properly. As quick fix, we can simply freeze the downloaded version to the last one with PHP7.0 support, similar to what I've done with Nextcloud on Jessie (max v13). Internal updater will inform user and block further updates.
Actually interesting that ownCloud currently still works on Jessie, so supports PHP5.6, but will drop PHP7.0 support directly with 10.1 according to docs 🤔.

However then (or already now start to) test all our offered software with PHP7.2, which most properly will cause less issues? I mean PHP7.3 can be tested as well, but since even ownCloud does not yet support it and I also read topics on Nextcloud forum/GitHub about some minor issues, I bet there will be much more. And especially on Jessie, the APT based packages are even older, so might cause even more problems then those on Stretch. Best might be to simply freeze or disable software versions with PHP5 support drop (as part of Jessie support fade) and concentrate testing on Stretch.

So we don't need to hurry from my point of view, but do careful tests instead, respectively read docs about PHP requirements of software that we pull from upstream.

d-tamm, 2019-01-01:
This affects not only the main Nextcloud/Owncloud installation. Also some addons have their own requirements on php, e.g. passwords (https://apps.nextcloud.com/apps/passwords) requires php >= 7.1.0. Therefore, these addons cannot currently be used in a standard DietPi installation. I will try to follow the instructions in @Fourdee's links and try to get a more recent php up and running. Otherwise I might go back to other distros :(

MichaIng, 2019-04-23:
### PHP7.3 support check
🈯️ Nextcloud
🈴 ownCloud
  ```
  This version of ownCloud is not compatible with PHP 7.3
  You are currently running PHP 7.3.4-1+0~20190412071350.37+stretch~1.gbpabc171.
  ```
- Damn, they even block install...

MichaIng, 2019-04-23:
Jep, so all web applications we offer work fine with PHP7.3 but ownCloud which will not support it for **at least** one month more (assuming current ~1 month patch schedule), perhaps longer until next minor or major release since PHP7.3 support does not sound like a patch?

MichaIng, 2019-04-24:
#### PR up: https://github.com/MichaIng/DietPi/pull/2735

MichaIng, 2019-04-26:
Merged! This is a giant one and I hope we find much testers during beta phase. I tested up and down on Jessie, Stretch and Buster with all relevant software titles installed and all webserver combinations, however it is hard to test all possible situations and it IS a major PHP version upgrade for Jessie at least.

Nextcloud <14 is required to be upgraded to NC14 as well. There should be only a hand full Jessie users with Nextcloud (13) installed, if any. Already since Nextcloud prompts with a big warning that this version has fully been dropped from support/EOL etc. So if any issue occurs (also I tested this several times now), then we will provide fast per-user support. Great is that our cloud migration script (created to make migration from DietPi v159 to v6.0 easier that time) serves greatly here to create a full recoverable Nextcloud backup: https://github.com/MichaIng/DietPi/blob/master/.meta/dietpi-cloud-migration
And in the end the benefit of PHP7.3 and Nextcloud 16 is enormous so should totally worth the risk of both upgrades 😉.

However I will go through survey files to check for amount Jessie + Nextcloud users.

closed 2019-05-12 by MichaIng