MichaIng/DietPi #2374: skip 	re-install python 3.6.3 (157 home-assistant)

MZorzy, 2018-12-28:
not a bis issue but every time
dietpi-software reinstall 157
it ask to reinstrall python 3.6.3 y/N , if No it end and not upgrade. if y ... redownload with lot of time consumping.

may be an option to check if exist 3.6.3 and skip OR continue upgrade/reinstall  ?

MichaIng, 2018-12-28:
@MZorzy 
Thanks for your request.

Python is installed within `pyenv`. Requires investigation in how to check for installed python version within `pyenv`. In question, to be failsafe, always (re)install Python at least assures functionality.

Why actually do you need to reinstall HA that often, that Python reinstall is an issue?

> it ask to reinstrall python 3.6.3 y/N , if No it end and not upgrade.

The related code line is: `su --shell /bin/bash --command "cd $ha_userroot; $ha_pyenv_activation; pyenv install $ha_python_version" $ha_user`
- Actually `dietpi-software` should go on fine (doesn't it?), if you choose `no`, so if there is no way reliable check for installed python version, maybe we can run the command non-interactively, so default answer (capital `N`, so no is default?) will be chosen.
- Or does running `pyenv-installer` before, somehow clears the environment (partly), so that python needs to be reinstalled to avoid errors? But these should not show up during install step, since nothing is error handled there, script should just do on in case of failure.
- Will test myself and try to replicate.

Generally the HA installer could retrieve some update. Looks like copy and paste from a guide, with a complexity that is not required for DietPi (but works!).

MichaIng, 2018-12-28:
Further research:
- `pyenv` actually is a Python version manager only. If HA is compatible with the one, installed by default via repo, it should work without pyenv?
> Home Assistant provides multiple ways to be installed. A requirement is that you have Python 3.5.3 or later installed.
- 3.5.3 available on Stretch: https://packages.debian.org/de/stretch/python3, so `pyenv` should be only required on Jessie.
- Actually I would drop support on Jessie to simplify the install code, just keep support (and uninstall code) for existing HA setups. But this would break reinstall for those systems as well. Maybe check survey for HA + Jessie systems first.

MZorzy, 2019-01-16:
> @MZorzy
> Thanks for your request.

> The related code line is: `su --shell /bin/bash --command "cd $ha_userroot; $ha_pyenv_activation; pyenv install $ha_python_version" $ha_user`

for tet i comment # this line, running dietpi-software reinstall 157 it upgrade HA as i want, without reinstall pyton. and work.
homeassisant add new device, and release new version, very very often

MichaIng, 2019-01-16:
There is no internal updater (from web UI) available, right?

Yeah so maybe we can check for existing install and skip dependency install the. Only thing is, if dependency min version raises, the update breaks HA. So at best check and compare installed and required dependency versions.

Or we skip the white pyenv, if possible.

MZorzy, 2019-01-16:
there is no an internal updater if installed as "157", 
but there is with docker-hassio (https://github.com/home-assistant/hassio-build/tree/master/install) with add-on feature
don't know how/if docker slowdown raspberry.

MZorzy, 2019-01-29:
> Why actually do you need to reinstall HA that often, that Python reinstall is an issue?
cause "update and fix" some time too fast...
Release 0.86.1 - January 23
Release 0.86.2 - January 24
Release 0.86.3 - January 26
Release 0.86.4 - January 28

meanwhilr i edit dietpi-software skipping "# Install Python which is needed for HA" on line 6891-6892

MichaIng, 2019-11-06:
PR up to inform users about long install process (+ Python update and slight install enhencements): https://github.com/MichaIng/DietPi/pull/3220
The info contains a link to this issue, however steps to reduce the (re)install times are still outstanding.

MichaIng, 2020-01-01:
With v6.27, install has been simplified, but the pyenv install + Python compiling stays. We cannot really skip it due to the vast amount of special module version needs. Only alternative would by venv from APT-installed Python. With this, Python does not need to be compiled anymore, but the instance it less independent and lower version.

I mark this issue as closed for now. Further discussion/enhancements can be done at a later date.

closed 2020-01-01 by MichaIng