MichaIng/DietPi #2377: MPD with NFS music doesn't work after upgrade 6.10 -> 6.19

mfeif, 2018-12-29:
## Creating a bug report/issue:

I just upgraded to 6.19 from 6.10. I was using MPD with a remote database of music, running over nfs to a linux server on the LAN. Note that mpd has its own [built-in nfs client stuff](https://www.musicpd.org/doc/html/plugins.html?highlight=nfs#nfs) not via a filesystem mount. This worked on DietPi 6.10. Now, I get this:

```
# mpd --no-daemon --stderr /etc/mpd.conf
exception: Unrecognized storage URI: nfs://10.0.0.2/srv/music
```

It looks like maybe someone disabled the compiled in storage plugin for NFS (look at Storage plugins, which ought to contain nfs):
```
# mpd --no-daemon --stderr -v -V /etc/mpd.conf
Music Player Daemon 0.20.21

Copyright (C) 2003-2007 Warren Dukes <warren.dukes@gmail.com>
Copyright 2008-2017 Max Kellermann <max.kellermann@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Database plugins:
 simple proxy upnp

Storage plugins:
 local curl

Neighbor plugins:
 upnp

Decoders plugins:
 [mad] mp3 mp2
 [mpg123] mp3
 [vorbis] ogg oga
 [oggflac] ogg oga
 [flac] flac
 [opus] opus ogg oga
 [dsdiff] dff
 [dsf] dsf
 [faad] aac
 [wavpack] wv
 [ffmpeg] 16sv 3g2 3gp 4xm 8svx aa3 aac ac3 adx afc aif aifc aiff al alaw amr anim apc ape asf atrac au aud avi avm2 avs bap bfi c93 cak cin cmv cpk daud dct divx dts dv dvd dxa eac3 film flac flc fli fll flx flv g726 gsm gxf iss m1v m2v m2t m2ts m4a m4b m4v mad mj2 mjpeg mjpg mka mkv mlp mm mmf mov mp+ mp1 mp2 mp3 mp4 mpc mpeg mpg mpga mpp mpu mve mvi mxf nc nsv nut nuv oga ogm ogv ogx oma ogg omg opus psp pva qcp qt r3d ra ram rl2 rm rmvb roq rpl rvc shn smk snd sol son spx str swf tak tgi tgq tgv thp ts tsp tta xa xvid uv uv2 vb vid vob voc vp6 vmd wav webm wma wmv wsaud wsvga wv wve
 [pcm]

Filters:
 soxr

Tag plugins:
 id3tag

Output plugins:
 shout null fifo pipe alsa httpd recorder

Encoder plugins:
 null vorbis opus lame wave flac

Archive plugins:
 [iso] iso

Input plugins:
 file alsa archive curl ffmpeg cdio_paranoia

Playlist plugins:
 extm3u m3u pls xspf asx rss soundcloud flac cue embcue

Protocols:
 file:// http:// https:// gopher:// rtp:// rtsp:// rtmp:// rtmpt:// rtmps:// cdda:// alsa://

Other features:
 avahi epoll icu inotify systemd tcp un
``` 

Additionally, I can't even run mpd; if I type mpd as root, I get `exception: No configuration file found`. I'm going to file another bug on this, because I think it's not the same.

#### Required Information:
- DietPi version
```
#!/bin/bash
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=19
G_DIETPI_VERSION_RC=7
G_GITBRANCH=master
G_GITOWNER=Fourdee
```

- Distro version	| `9.6`
- Kernel version	| `Linux PiFi 3.10.38 #22 SMP PREEMPT Fri Dec 22 10:45:41 IST 2017 armv7l GNU/Linux`
- SBC device		| `Sparky SBC (armv7l)`

#### Additional Information (if applicable):
- Software title	| (EG: MPD)
- Was the software title installed freshly or updated/migrated? updated
- Can this issue be replicated on a fresh installation of DietPi? Unknown; fresh update.
- `dietpi-bugreport` 94df705b-f681-420c-b01b-3217e7eeb3db

#### Steps to reproduce:
1. have a config that references an nfs music directory, eg `music_directory "nfs://10.0.0.2/srv/music"`
2. try to run mpd
3. Get error

#### Expected behaviour:
...

#### Actual behaviour:

- ...

#### Extra details:

- ...

mfeif, 2018-12-29:
For reference, here's `mpd -V` on an Ubuntu 18.10 box:

```
Music Player Daemon 0.20.21

Copyright (C) 2003-2007 Warren Dukes <warren.dukes@gmail.com>
Copyright 2008-2017 Max Kellermann <max.kellermann@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Database plugins:
 simple proxy upnp

Storage plugins:
 local smbclient nfs curl

Neighbor plugins:
 smbclient upnp

Decoders plugins:
 [mad] mp3 mp2
 [mpg123] mp3
 [vorbis] ogg oga
 [oggflac] ogg oga
 [flac] flac
 [opus] opus ogg oga
 [sndfile] wav aiff aif au snd paf iff svx sf voc w64 pvf xi htk caf sd2
 [audiofile] wav au aiff aif
 [dsdiff] dff
 [dsf] dsf
 [faad] aac
 [mpcdec] mpc
 [wavpack] wv
 [modplug] 669 amf ams dbm dfm dsm far it med mdl mod mtm mt2 okt s3m stm ult umx xm
 [mikmod] amf dsm far gdm imf it med mod mtm s3m stm stx ult uni xm
 [sidplay] sid mus str prg P00
 [wildmidi] mid
 [fluidsynth] mid
 [adplug] amd d00 hsc laa rad raw sa2
 [ffmpeg] 16sv 3g2 3gp 4xm 8svx aa3 aac ac3 adx afc aif aifc aiff al alaw amr anim apc ape asf atrac au aud avi avm2 avs bap bfi c93 cak cin cmv cpk daud dct divx dts dv dvd dxa eac3 film flac flc fli fll flx flv g726 gsm gxf iss m1v m2v m2t m2ts m4a m4b m4v mad mj2 mjpeg mjpg mka mkv mlp mm mmf mov mp+ mp1 mp2 mp3 mp4 mpc mpeg mpg mpga mpp mpu mve mvi mxf nc nsv nut nuv oga ogm ogv ogx oma ogg omg opus psp pva qcp qt r3d ra ram rl2 rm rmvb roq rpl rvc shn smk snd sol son spx str swf tak tgi tgq tgv thp ts tsp tta xa xvid uv uv2 vb vid vob voc vp6 vmd wav webm wma wmv wsaud wsvga wv wve
 [gme] ay gbs gym hes kss nsf nsfe sap spc vgm vgz
 [pcm]

Filters:
 libsamplerate soxr

Tag plugins:
 id3tag

Output plugins:
 shout null fifo pipe alsa roar ao oss openal pulse jack httpd recorder

Encoder plugins:
 null vorbis opus lame wave flac

Archive plugins:
 [bz2] bz2
 [zzip] zip
 [iso] iso

Input plugins:
 file alsa archive curl ffmpeg smbclient nfs mms cdio_paranoia

Playlist plugins:
 extm3u m3u pls xspf asx rss soundcloud flac cue embcue

Protocols:
 file:// http:// https:// mms:// mmsh:// mmst:// mmsu:// gopher:// rtp:// rtsp:// rtmp:// rtmpt:// rtmps:// smb:// nfs:// cdda:// alsa://

Other features:
 avahi epoll icu inotify ipv6 systemd tcp un
```

MichaIng, 2018-12-29:
@mfeif 
Just rechecked and jep, on Stretch we install a self-compiled package with newer version (0.20.21 vs 0.19.21 on Stretch repo) where the NFS protocol is not implemented.

I am not sure now if this was different in the past, however multiple reinstalls within update path show that most likely not only the config file has changed. Sadly I just now on DietPi v6.20 dev I managed to clean up the reinstalls to be only done once, from whatever version you are coming, however is is not related.

Since there is no NFS related entry in `/etc/mpd.conf` (also not the default one from Stretch repo), I guess this needs to be enabled on compile time?
_________
Testing it with default Debian Stretch package...

Fourdee, 2019-01-02:
@MichaIng 

> Any chance we have our binaries updated with smbclient and nfs storage plugins, and if already also to current upstream version?

Yep, i'll update the binaries.

> Required when compiliing: libnfs-dev libsmbclient-dev

Legend 👍

mfeif, 2019-01-02:
Thanks guys!

MichaIng, 2019-01-02:
@Fourdee 
> Fails:` option('systemd', type: 'feature', description: 'systemd support')` `libsystemd-dev` installed.

Strange, although no big issue since we create the systemd unit ourself on install? I guess this option would just implement the systemd unit into the installer/deb package?
**EDIT: Ah, did you install `libwrap0-dev` as well? At least it is in the same line of the listed build dependencies: https://www.musicpd.org/doc/html/user.html#compiling-from-source**

UPnP for sure requires `libupnp-dev`, but I guess you had that installed?

Fourdee, 2019-01-02:
@MichaIng 

> UPnP for sure requires libupnp-dev, but I guess you had that installed?

Yep, was installed. Reverted back to to 0.20.X for now.

Fourdee, 2019-01-03:
Local testing passed, however, issue with https://github.com/Fourdee/DietPi/issues/2378 still exists.

Completed.

closed 2019-01-03 by Fourdee