MichaIng/DietPi #2379: DietPi-Software | Kodi H.265 support

MichaIng, 2018-12-29:
Seems to be not supported on Kodi 17. Kodi 18 is close to release, but will surely not be backported to Debian Stretch repo, but at least hopefully to Meverics Odroid repo/backports.

However perhaps it's possible to add H.265 support also to older versions. Not the topic I know much about 😉.

Ref Kodi v18 test: https://github.com/Fourdee/DietPi/issues/2371
Ref H.265 request: https://dietpi.com/phpbb/viewtopic.php?p=15874#p15863

msdos, 2019-04-01:
It seems to be some patches in `ffmpeg`. More info in https://forum.kodi.tv/showthread.php?tid=335837

msdos, 2019-04-01:
Actually I now ain't so sure about it. I've upgraded Kodi from 17 to 18, [using `apt upgrade`](https://github.com/MichaIng/DietPi/issues/2371#issuecomment-478427670) and now I can execute H265 videos.

MichaIng, 2019-04-02:
@msdos 
Thanks for testing. So it is indeed about the software here. RPi itself seems to have enabled/improved H.265 hardware acceleration if I do not misunderstand: https://raspberrypi.stackexchange.com/a/81858

I hope also Debian ships Kodi 18 soon, currently even on Buster stuck at Kodi 17: https://packages.debian.org/de/buster/kodi

MichaIng, 2019-04-27:
I mark this as closed as it is nothing we can influence currently. We rely on the Kodi version shipped by the repo, especially since it is mostly customized per SBC.

RPi ships Kodi 18 now via Raspbian Stretch repo.

closed 2019-04-27 by MichaIng