MichaIng/DietPi #2382: Wrong timestamp in Pi-hole query log

tillcash, 2018-12-31:
Required Information:
- DietPi version	| v6.19.7
- Distro version	| stretch / 9.6
- Kernel version	| Linux DietPi 4.14.90-v7+ #1183 SMP Fri Dec 21 14:03:50 GMT 2018 armv7l GNU/Linux
- SBC device		| RPi 3 Model B (armv7l)
- Power supply used	| 5V 1A RAVpower
- SDcard used		| SanDisk

Additional Information (if applicable):
- Software title	| Pi-hole
- Was the software title installed freshly or updated/migrated? freshly
- Can this issue be replicated on a fresh installation of DietPi? yes
- 4a2260f4-e4d5-4d76-9ebc-7632903be267

Steps to reproduce:
Changed device time zone to asia/kolkata but its not reflecting in pi-hole query log (shows UK time)

Extra details:

This issue (and ones like it) might be caused by an incorrect timezone on the Pi-hole or the browser. The queries are stored in UTC time, based on the device's timezone, and then converted to the browser's time zone when displayed in the query log. Usually one of those two timezones is incorrect.

Fourdee, 2018-12-31:
@tillcash 

Many thanks for the report 👍  we'll try to replicate and investigate.

Fourdee, 2018-12-31:
@tillcash 

Appears to be a documented issue. Please can you try the following commands in this post?
https://discourse.pi-hole.net/t/query-log-entries-getting-wrong-time-stamp/8114

Optional: Might be advisable to backup system before hand with:
```
dietpi-backup
```

Then run repair:
```
mv /etc/pihole/pihole-FTL.db /etc/pihole/pihole-FTL.db.bak
pihole -r
```

Then verify if timestamps are valid.

tillcash, 2019-01-01:
I tried before submitting the issue. /r/pihole suggested to check with DietPi team for support.

closed 2019-01-01 by tillcash