MichaIng/DietPi #2384: [proposed Help wanted :sos:] DietPi-Software | Mopidy playlists not working due to missing config/permission?!?

cyberlussi, 2019-01-01:
I am new to dietpi and mopidy but it seems for me that  playlists are not working in the vanilla dietpi setup of mopidy. I have a workaround (see below) but I am not sure whether this is perfect under all circumstances. I mean, I do not know whether mopidy-m3u is then still able to scan for m3u below the music directory or not ...

#### Required Information:
- DietPi version	| 6.19.7
- Distro version	| 9.6
- Kernel version	| inux DietPi 4.14.79+ # 1159 Sun Nov 4 17:28:08 GMT 2018 armv6l GNU/Linux
- SBC device		| RPi3
- Power supply used	| 5V 1A
- SDcard used		| Kingston 8GB

#### Additional Information (if applicable):
- Software title	| Mopidy
- Was the software title installed freshly or updated/migrated? YES
- Can this issue be replicated on a fresh installation of DietPi? I think so ...
- `dietpi-bugreport` ID

#### Steps to reproduce:
1. Install with dietpi-software vanilla mopidy
2. Reboot
3. Open mopidy frontend, add file (in my case a stream) to queue
4. Try to save queue as playlist

#### Expected behaviour:
- new playlist with file should be present in mopidy frontend section "Playlists"

#### Actual behaviour:
- mopidy frontend section "Playlists" is empty

#### Extra details:
- /var/log/mopidy/mopidy.log says: 2019-01-01 18:55:08,061 ERROR [2333:M3UBackend-4] mopidy.m3u.playlists: Error creating playlist [Radio Streams]: Permission denied
- /etc/mopidy/mopidy.conf says in section [m3u] playlists_dir = /mnt, which is owned by root and is not writable by user mopidy

#### My workaround (possible solution for dietpi setup scripts?!)
- mkdir /mnt/dietpi_userdata/mopidy/m3u, chmod/chown mopidy as owner (like the data and cache dir)
- nano /etc/mopidy/mopidy.conf, in section [m3u] playlists_dir = /mnt/dietpi_userdata/mopidy/m3u
- systemctl restart mopidy
- ... then I can add a playlist in mopidy frontend which is stored as .m3u8

cyberlussi, 2019-01-02:
Thank you for the extensive answer. So effectively it should work with my proposed workaround and also with yours. Using the already existent Music dir makes sense.
So I changed mopidy's m3u/playlists_dir to `/mnt/dietpi_userdata/Music`. But at least in my installation the group `dietpi` does **not** have `mopidy` as member: `getent group dietpi` gives `dietpi:x:1000:` and mopidy complains as before to have no permissions to write in `/mnt/dietpi_userdata/Music`. So I did ` usermod -a -G dietpi mopidy` as you suggested, then mopidy is able to write the m3u8 file after a service restart: `ls -la /mnt/dietpi_userdata/Music/` gives `-rw------- 1 mopidy audio       56 Jan  2 14:47 [Radio Streams].m3u8`

MichaIng, 2019-01-02:
@cyberlussi 
Jep, we used `useradd` with option to define user groups and such, but the (new) APT install already creates the user, which is why we need to use `usermod`. I resolved that with the PR. I guess either this changed with moreless recent Mopidy versions or we used a different install method in the past (e.g. build from source), which did not include the user creation.

Fourdee, 2019-01-04:
@MichaIng 

Thanks, great work, appears to be resolved in v6.20:
[image]

Patch reinstall for existing installations?

closed 2019-01-28 by Fourdee