MichaIng/DietPi #2386: DietPi-Software | Allo Web Gui - SqueezeLite DSD - DoP KO

volpone, 2019-01-02:
#### Required Information:
- DietPi version: 6.19.7 master
- Distro version: stretch
- Kernel version: Linux USBridge 3.10.38
- SBC device: Sparky SBC (armv7l)

#### Steps to reproduce:
- Using Allo Web Gui
- SqueezeLite output set to DSD native "disabled" 
- Play DSF64 from LMS 7.9.2 ( Settings > Filetype > DSF/DFF set to "native")
- SqueezeLite output converted PCM to my DAC vs DSD encapsulated via DoP
- nano /etc/systemd/system/squeezelite.service:
[Unit]
Description=squeezelite
[Service]
Type=simple
ExecStart=/usr/bin/squeezelite -a 4096:8096:32:0 -C 5 -n 'DietPi-Squeezelite' -f /var/log/squeezelite.log
- SqueezeLite parameter -D (or -D :dop) is not set so according to SqueezeLite manual DSD is converted to PCM.

#### Expected behaviour:
ExecStart=/usr/bin/squeezelite -a 4096:8096:32:0 -C 5 -n 'DietPi-Squeezelite' -f /var/log/squeezelite.log -D (or -D :dop)

OK confirmation when i launch SqueeLite manually from:
/usr/bin/squeezelite -D -a 4096:8096:32:0 -C 5 -n 'DietPi-Squeezelite' -$

#### Extra details:
However DSD Native output seems OK via Allo Web Gui (-D :u32le or u32be)
Usage: squeezelite [options]
...
  -D [delay][:format] Output device supports DSD, delay = optional delay switching between PCM and DSD in ms
   format = dop (default if not specified), u8, u16le, u16be, u32le or u32be.

volpone, 2019-01-03:
Perhaps the best way is to modify Allo Web Gui "DSD Native Output" UI dropdown selector and add a new DoP option:

**- Disabled:**: _unchanged (PCM output)_ -> No -D option
**- DoP** : _add this new option to allow DoP Output_ -> **-D :dop**
**- u8 | u16le | u16be | u32be | u32be** : _unchanged (native DSD output)_ -> -D :u8|u16le|u16be|u32be|u32be

**Change UI and help tool tip wording:**
"DSD Output:" vs "DSD Native Output:"
- Disabled: DSD is converted to PCM
- DoP: DSD is sent as DoP
- All other values ... unchanged

Fourdee, 2019-01-03:
@volpone 

Hi many thanks for the report/request 👍 

We'll open for investigations.

volpone, 2019-01-03:
Thank's Dan,
IMHO this is a bug and not a "feature request".
DoP output is not possible, and a lot of users use DoP to feed DACs. 
Disabled option output PCM and not DoP DSD

volpone, 2019-01-23:
Thank's @sparkysbc  !

Seems OK with current Squeezelite version used with DietPi (v1.8.7-1020) 
However, last DietPi version (not used) seems to handle differently -D options. See  
https://ralph-irving.github.io/squeezelite.html

**-D [delay]**
_Output device supports DSD over PCM (DoP). DSD streams will be converted to DoP before output. If this option is not supplied, DSD streams will be converted to PCM and resampled, so they can be played on a PCM DAC. Delay is an optional time to wait when switching between PCM and DoP between tracks, in milliseconds._

MichaIng, 2019-05-01:
@volpone 
I just checked pre-compiled binaries from: https://sourceforge.net/projects/lmsclients/files/squeezelite/linux/

First I accidentally downloaded the pulse version, which indeed shows (after installing pulseaudio):
```
root@VM-Stretch:/tmp# ./squeezelite -?
...
Squeezelite v1.9.1-1137, Copyright 2012-2015 Adrian Smith, 2015-2019 Ralph Irving. See -t for license ...
  -D [delay]            Output device supports DSD over PCM (DoP), delay = optional delay switching between PCM and DoP in ms
```

Then I downloaded the non-pulse version:
```
root@VM-Stretch:/tmp# ./squeezelite -?
Squeezelite v1.9.1-1137, Copyright 2012-2015 Adrian Smith, 2015-2019 Ralph Irving. See -t for license terms
...
  -D [delay][:format]   Output device supports DSD, delay = optional delay switching between PCM and DSD in ms
                         format = dop (default if not specified), u8, u16le, u16be, u32le or u32be.
```

So the man page seems to be related to the pulse version which seems to only allow switching between native DSD (no argument) and DoP (`-D`).
However both binaries can simply be used with `-D` since: `format = dop (default if not specified)`

I already wanted to simply overwrite our binaries with the v1.9 ones but found the enormous difference in size:
```
-rwxrwxr-x 1  501  501 2642416 Mar 18 15:36 squeezelite # from SourceForge
-rwxr-xr-x 1 root root  144512 Nov 30  2017 squeezelite_amd64 # from dietpi.com
```

The argument could be added/removed by adding/removing `/etc/systemd/system/squeezelite.service.d/dietpi-dop.conf` drop-in config which overrides the `ExecStart=` or environment file, e.g. `/etc/default/squeeselite` which contains a `DOP=''` variable and if already there an `OPTIONAL_ARGS=''` so users can further customize.
- So this can be used by Allo GUI then but of course as well when installing SqueezeLite as standalone application, adding the info about this to our online docs.

@Fourdee 
Which magic did you use to create such tiny binaries? Or is it some compile time option (e.g. ffmpeg and such) which raise the size?
Recompiling could be done but IMO is not critical since above request/issue can be solved with v1.8.7 as well.

I found code to remove an init.d service and purge a package when uninstalling SqueezeLite in DietPi-Software. Did we ship this via APT package? "Blame" does not work on dietpi-software since, I guess, the script is too large. Just face the angry Unicorn that page loading takes too long 😆.
... Answered it myself: https://packages.debian.org/stretch/squeezelite
- Sadly even on Buster and Sid there is no current version shipped by Debian.

volpone, 2019-05-02:
Thank's @MichaIng !

volpone, 2019-05-30:
Some hope next DiePi release will include latest SqueezeLite ?. 
Current version (1.8.4) is quite old.
Thank you.

MichaIng, 2019-05-30:
@volpone 
It's on the list, see milestone 🙂.

volpone, 2019-05-30:
Great @MichaIng !

MichaIng, 2020-06-30:
New squeezelite 1.9.7 packages are ready for testing:
```
systemctl disable --now squeezelite
mv /etc/systemd/system/squeezelite.service{,_bak}
cd /tmp
wget https://dietpi.com/downloads/binaries/$G_DISTRO_NAME/squeezelite_$G_HW_ARCH_NAME.deb
apt install ./squeezelite_$G_HW_ARCH_NAME.deb
rm squeezelite_$G_HW_ARCH_NAME.deb
```
SSL support has been additionally enabled.
The service runs now as own user "squeezelite", so especially check if any permissions are missing. It is added to "audio" group, so generally audio device access is granted.

Allo web GUI support for switching native DSD will be at latest added with next DietPi release, but this one is in beta stage already.