MichaIng/DietPi #2388: Docker logging to filesystem - suggesting journald log-driver as default

SaturnusDJ, 2019-01-02:
Correct me if I am wrong but:

I found that docker containers log to  _/mnt/dietpi_userdata/docker-data/containers/[Container ID]/[Container ID]-json.log._ This is on non tmpfs, thus SD card in many cases.

DietPi by default has journald as logger log to _/run/log_, which is tmpfs.

Docker can logs can be passed to journald.
Therefor on a stock DietPi the /etc/docker/daemon.json file should look like:
```
{
    "data-root": "/mnt/dietpi_userdata/docker-data",
    "log-driver": "journald"
}
```
I am suggesting to discuss here if it would be a good idea to have this as default with DietPi.

Reporting this as bug since a) I consider it as design flaw on SD-card systems (as many DietPi users will have) for the many writes that can occur on e.g. info level logging, which is default in some applications, and b) as it is not in line with the logging philosophy presented in _dietpi-software_. Welcoming other's opinions on this.

MichaIng, 2019-01-02:
@SaturnusDJ 
Thanks for sharing your suggestion.

Yeah if it's about me, I would move really all logging to systemd (I did on my system, where possible). It is by default on RAM, but can be as well (with custom rotation as desired) written permanently to disk. Sadly it is not even possible for all system and software logs. APT, dpkg and other system logs are still forcefully to /var/log (without moving them manually/hackish), and many software titles do not allow logging to syslog daemon.

The only advantage of disk log I see is that they can be read out easily on any other ext4 supporting system with text editor, without special systemd reader.

To resolve `/var/log` to disk, we have DietPi-RAMlog. But again, if the software supports journald logging, is does not make much sense to let it log to `/var/log` and mount this to RAM manually. And of course it does not apply to log dirs outside of /var/log.

In case of docker, the main question is, if users mostly wish to have container logs persistent or not. If there is some need to have this persistent, then at least `/mnt/dietpi_userdata/...` on SDcard systems is the better solution, since this should on production system anyway be moved to an external HDD/SSD, where you potentially have better performance and which deal muuuch better with constant I/O than SDcards.
____________
**To sum up: I would go with your journald solution and add info to our online docs, how to (re)apply persistent disk logging, if desired.**
@Fourdee 
What do you think?

SaturnusDJ, 2019-01-06:
@MichaIng 
I am familiar with basic Docker functionality for some weeks now, not an expert. What I found is https://docs.docker.com/v17.09/engine/admin/logging/overview/#use-environment-variables-or-labels-with-logging-drivers.

_None_ would result in no logs.
_Json-file_ is default. It logs to a json-file in the root of the container meta volume. You would expect to be able to log to custom filesystem location, but I could find a single hint on the whole internet. A soft link would be an option but then a program needs to intercept the json-file and create a soft link each time a new container is created. Not nice.
_Syslog_ seems to be not running on DietPi?
_Journalct_, I don't know it well. It just worked for me. If this has options to log to folder, then that would come closest.
The other options require external programs, which extremely likely would make 'raw' logging to /var/log possible.

Then there is the option of using custom 'plug-in' logging drivers, yet I couldn't find one.

SaturnusDJ, 2019-01-06:
Yes I was sampling a log. 😃
[Docker_Home_Assistant_30_mins_log.txt](https://github.com/Fourdee/DietPi/files/2730561/Docker_Home_Assistant_30_mins_log.txt)

This is Home Assistant on default logging level.
For example why scan Bluetooth that much? Because if the car comes back the key to which the Bluetooth tracker is attached needs to be found quickly in order to immediately turn on the hallway light.

I found that the Docker container log contains a lot more entries than the log that Home Assistant shows you from inside its webinterface. Which at first was a deception to me for the real log.

The worst is when there is a configuration error. It's possible that every x seconds a whole block of 20 lines of code will be logged. This happened for me in a lesser way with DSMR Reader. When the MQTT element is not configured, it will keep logging that it spawned a MQTT process but didn't find a configuration.

Meanwhile I've put the Home Assistant default logging level to warn and stopped the DSMR Reader MQTT process spawning. Except for on purpose Home Assistant debugging and testing, 4 days without logs for non Home Assistant containers.

It's absolutely clear that in no way DietPi or its config is to blame. It's just that software does not expect the storage to be a potential problem thus DietPi can be a life saver here.

Fourdee, 2019-01-09:
@SaturnusDJ 

Many thanks for the report 👍 

@MichaIng 

> To sum up: I would go with your journald solution and add info to our online docs, how to (re)apply persistent disk logging, if desired.
@Fourdee
What do you think?

Yep, definitely 👍

MichaIng, 2019-01-18:
Hmm, the amount of `INFO` logs IMO is too much for systemd. I mean its personal state and how you use journalctl, but I personally regularly check it to get an overview about if everything is in order, any service, system or kernel errors and such, and this only works comfortable, if there are no continuous info logs, but at best warnings and system/kernel related entries.

But the logging mode can be adjusted:
- https://docs.docker.com/v17.09/engine/reference/commandline/dockerd/#daemon-configuration-file
- `"debug": false,` # If not yet the case
- `"log-level": "warn",`

Indeed there seems to be no way to customize the json-file log location. This will always be within the individual docker containers data dir. journald logging mode thus is the only RAMlog solution, AFAIK.

MichaIng, 2019-07-17:
Log level applies to `dockerd` but is not passed through to `containerd`, thus all container logs revert to default level "info", which is IMO too much to flood journald with.

There was a patch to address exactly this issue, but it seems to have gone until current version: https://github.com/moby/moby/commit/aaa1392279ffe67a63581d2a43b3a3c7b0fc47b1

MichaIng, 2020-09-26:
PR up: #3791
Works well on first test, good reason to finally add some online docs to inform users where to search for Docker and container logs:
- `journalctl -u docker`
- `journalctl -u containerd`

Log level reduced to warnings to prevent journald flood.

Since the Docker config file does not exist by default, exposing it is a bonus by itself and we might want to add a link to docs inside so users easily find out how to further configure Docker or restore file logs 😉.

closed 2020-10-17 by MichaIng