MichaIng/DietPi #2389: Native PC cannot update from v6.16 to v6.19

Elluminatus, 2019-01-03:
#### Details:
- Date           | Thu  3 Jan 22:25:13 CET 2019
- Bug report     | N/A
- DietPi version | v6.16
- Img creator    | n/a
- Pre-image      | n/a
- SBC device     | Native PC (x86_64) (index=21)
- Kernel version | #1 SMP Debian 4.9.88-1+deb9u1 (2018-05-07)
- Distro         | stretch (index=4)
- Command        | G_AGI: xserver-xorg-video-intel libgl1-mesa-dri
- Exit code      | 100
- Software title | DietPi-Config

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:
Corret update from 6.16 to 6.19
- ...

#### Actual behaviour:
I try to update via dietpi-update from 6.16 to 6.19, but it is stopping with failure

#### Extra details:
Tried to remove, reinstall and install firmware-amd-graphics but it is still not working. Every time there is the failure below.

#### Additional logs:
```
Log file contents:
dpkg: error processing package firmware-amd-graphics (--configure):
 package is in a very bad inconsistent state; you should
 reinstall it before attempting configuration
Errors were encountered while processing:
 firmware-amd-graphics
E: Sub-process /usr/bin/dpkg returned an error code (1)

Elluminatus, 2019-01-03:
I dont need any driver. Best case would be that there is no amd-graphics. But I cant delete it from the system. I have a headless server.

MichaIng, 2019-01-03:
@Elluminatus 
Thanks for your report.

I think it's coming from here: https://github.com/Fourdee/DietPi/blob/a61184cccb1cd8eaeaf5fe03d76d3e9c23a7b643/dietpi/dietpi-config#L925

Must be called from a reinstall, but I can't find the related one, especially no sure how a certain drive is directly toggled 🤔.

And now `xserver-xorg-video-intel` is an Intel video driver for xserver, while `firmware-amd-graphics` is the AMD video driver package. Possible that those conflict each other.

If you have a headless server, remove both of them:
- `G_AGP firmware-amd-graphics` did not work you say? Did you try `dpkg -P firmware-amd-graphics` otherwise?
- Looks like Intel driver was selected through dietpi-config menu. Please go to `dietpi-config > Display options > Change Resolution` and check where it say `Intel: Uninstall GPU driver` and if yes, do it.

Just to be sure, which software titles do you have installed? Maybe I we can identify which one was reinstalled and toggled the Intel driver install: `dietpi-software list | grep ' =2 '`

Elluminatus, 2019-01-04:
Hi MichaIng thank you for the fast reply. I tried `dpkg -P firmware-amd-graphics` with the same result:

> dpkg: error processing package firmware-amd-graphics (--purge):
 package is in a very bad inconsistent state; you should
 reinstall it before attempting configuration
Errors were encountered while processing:
 firmware-amd-graphics

Intel driver is nbot selected in the menu. There are no GPU driver installed there.

This is the output from `dietpi-software list | grep ' =2 '`

> id 0 | =2 | openssh client:  |
id 1 | =2 | samba client: access network shares | | https://dietpi.com/phpbb/viewtopic.php? dietpi-config > network options: nas/misc
id 3 | =2 | mc: midnight commander, powerful file manager |
id 7 | =2 | ffmpeg: audio & visual libary |
id 8 | =2 | java: openjdk 8 + jre libary |
id 9 | =2 | node.js: javascript runtime |
id 16 | =2 | build-essentials: common packages for compile |
id 17 | =2 | git client: git clone etc |
id 74 | =2 | influxdb: time-series database | | https://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=120#p12523
id 77 | =2 | grafana: platform for analytics and monitoring | | https://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=12524#p12524
id 87 | =2 | sqlite: database | | https://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=1335#p1335
id 96 | =2 | samba: feature-rich file server | | https://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=10#p56
id 103 | =2 | dietpi-ramlog: minimal, optimized logging | | https://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=20#p68
id 105 | =2 | openssh server:  | | https://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=10#p63
id 123 | =2 | mosquitto : mqtt messaging broker | | https://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=90#p4293
id 130 | =2 | python pip: python pip package installer |
id 150 | =2 | mono: runtime libraries and repo |

The INtel driver was a failure from myself. I tried it in the menu. 

Do you need further output or other Infos from me?

Greetings Elluminatus

MichaIng, 2019-01-04:
@Elluminatus 
> I tried dpkg -P firmware-amd-graphics with the same result:

Strange that it wants to configure the package, when attempting to purge. However perhaps try to follow the advice: `you should reinstall it before attempting configuration`: `apt install --reinstall firmware-amd-graphics`

If that throws the same error, the last idea I have so far is manually downloading the deb package: https://deb.debian.org/debian/pool/non-free/f/firmware-nonfree/firmware-amd-graphics_20161130-4_all.deb
or search for it in cache: `/var/cache/apt/archives/`
and install via `dpkg -i <name>.deb`.

MichaIng, 2019-01-04:
Checked reinstalls: Only `Mosquitto` during v6.17 => v6.18 patch, but that is a very simple install without much configuration.

So I guess you copied the error prompt when trying to switch to Nvidia driver manually? However so the main issue is the broken AMD driver package that breaks all APT attempts.

Elluminatus, 2019-01-04:
Perfect: apt install --reinstall firmware-amd-graphics wortes fine and nie its runing smooth. Update complete v6.19.7 Thank you

MichaIng, 2019-01-04:
@Elluminatus 
Great to hear. Now you should be also able to uninstall it: `G_AGP firmware-amd-graphics`

I will mark this issue as closed. Feel free to still use it or reopen, if required.

closed 2019-01-04 by MichaIng