MichaIng/DietPi #2390: RPi 3 Bluetooth | Cannot access adapter

jimbroze, 2019-01-03:
## Creating a bug report/issue:

#### Required Information:
- DietPi version	| `6.19.7`
- Distro version	| `Stretch`
- Kernel version	| `Linux DickWeb 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux
`
- SBC device		| `RPI 3`
- Power supply used	| (Official Raspberry Pi Power Supply for RPi 3)
- SDcard used		| (SanDisk ultra)

#### Steps to reproduce:

- Attempting a `hcitool scan` gives:
`Device is not available: Address family is not supported by protocol`
- `bluetoothctl` gets stuck on:
`Waiting to connect to bluetoothd`
- Trying to start the bluetooth service does not appear to fail but then the status shows as:
```
● bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
   Active: inactive (dead)
     Docs: man:bluetoothd(8)
```
- Attempting to restart hciuart.service results in:
```
● hciuart.service - Configure Bluetooth Modems connected by UART
   Loaded: loaded (/lib/systemd/system/hciuart.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2019-01-03 22:13:54 GMT; 8s ago
  Process: 9812 ExecStart=/usr/bin/btuart (code=exited, status=1/FAILURE)

Jan 03 22:13:24 DickWeb systemd[1]: Starting Configure Bluetooth Modems connected by UART...
Jan 03 22:13:54 DickWeb btuart[9812]: Initialization timed out.
Jan 03 22:13:54 DickWeb btuart[9812]: bcm43xx_init
Jan 03 22:13:54 DickWeb systemd[1]: hciuart.service: Control process exited, code=exited status
=1
Jan 03 22:13:54 DickWeb systemd[1]: Failed to start Configure Bluetooth Modems connected by UAR
T.
Jan 03 22:13:54 DickWeb systemd[1]: hciuart.service: Unit entered failed state.
Jan 03 22:13:54 DickWeb systemd[1]: hciuart.service: Failed with result 'exit-code'.
```

- Reinstalling `pi-bluetooth` package also results in a similar hciuart failure:
```
● hciuart.service - Configure Bluetooth Modems connected by UART
   Loaded: loaded (/lib/systemd/system/hciuart.service; enabled; vendor preset:
enabled)
   Active: failed (Result: exit-code) since Thu 2019-01-03 19:00:53 G
MT; 2h 51min ago
  Process: 346 ExecStart=/usr/bin/btuart (code=exited, status=1/FAILURE)

Jan 03 19:00:53 DickWeb btuart[346]: Can't set line discipline: Invalid argument
Jan 03 19:00:53 DickWeb btuart[346]: Can't initialize device: Invalid argument
Jan 03 19:00:53 DickWeb btuart[346]: bcm43xx_init
Jan 03 19:00:53 DickWeb btuart[346]: Flash firmware /lib/firmware/brcm/BCM43430A
1.hcd
Jan 03 19:00:53 DickWeb btuart[346]: Set BDADDR UART: b8:27:eb:79:5f:f8
Jan 03 19:00:53 DickWeb btuart[346]: Set Controller UART speed to 921600 bit/s
Jan 03 19:00:53 DickWeb systemd[1]: hciuart.service: Control process exi
ted, code=exited status=1
Jan 03 19:00:53 DickWeb systemd[1]: Failed to start Configure Bluetooth
Modems connected by UART.
Jan 03 19:00:53 DickWeb systemd[1]: hciuart.service: Unit entered failed
 state.
Jan 03 19:00:53 DickWeb systemd[1]: hciuart.service: Failed with result
'exit-code'.
```

#### Extra details:

- The SD card in a different RPI 3 provides the exact same result; suggesting that this is not a hardware issue.

MichaIng, 2019-01-03:
@jimbroze 
Thanks for your report.

I mark for investigation.

jimbroze, 2019-01-07:
Also tried a completely fresh DietPi installation on another SD card, Installed nothing but pi-bluetooth and experienced the same issue.

MichaIng, 2019-01-07:
@jimbroze 
Did you actually enable bluetooth functionality via `dietpi-config` or:
`/DietPi/dietpi/func/dietpi-set_hardware bluetooth 1`

@Fourdee 
Just recognized that we don't have a bluetooth setting in `dietpi.txt`. I mean enabling it after 1st run setup is quite simple, but this would also give users a hint THAT you can toggle it and it's actually disabled by default.

jimbroze, 2019-01-07:
@MichaIng 
Wow, I feel incredibly stupid now. I looked in dietpi.txt and in the network settings but never thought to check advanced settings.
Problem solved!

closed 2019-01-07 by jimbroze

MichaIng, 2019-01-07:
@jimbroze 
No worries, indeed it might not fit too good into advanced settings.

We already started to think about a better fitting reorder of `dietpi-config`, but didn't manage to start with this until now: https://github.com/Fourdee/DietPi/issues/639

There was the idea to merge all hardware/protocol connection related settings into a "Connectivity" menu.