MichaIng/DietPi #2391: Errors at first boot

svh1985, 2019-01-04:
## Creating a bug report/issue:
Errors at first boot

#### Required Information:
- DietPi version	| `cat /DietPi/dietpi/.version`
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=17
G_DIETPI_VERSION_RC=0
G_GITBRANCH=master
G_GITOWNER=Fourdee
- Distro version	| `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
stretch
- Kernel version	| `uname -a`
Linux DietPi 4.14.71-v7+ #1145 SMP Fri Sep 21 15:38:35 BST 2018 armv7l GNU/Linux
- SBC device		| `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
RPi 2 Model B (armv7l)
- Power supply used	| (EG: 5V 1A RAVpower)
5V  2.6A
- SDcard used		| (EG: SanDisk ultra)
Samsung 32GB microSD EVO

#### Steps to reproduce:

1. Flash SD (w/ Etcher)
2. Set Wifi Password
3. Changed Hostname
4. Enabled WiFi for first boot
5. SSH into DietPi and got the errors

#### Expected behaviour:

- A freshly installed Pi without errors 🍻

#### Actual behaviour:

- Seeing errors:

> The programs included with the Debian GNU/Linux system are free software;
> the exact distribution terms for each program are described in the
> individual files in /usr/share/doc/*/copyright.
> 
> Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
> permitted by applicable law.
> chmod: changing permissions of '/DietPi/dietpi/.network': Operation not permitted
>  ────────────────────────────────────────────────
>  DietPi | Sun 28/10/18 - 15:03
>  ────────────────────────────────────────────────
>  v6.17.0 | RPi 2 Model B (armv7l)
>  ────────────────────────────────────────────────
>  wlan0 | 172.16.0.5
>  ────────────────────────────────────────────────
> [FAILED] DietPi-Software | Root privileges required. Please run the command with "G_SUDO", or, "sudo -i".
> mv: failed to access '/var/tmp/dietpi/logs/dietpi-firstrun-setup.log': Permission denied

svh1985, 2019-01-04:
Addition:
Running dietpi-config requires Root privileges, I think somethings wrong with the system privileges.

> dietpi@DietPi:~$ dietpi-config
> [FAILED] DietPi-Config | Root privileges required. Please run the command with "G_SUDO", or, "sudo -i".

svh1985, 2019-01-04:
I realised that I logged in with user dietpi and not root. Issue can be closed.

closed 2019-01-04 by svh1985

mikegilchrist, 2019-01-29:
So to be clear, after an upgrade one apparently needs to login as root, not dietpi to complete the upgrade.