MichaIng/DietPi #2394: WaveShare35 support

PDXUser, 2019-01-05:
Please add support for the [Waveshare 3.5" screen.  ](https://www.waveshare.com/wiki/3.5inch_RPi_LCD_(B)#Driver)

I have the Waveshare 3.5" Model B revision 2.0 which does not work, Dietpi does not turn the screen on when telling it to use the waveshare32 setups.

MichaIng, 2019-10-16:
Ref: https://dietpi.com/phpbb/viewtopic.php?f=15&t=6323

navdotnetreqs, 2023-05-24:
Answering my own questiong, killing off "fbcp" did the trick! Where is this started so I can disable it from running on boot?

MichaIng, 2023-05-25:
The preboot service should have long finished when login prompts appear, of course. What does this `fbcp` do? Does it somehow keep copying content from the console framebuffer to the LCD over and over again?