MichaIng/DietPi #2396: Minimizing write ops to SD-card

WolfganP, 2019-01-05:
#### Required Information:
- DietPi version	| 159 (yes, I know, old version... planning to update soon)
- Distro version	| 9.6
- Kernel version	| Linux DietPi 4.9.62-v7+ #2 SMP Fri Nov 17 23:52:26 GMT 2017 armv7l GNU/Linux
- SBC device		| RPi2
- Power supply used	| 5V 2A
- SDcard used		| (EG: SanDisk ultra)

#### Steps to reproduce:
So, as my RasPi acts as a 24/7 media / file / IoT server, I started a mission to reduce power usage (ie disconnect HDD while media/file server is not needed) and avoid unnecessary write ops to the SD card to improve the component life and system stability.

So, using `fnotifystat` I was relocating files to tmpfs / ext HDD and dynamically taking down unneeded processes, but there are a couple of functions that still keep writing to the SD card unnecessarily:

```
pi@DietPi:~$ sudo fnotifystat -vnT -x /mnt,/proc,/dev,/run,/tmp,/var/log | grep W\)

05/01/19 12:10:47   0.0  (-C-W)  1330 nmbd            /var/cache/samba/browse.dat.
05/01/19 12:11:07   0.0  (-C-W)  1330 nmbd            /var/cache/samba/browse.dat.
05/01/19 12:17:03   0.0  (---W)  4380 date            /etc/fake-hwclock.data
05/01/19 12:17:03   0.0  (-C-W)  4378 fake-hwclock    /etc/fake-hwclock.data
05/01/19 12:22:50   0.0  (OC-W)  1330 nmbd            /var/cache/samba/browse.dat
05/01/19 12:23:10   0.0  (OC-W)  1330 nmbd            /var/cache/samba/browse.dat
05/01/19 12:27:36   0.0  (OC-W)  1330 nmbd            /var/cache/samba/browse.dat (deleted)
05/01/19 12:27:36   0.0  (OC-W)  1330 nmbd            /var/cache/samba/browse.dat
05/01/19 12:34:52   0.0  (OC-W)  1330 nmbd            /var/cache/samba/browse.dat (deleted)
05/01/19 12:34:52   0.0  (OC-W)  1330 nmbd            /var/cache/samba/browse.dat
05/01/19 12:39:34   0.0  (OC-W)  1330 nmbd            /var/cache/samba/browse.dat
05/01/19 12:39:34   0.0  (OC-W)  1330 nmbd            /var/cache/samba/browse.dat
```

For `/var/cache/samba/browse.dat`I found this patch https://lists.samba.org/archive/samba-technical/2015-May/107144.html and I'm still investigating the issue (I'll report back if found more info / a solution)

For `/etc/fake-hwclock.data` being re-written regularly, I found https://github.com/Fourdee/DietPi/issues/2041 relevant, but all the discussion centers on needing it at boot time because the clock is not yet updated by ntp. 
So, is there any way to avoid that file write after the system is completely up and running?

Thx as always for any tip.

WolfganP, 2019-01-05:
Just for the sake of documentation in one single place, the write ops to the sdcard can be checked via `awk '/mmc/ {print $3"\t"$7}' /proc/diskstats` as per https://www.kernel.org/doc/Documentation/iostats.txt

For the `/var/cache/samba/browse.dat` case, I found this https://ubuntuforums.org/showthread.php?t=2166829 relevant and tested it succesfully. Converted to systemd specs, this can be easily automated as:

```
$ sudo systemctl edit nmbd

[Service]
ExecStartPre=sudo mkdir /var/run/samba-cache
```

I'll keep adding info if any additional findings.

MichaIng, 2019-01-05:
@WolfganP 
Thanks for opening this issue.

`fnotifystat` looks like a very interesting tool to improve any system according to disk I/O.

#### /var/cache/samba/browse.dat
- As long as there is no fix (only write that file if actual values have changed) available, you can surely link that file to RAM, e.g.:
```
mv /var/cache/samba/browse.dat /tmp/samba-browse.dat
ln -s /var/cache/samba/browse.dat /tmp/samba-browse.dat
```
- Just assure (although I am pretty sure) that:
  - a) It is really no issue that the content is lost on reboot
  - b) The file itself is re-created, if non-existent
- In case do a backup of it first, if you face any issues.

#### /etc/fake-hwclock.data
- Indeed this one should stay in place, otherwise on early boot (before network time sync) you will have 1970th timestamps, which might cause further issues down the road, especially if then network time sync fails for some reason. RPi has no own hardware clock 😉.
- But: `fake-hwclock` updates the file very late on shutdown to have the most current time on startup. The hourly update via `/etc/cron.hourly/fake-hwclock` is just a failsafe, if the system crashes (e.g. power loss), so no ordinary shutdown with fake-hwclock update is done.

WolfganP, 2019-01-05:
Thx @MichaIng , we were writing at the same time with the findings. See my previous reply with a solution for `/var/cache/samba/browse.dat`

Regarding `/etc/fake-hwclock.data` I'll adopt your suggestion, and reschedule to run more spaced, ie once a day, or every 6 hours, or ...

MichaIng, 2019-01-05:
@WolfganP 
Jap, I updated my post accordingly, see my suggestion about linking only the file, not the whole folder, and link target. But both solutions should work fine.

To have fake-hwclock running daily is pretty simply:
`mv /etc/cron.hourly/fake-hwclock /etc/cron.daily/fake-hwclock`

To have it run every 6 hours you need to create a custom crontab entry or systemd timer.

WolfganP, 2019-01-05:
> Jap, I updated my post accordingly, see my suggestion about linking only the file, not the whole folder, and link target. But both solutions should work fine.

I tested linking just the `/var/cache/samba/browse.dat` file but it didn't work. `nmbd` deletes the symlink and recreates the file on the folder at start; no issues with linking the folder thou. I'll keep investigating.

Fourdee, 2019-01-05:
In ```/etc/default/fake-hwclock``` you can set:
```
FILE=/tmp/fake-hwclock.dat
```

```
root@DietPi:~# ls -lha /tmp
total 8.0K
drwxrwxrwt  7 root root    180 Jan  5 19:42 .
drwxr-xr-x 22 root root   4.0K Oct 30 15:26 ..
-rw-r--r--  1 root root     20 Jan  5 19:42 fake-hwclock.dat
```

MichaIng, 2019-01-05:
@Fourdee 
> FILE=/tmp/fake-hwclock.dat

This does not work, since the file must be reboot persistent to do it's job. Even external drive is dangerous, if it is required before external drives are fully mounted, and of course, if the drive got detached. So this must stay on RootFS and I would also not touch the hourly safe to better prevent issues in case of power loss. Leave this to end user custom change, if really desired. But this file is tiny, and hourly disk touch should be really no issue, besides you are 100% enthusiast 😉.

Fourdee, 2019-01-05:
@MichaIng 

> This does not work, since the file must be reboot persistent to do it's job.

Ah good point 👍  we can do a service to save this to disk during shutdown and restore during boot.

MichaIng, 2019-01-05:
@Fourdee 
Ah good point 👍 we can do a service to save this to disk during shutdown and restore during boot.

Simply removing the cron job `/etc/cron.hourly/fake-hwclock` has same result: 1970th system clock, if system crashes due to power loss or other critical issue.
`fake-hwclock` creates/writes to this file only on late shutdown stage itself, while the cron job is to prevent urgently outdated system clock in case of unordered shutdown.

Fourdee, 2019-01-05:
@MichaIng 

> /etc/cron.hourly/fake-hwclock

Yep, so only issue now is if power loss, time will be reset to last boot time saved.

Shame ```/etc/cron.hourly/fake-hwclock``` doesnt simply run ```systemctl restart fake-hwclock```

MichaIng, 2019-01-05:
@Fourdee 
> Shame /etc/cron.hourly/fake-hwclock doesnt simply run systemctl restart fake-hwclock

? Would be just overload, since the service does nothing else (on stop) then what the cron job does?

As said, the current solution is a workaround for a result that we can achieve by simply removing the cron job (which will be due to our apt.conf.d settings) not automatically re-created on APT upgrade.

But I still do not like the workaround/solution. This is a very tiny file touch once an hour that prevents potentially large issues when the system starts with outdated clock due to crash/power loss or simply because SSH connection does not work on HDMI-less SBCs and users are forced to unplug power cable.

I vote to revert all changes about fake-hwclock here. This is really no issue, but, as it is by default, a meaningful implementation.

WolfganP, 2019-01-05:
Totally agreed with @MichaIng on `/etc/fake-hwclock.dat` case . Leave it as it is and if some advanced user wants to manually change the balance on wearing vs reboot stability, it's all documented in this issue.

The samba solution however, is transparent to the user and bigger offender in terms of SDcard writing, so it may be worthy to tackle for all users.

MichaIng, 2019-01-05:
@Fourdee 
Yes, it makes sense that it starts as early as every possible, also to have all following boot logs with correct time stamps and of course if something else depends on moreless correct system time.

Having another look onto the `FORCE` option: https://manpages.debian.org/stretch/fake-hwclock/fake-hwclock.8.en.html#COMMANDS
Actually from my point of view, besides for testing reasons, it does no make much sense to set this true? E.g. `systemd` already does an earlier check and sets the system clock to it's own release date (in case no real hwclock available):
`[    1.443606] systemd[1]: System time before build time, advancing clock.`
If we now ship an image, which was created before systemd release, with `FORCE=force`, `fake-hwclock` would set the time further into the past.
And of course if somehow before shutdown (and after last cron job update) the system clock got out of order, it would safe an older time.
All this should be relevant only in very rare cases, but at least I see no reason to actively set the FORCE option, but simply leave it as it is by default.
Or did some reason for this appear in the past?

MichaIng, 2019-01-06:
Okay about the samba cache:
- So this cache dir is used by different processes/services right? So we cannot bin the tmp dir creation to only one service.

Me by chance found the tmpfile systemd feature: https://manpages.debian.org/stretch/systemd/tmpfiles.d.5.en.html
- Used by systemd/packages: `/usr/lib/tmpfiles.d`
- To be used by admin: `/etc/tmpfiles.d`

So the following should work well, at best BEFORE samba APT install:
# Remove disk cache dir or previous symlink
[[ -e /var/cache/samba ]] && rm -R /var/cache/samba
# Pre-create RAM cache dir
mkdir -p /var/run/samba-cache
# Link disk cache to RAM
ln -s /var/run/samba-cache /var/cache/samba
# Create RAM cache dir automatically on boot
echo 'd /var/run/samba-cache - - - - -' > /etc/tmpfiles.d/dietpi-samba_cache.conf

Fourdee, 2019-01-07:
- Existing install patch + test
- New install patch + test

closed 2019-01-28 by MichaIng