MichaIng/DietPi #2403: Networking issue and a bug

krishnasen2401, 2019-01-09:
Problem 1
Config 1
1.wired connect DHCP (to tplink portable router)(just for control and readyshare media transfer)
2.wifi DHCP to mobile hotspot(for internet)
Config 2 (to confirm problem)
1.wired connect DHCP and static (normal router as lan)(just for control and readyshare media speed) with internet access
2.connected to the mobile hotspot

Observation
The raspberry pi is not able to connect to internet.
That the system is not able to connect to internet like normal system based on speed or connection status

The senario raspberry pi is used as media center which downloads(sonarr +transmission) and streams media connected to college wifi(medium strength signal) to stream to phone buffering occurs and disconnection if signal interruption so I had to connect via ethernet using the tplink wr702n router to control the device via android phone

Bug(install and uninstall)
Netdata monitoring doesn't appear in uninstall menu

Fourdee, 2019-01-10:
@krishnasen2401 

Many thanks for the report 👍 

> Netdata missing from uninstall

Unable to replicate on our test system:
[image]

The installation may have failed. Please try the following to force install flag, and allow removal:
- edit ```/DietPi/dietpi/.installed```
- change ```aSOFTWARE_INSTALL_STATE[69]=0``` to ```=2```
- Save changes
- ```dietpi-software``` will now show the uninstall option for netdata

> Observation
The raspberry pi is not able to connect to internet.
That the system is not able to connect to internet like normal system based on speed or connection status

DietPi is designed to be used with a single connection with internet (eg: ETH or WiFi). For advanced connections with multiple network cards, you'll need to manually configure routing on the device.
https://www.cyberciti.biz/faq/howto-linux-configuring-default-route-with-ipcommand/

krishnasen2401, 2019-01-10:
The write error guess due to unexpected shutdown maybe

Have also noticed that external HDD doesn't get uncounted similar to PC just remove the USB cord(only noticed after hearing the disk sound when we just remove it in middle)
sed: couldn't open temporary file /boot/sed62qGCF: Read-only file system         ( for changing autobot option)

This was when installing netdata
[ SUB1 ] DietPi-Software > Configuring Sonarr: automatically download TV shows sed: couldn't open temporary file /boot/sed70sEYO: Read-only file system
sed: couldn't open temporary file /boot/sedudxWWR: Read-only file system
sed: couldn't open temporary file /boot/sedRngwrS: Read-only file system

Fourdee, 2019-01-10:
@krishnasen2401 

> The write error guess due to unexpected shutdown maybe

Yep, you need to shutdown the system before pulling out the power. 

Also, before unplugging USB harddrives, you must dismount them using ```dietpi-drive_manager```. Not doing this will render filesystem corruptions.

krishnasen2401, 2019-01-10:
The unexpected power off was due to battery

I asking in case of "poweroff" command for system shutdown which stop each and every service I didn't notice it umounting the storage

Fourdee, 2019-01-10:
@krishnasen2401 

> I asking in case of "poweroff" command for system shutdown which stop each and every service I didn't notice it umounting the storage(the user data is not transferred only sonarr download are stored)

Once the device has powered off, the drive may still be powered. Depends on the SBC device used.

Regardless, once ```poweroff``` is used, and the system has completely shutdown, it is safe to unplug the HDD. No need to dismount for poweroff.

MichaIng, 2019-01-10:
In most cases poweroff should stop services and disk I/O gracefully,  so you can power/plug off the drives then safely. But especially with temporary mounted drives (not via fstab) it is possible that e.g. writes hang and the drive is not unmounted gracefully. So jep to be safe, unmount it manually first.

If you added the drive via dietpi-drive_manager, it is added to fstab, so included into the regular systemd schedule and there is no need to manually unmount for sure.

Fourdee, 2019-01-10:
@MichaIng 

>  it is possible that e.g. writes hang and the drive is not unmounted gracefully. So jep to be safe, unmount it manually first.

Good point 👍 Some USB <> SATA converters may cache writes on chip.

krishnasen2401, 2019-01-10:
I noticed it unmount when reboot(applying dietpi-config) it might be unmounted when poweroff as I can see boot partion also unmounted

closed 2019-01-10 by krishnasen2401