MichaIng/DietPi #2404: DietPi-Autostart | Auto login prompts for password

Fourdee, 2019-01-10:
> https://github.com/Fourdee/DietPi/issues/2352#issuecomment-452883078

Something has changed in SystemD?

MichaIng, 2019-01-14:
@Fourdee 
Perhaps due to changed boot script order?

Ah wait, nope above should be not related, but I just recognized, if not mistaken, that we do not enable lightdm service on choice: https://github.com/Fourdee/DietPi/blob/dev/dietpi/dietpi-autostart
But this is already for longer 🤔.

MichaIng, 2019-01-28:
Should be retested now with v6.20, as we did some systemd reorder which might have resolved the issue.

arrudagates, 2019-02-14:
same problem here, but on custom.sh

MichaIng, 2019-02-14:
@arrudagates 
Verify that you actually don't need to login and only the prompt shows up. You should be able to run commands etc.

I will provide some command step later to hide those login prompt in auto login case, until we found a nice solution with v6.22.

MichaIng, 2019-03-02:
- Login prompt will only be shown, if no autologin mode has been selected: https://github.com/MichaIng/DietPi/commit/98ffdf96ba1f8059d101b0f8f580d73a444086f0#diff-93b9c3871f032c905ec8b4505d408ca9
- An issue where custom.sh caused a dependency/order loop with postboot and/or other idle and/or multi-user.target has been fixed.

I mark this as closed.

closed 2019-03-02 by MichaIng