MichaIng/DietPi #2406: Running Certbot requires a web server but does not install one

noGitForYou, 2019-01-11:
## Creating a bug report/issue:

#### Required Information:
- DietPi version	| `cat /DietPi/dietpi/.version`

G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=19
G_DIETPI_VERSION_RC=7
G_GITBRANCH=master
G_GITOWNER=Fourdee

- Distro version	| `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
9.6

- Kernel version	| `uname -a`
Linux rpi3 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux

- SBC device		| `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
RPi 3 Model B (armv7l)

- Power supply used	| (EG: 5V 1A RAVpower)
5V 3.5 A USB power station

- SDcard used		| (EG: SanDisk ultra)
Sandisk Ultra 16 GB

#### Additional Information (if applicable):
- Software title	| (EG: Nextcloud)
Certbot

- Was the software title installed freshly or updated/migrated?
Yes, the Dietpi image was freshly installed.  Certbot and a few other optimized software titles were installed from dietpi-software.

- Can this issue be replicated on a fresh installation of DietPi?
Yes.
- `dietpi-bugreport` ID

#### Steps to reproduce:

1. Install Certbot, without having installed a web server.
2. Run dietpi-letsencrypt.

#### Expected behaviour:

- Dietpi-letsencrypt should run through without error
- If a web server is not installed when installing Certbot, it should install one based on the user's selected web server preference.

#### Actual behaviour:

- dietpi-letsencrypt throws an error when selecting 'apply'
- the error states that a web server is not installed (because it is not).

#### Extra details:

- This is easy to workaround by installing another package containing a web server, but ideally, Certbot would bring along a web server.  Lighttpd is also not available as a standalone install, so the user is forced to install additional packages that they may not need (php, mariadb).  It would be nice to have Lighttpd available as a standalone install too.

MichaIng, 2019-01-12:
@noGitForYou 
Thanks for your suggestion.

However I think it is fine as it is, for the following reason:
- `CertBot` itself does not depend on a webserver. It even brings it's own internal standalone webserver that can be used to get a certificate. E.g. you have a `Java` or `Node.js` application with web interface, so do not need a dedicated webserver, but certificates.
- `DietPi-LetsEncrypt` is another topic. it uses `CertBot` to apply SSL certificate and installs them to an installed webserver, so it depends on CertBot as well as on a webserver do something, that's true.
- So what we could do, is offering automated install of CertBot and the chosen webserver via `dietpi-software` directly from `dietpi-letsencrypt`, so users don't need to manually do that. But it is not 100% perfect, since currently `Minio S3` is also detected as webserver by DietPi-LetsEncrypt 🤔.

What I have in mind for the future of DietPi-LetsEncrypt is the following:
- Allow to choose whether to use CertBot, acme.sh (https://github.com/Fourdee/DietPi/issues/2363), create a self-singed certificate or enter the path to key+cert from external provider.
- It will download/renew the certificate independently from installed software, optionally use webroot or webserver method, if available, but otherwise use an independent method (standalone webserver or DNS validation).
- Then a separate menu allows to apply the certificate and enable SSL for installed software titles, e.g. webservers but as well other web interface software.

noGitForYou, 2019-01-13:
@MichaIng 
Thanks for your detailed reply.  I think your plan for the future makes good sense.  Your last comment highlights where my real concern lies.  First I got the error from dietpi-letsencrypt about not having a web server installed, and I would have happily installed Lighttpd manually from dietpi-software, but didn't want the whole stack.  Especially since installing a whole stack gives a somewhat cryptic warning message that makes you worry about screwing up your config.

It seems that if you view the software list using the command line with `dietpi-software list` instead of dietpi-software, there is an option to just install Lighttpd and PHP together (software ID 84), but it is not visible as an option in dietpi-software.  Also no entry for just installing the server without PHP.  I guess its not the end of the world to install a whole stack, I'm just a minimalist by nature and I don't need the database or PHP.

MichaIng, 2019-01-13:
@noGitForYou 
The initial intention was to clean up the menu and make it easier for users to fast find and select the stack they require. If we show the webservers and databases as well in the menu, the whole webserver category will look somehow messy, also due to the chosen install IDs (which we do not want to rearrange, every time out software offer changes). So I think either the stacks or only the individual webservers, databases and PHP (no stacks then) should show up.

What you can always do to install only a single software ID: `dietpi-software install 84`
But currently, webservers will always install PHP as hard coded dependency.
___________
Okay some thoughts:
- DietPi-Software has a search function for some time, which makes it easier to search through the large menu.
- There are indeed cases where one requires a webserver, database or PHP only.
- (Coding) We already separated webserver and PHP requirement arrays for this reason.
- For flexibility and to be consistent with the more flexible dependencies, I think we can switch/revert from webserver stack to individual webserver, database and PHP selection. If anything as dependency is missing, `dietpi-software` will add it automatically anyway.
- Only thing that we could enhance, although not required urgently:
  - Currently all webservers have PHP as dependency and also from config code it is expected to be installed.
  - We could make it more flexible by configure the webserver to use PHP only, if it's actually installed or marked for install.
  - But this is a larger effort, so something that would be done by times, webserver by webserver.

@Fourdee 
What do you think?

Fourdee, 2019-01-21:
@MichaIng 

> So I had to select a stack because dietpi-letsencrypt did not automatically install the required web server (and the web server is not available individually for install in dietpi-software). 

Webserver stack should be a pre-req for Certbot, to be consistent with what users expect from our install system.

> , I think we can switch/revert from webserver stack to individual webserver, database and PHP selection.

I believe this will open a can of worms and increase our support level. I'd personally prefer to only allow users to install a web-server stack that we pre-define.

Hmm, these should not install PHP/SQL unless they are specified.
```
		aSOFTWARE_REQUIRES_WEBSERVER[$software_id]=1
			  aSOFTWARE_REQUIRES_PHP[$software_id]=1
		   aSOFTWARE_REQUIRES_SQLITE[$software_id]=1
```

MichaIng, 2019-01-21:
@Fourdee 
Jep, databases can already be skipped, PHP currently not, if webserver selected. The auto pre-req selection will select PHP with any webserver automatically, although we have it as separate dependency array.

> Webserver stack should be a pre-req for Certbot, to be consistent with what users expect from our install system.

At least currently Minio S3 object server is handled separately without any webserver. So DietPi-LetsEncrypt will not show any error, if this is installed.

However, issue should be solved with: https://github.com/Fourdee/DietPi/issues/2363#issuecomment-449649984
- So the script would allow to enable HTTPS for various software titles without webserver as well, if installed. It will not depend on any webserver, but, in case of CertBot, using it's standalone module to receive the cert.
- Only downside in case of webserver is, that we need to configure HTTPS manually. But we do that for Lighttpd already and not hard to achieve for Nginx/Apache via drop-in configs, based on: https://mozilla.github.io/server-side-tls/ssl-config-generator/

MichaIng, 2026-01-05:
Done with DietPi v9.10: https://github.com/MichaIng/DietPi/commit/9507569