MichaIng/DietPi #2411: DietPi-Software | Tomcat version 9 available in Debian repo

MichaIng, 2019-01-13:
- For Stretch via backports: https://packages.debian.org/stretch-backports/tomcat9

Fourdee, 2019-01-16:
62 installs, will flag as low priority + info

MichaIng, 2019-01-16:
@Fourdee 
Jep basically the question is, if an existing Tomcat8 instance can be upgraded safely to Tomcat9. Simply installing the APT package at least is a simple task.

MichaIng, 2021-05-27:
Added to next version milestone. We'll basically remove the Tomcat 8 install option, which is only available on Debian Stretch, where we don't ship any images for since a while, and replace it with Tomcat 9, only available on Debian Buster and Bullseye, and theoretically via `stretch-backports`.

MichaIng, 2021-06-22:
Actually there is not really a point in keeping Tomcat in our software list. We'd now migrate to tomcat9, tomcat10 is already in the pipeline (experimental suite) and the tomcat8 package in Raspbian Buster and above is actually a pretty outdated one while on Stretch there is a much newer one available already. And all that we do aside of installing the APT package is setting the JAVA_HOME setting, which is not required as Tomcat is able to find it automatically, when unset, based on common OpenJDK and Oracle JDK paths.

I'll hence remove the install option now, refer to simply using APT to install tomcat9 instead on current Debian releases. If a new reason appears, e.g. for reasonably change the default config or setup another Java application with Tomcat automatically, we can re-implement it with new version and proper config block.

closed 2021-06-22 by MichaIng