MichaIng/DietPi #2414: No reaction when double-clicking on the desktop.

Hoffnungsschimmer, 2019-01-15:
Hello, I am not sure if it is a bug or it is up to me, fact is I can not find anything on the internet. 
Am also more of a user than a programmer (and English is not my mother tongue).
I explain it briefly, I can not start desktop icons by double-clicking or open a folder. 
If I try to open a folder by double-clicking a "Rename" window appears. Nothing happens with an icon, neither Htop nor the standard Dietpi icons.
I run a right click and click open, it works. Even if I click on the same in / root / Desktop it works.

I reinstalled everything to test, with the same result. What am I missing ??

- DietPi version	v6.19.7
- Distro version	stretch
- Kernel version	4.14.79-V7
- SBC device		RPi 3 Model B
- Power supply used	5V 2A
- SDcard used		SanDisk ultra

MichaIng, 2019-01-15:
@Hoffnungsschimmer 
Thanks for your report.

Which desktop are you using? LXDE?

Hoffnungsschimmer, 2019-01-15:
I use (as always) AutoStart Option 2 (automatic login). I think that's the LightDM.

Hoffnungsschimmer, 2019-01-15:
root@DietPi:~# grep '\]=2' /DietPi/dietpi/.installed
aSOFTWARE_INSTALL_STATE[6]=2
aSOFTWARE_INSTALL_STATE[23]=2
aSOFTWARE_INSTALL_STATE[103]=2
aSOFTWARE_INSTALL_STATE[104]=2
aSOFTWARE_INSTALL_STATE[120]=2
root@DietPi:~#

Hoffnungsschimmer, 2019-01-15:
aSOFTWARE_INSTALL_STATE[6]=2
aSOFTWARE_INSTALL_STATE[23]=2
aSOFTWARE_INSTALL_STATE[103]=2
aSOFTWARE_INSTALL_STATE[104]=2
aSOFTWARE_INSTALL_STATE[120]=2
root@DietPi:~#

MichaIng, 2019-01-15:
@Hoffnungsschimmer 
I have no RPi free for testing here, but will check in VM for obvious issues.

MichaIng, 2019-01-15:
@Hoffnungsschimmer 
Okay on VM double clock works fine.

As you installed a VNC server, could you check if when connecting remotely to the desktop, the double click works? This could sort out some local configuration issue.

Hoffnungsschimmer, 2019-01-16:
Yes I use VNC.
It had already tested yesterday, with the same result.
I just mark the icon, the text background turns blue. For a folder on the desktop, the rename window appears. Icons on the taskbar work without problems.
For testing I took a RPi 3 B + (2017), but there also the phenomenon.
Then I flashed a second SD card for the B + (DietPi_v6.17_RPi-ARMv6-Stretch) which then updated to 6.19.
But, also here the same double-click problem .....

Fourdee, 2019-01-21:
@Hoffnungsschimmer 

Very strange, I was unable to replicate in testing with RealVNC. Both shared and dedicated desktops were fine.

Please can you confirm on the client system (not the RPi):
- VNC client used?
- OS running on the VNC client?
- Any ALT/CTRL keys that might be enabled?

Hoffnungsschimmer, 2019-01-23:
It's a bit frustrating.
I use the RealVNC on a WIN10 and on a Win7 PC.
Have also accessed from another Raspberry (Raspbian V9 stretch) via VNC on the Dietpi.
Again, I can not double-click, the same as the other try.
I also tried it with two Cherry keyboards G84-4400LUBEU and DW5100.
Languages tested GB and DE. Also with a radio and cable mouse.
Different mouse button combinations tried ...

I still use some with the Dietpi v6.2, there everything works.
If I make an update (v6.19.7), the "error" comes.

Fourdee, 2019-01-25:
@Hoffnungsschimmer 

Very strange.

Things to try:
- Uninstall RealVNC and try VNC4 server instead. The RealVNC viewer will work with this also.
- Unplug any USB peripherals on the RPi

If you can zip/7z the ```.img``` once read, will reduce the size for upload.

Hoffnungsschimmer, 2019-02-11:
I try all these things, with the same result.

closed 2019-02-11 by Hoffnungsschimmer