MichaIng/DietPi #2416: DietPi-Software | LXDE: Weather plugin fails to retrieve data and find new locations

MichaIng, 2019-01-15:
- On fresh install it shows a warning rectangle and `N/A`.
- If you click the tray icon, it says: `Forecast for london unavailable`
- If you go into plugin settings, London UK is pre-configured as location, but it is not possible to change that: No other location if found, not via postal code, city name, country name, in English or native language or any combination.
- The server where the app gets it's data from simply seems to be unavailable, or it has something to do with my VM network setup?

Fourdee, 2019-01-16:
Confirmed on Pinebook which was working the other day. Appears the Yahoo weather API has changed, or, temp offline.

MichaIng, 2019-01-21:
@Fourdee 
Is this implemented fix into LXDE? In case remove it during config step or provide a working replacement?

Fourdee, 2019-01-23:
@MichaIng 

Yep, appears the URL defined in the sourcecode for LXpanel is hard coded for Yahoo API, which is a 404:
https://git.lxde.org/gitweb/?p=lxde/lxpanel.git;a=blob;f=plugins/weather/yahooutil.c;h=8c0f82117fa598c0dcc0a92be73454b2b7dcd37e;hb=a5a4d42e7f18f413033d1802fb99c59d9d1a2e6f#l53
- http://query.yahooapis.com/

Will patch for removal.

Fourdee, 2019-01-23:
Completed, resolved for fresh installs. Existing users will need to remove the panel app using rightclick.

closed 2019-01-23 by Fourdee