MichaIng/DietPi #2417: DietPi-Config | RPi: Edit only gpu_mem_* entry that matches current RPi version

MichaIng, 2019-01-15:
Ref: https://dietpi.com/phpbb/viewtopic.php?p=16036#p16036

Currently, when switching the SDcard between RPi's with different RAM sizes, the memory split is always set for all of them, while you might want to have a lower value on lower RAM device and higher value on higher RAM device.
It is even possible to set a value that is outside of the allowed range for the lower RAM devices or even higher then the overall RAM size.

So we should only apply the value to the single `config.txt` entry that matches the current RPi version/RAM size.

Fourdee, 2019-01-16:
@MichaIng 

> It is even possible to set a value that is outside of the allowed range for the lower RAM devices or even higher then the overall RAM size.

Yep, was thinking of this the other day. 

Always set actual value to ```gpu_mem_1024```, cap the others during set. Scrape from  ```gpu_mem_1024```

Fourdee, 2019-01-16:
```
[ SUB1 ] DietPi-Set_Hardware > gpumemsplit (512)
[  OK  ] DietPi-Set_Hardware | Setting in /DietPi/config.txt adjusted: gpu_mem_1024=512
[  OK  ] DietPi-Set_Hardware | Desired setting in /DietPi/config.txt was already set: gpu_mem_512=448
[  OK  ] DietPi-Set_Hardware | Desired setting in /DietPi/config.txt was already set: gpu_mem_256=192
[  OK  ] gpumemsplit 512 | Completed
```

🈯️

closed 2019-01-16 by Fourdee

MichaIng, 2019-01-16:
@Fourdee 
Can't we use the `HW_MEMORY_SIZE` from `/DietPi/dietpi/.hw_model` to edit only the setting relevant for the current device?

If there is a unknown RPi model, so the above value will be `0`, then applying to all, respecting max values is great. Generally for these entries max allowed value check makes sense.