MichaIng/DietPi #2422: LMS Fails to recognize ALAC files

Sedim, 2019-01-17:
## Creating a bug report/issue:

#### Required Information:
- DietPi version	| `cat /DietPi/dietpi/.version`
- Distro version	| `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version	| `uname -a`
- SBC device		| `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
- Power supply used	| (EG: 5V 1A RAVpower)
- SDcard used		| (EG: SanDisk ultra)

#### Additional Information (if applicable):
- Software title	| (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?
- `dietpi-bugreport` ID

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

MichaIng, 2019-01-17:
@Sedim 
Thank you very much for your report.

We will investigate, perhaps some additional library is required for this.

MichaIng, 2020-07-01:
@Sedim 
Sorry for delay. You use SqueezeLite as player, right? I'm not 100% sure but as far as I understand, the audio file/container/codec is a player issue, not one of the LMS.

SqueezeLite can be compiled with and without ALAC support... damn and now I see I didn't compile it in...

Are your players Debian systems as well? Then I could redo the package for your system with enabled ALAC flag to test.

MichaIng, 2020-07-02:
Okay, in the LMS settings, file formats, Apple lossless, changing the decoder from "nativ" to "disabled" works, as long as the SqueezeBox/SqueezeLite supports ALAC decoding. It seems like with this file is sent to the player undecoded and our current SqueezeLite version successfully decodes ALAC files via ffmpeg. Not sure if LMS simply is not able to use ffmpeg for decoding, since it ran on the same system with ffmpeg libraries installed that SqueezeLite successfully used.

However since things are working from our side, I'll mark this issue as closed. Feel free to reopen if required.

closed 2020-07-02 by MichaIng