MichaIng/DietPi #2423: DietPi-Software | Monica

JonasDralle, 2019-01-17:
# Monica
Personal CRM - Monica helps you organize the social interactions with your loved ones.

#### Some formal software information:
- software title | Monica 
- Short description | Monica helps you organize the social interactions with your loved ones.
- GitHub URL | https://github.com/monicahq/monica
- External URL (if available) | https://monicahq.com/
- install documentation | https://github.com/monicahq/monica/blob/master/docs/installation/index.md

#### Are there similar/alternative software titles available with DietPi-Software?
None that I'm aware of

#### What makes your requested software better than the above solutions, if available?
not applicable

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?

- database: manual install of MariaDB (easy to do but it's 6+ steps, a script can do so too)
- PHP+dependencies: managed via composer
- enviroment variables need to be configured
- cronjobs need to be configured
- Apache webserver needs to be configured

#### Can you provide the installation steps that you would suggest DietPi-Software to do?

<details>
<summary>Installation Steps (click to open spoiler)</summary>

# Installation copied from [official documentation](https://github.com/monicahq/monica/blob/master/docs/installation/debian.md)

---

# Installing Monica on Debian

[image]

Monica can run on Debian Stretch.

## Prerequisites

1. Install the required packages:

```sh
sudo apt install apache2 mariadb-server php7.1 php7.1-cli php7.1-common \
    php7.1-json php7.1-opcache php7.1-mysql php7.1-mbstring php7.1-mcrypt \
    php7.1-zip php7.1-fpm php7.1-bcmath php7.1-intl php7.1-simplexml \
    php7.1-dom php7.1-curl php7.1-gd git curl
```

2. Install composer

Download and install the binary by following the [Command-line installation of composer](https://getcomposer.org/download/).

Move it to the bin directory.
```sh
sudo mv composer.phar /usr/local/bin/composer
```

## Installation steps

Once the softwares above are installed:

### 1. Clone the repository

You may install Monica by simply cloning the repository. Consider cloning the repository into any folder, example here in `/var/www/monica` directory:
```sh
sudo git clone https://github.com/monicahq/monica.git /var/www/monica
```

You should check out a tagged version of Monica since `master` branch may not always be stable.
Find the latest official version on the [release page](https://github.com/monicahq/monica/releases)
```sh
cd /var/www/monica
# Clone the desired version
sudo git checkout tags/v1.6.2
```

### 2. Setup the database

First make the database a bit more secure.
```sh
sudo mysql_secure_installation
```

Next log in with the root account to configure the database.
```sh
sudo mysql -uroot -p
```

Create a database called 'monica'.
```sql
CREATE DATABASE monica;
```

Create a user called 'monica' and its password 'strongpassword'.
```sql
CREATE USER 'monica'@'localhost' IDENTIFIED BY 'strongpassword';
```

We have to authorize the new user on the `monica` db so that he is allowed to change the database.
```sql
GRANT ALL ON monica.* TO 'monica'@'localhost';
```

And finally we apply the changes and exit the database.
```sql
FLUSH PRIVILEGES;
exit
```

### 3. Configure Monica

`cd /var/www/monica` then run these steps with `sudo`:

1. `cp .env.example .env` to create your own version of all the environment variables needed for the project to work.
1. Update `.env` to your specific needs. Don't forget to set `DB_USERNAME` and `DB_PASSWORD` with the settings used behind.
1. Run `composer install --no-interaction --no-suggest --no-dev --ignore-platform-reqs` to install all packages.
1. Run `php artisan key:generate` to generate an application key. This will set `APP_KEY` with the right value automatically.
1. Run `php artisan setup:production` to run the migrations, seed the database and symlink folders.
1. Optional: run `php artisan passport:install` to create the access tokens required for the API (Optional).

### 4. Configure cron job

Monica requires some background processes to continuously run. The list of things Monica does in the background is described [here](https://github.com/monicahq/monica/blob/master/app/Console/Kernel.php#L33).
Basically those crons are needed to send reminder emails and check if a new version is available.
To do this, setup a cron that runs every minute that triggers the following command `php artisan schedule:run`.

1. Open the crontab edit:
```sh
sudo crontab -e
```
2. Then, in the text editor window you just opened, copy the following:
```
* * * * * sudo -u www-data php /var/www/monica/artisan schedule:run
```

### 5. Configure Apache webserver

Give proper permissions to the project directory by running
```sh
sudo chown -R www-data:www-data /var/www/monica
```

Enable the rewrite module of the Apache webserver:
```sh
sudo a2enmod rewrite
```

Edit `/etc/apache2/sites-enabled/000-default.conf` file.

* Update `DocumentRoot` property to:
```
DocumentRoot /var/www/monica/public
```
* and add a new `Directory` directive:
```
<Directory /var/www/monica/public>
    Options Indexes FollowSymLinks
    AllowOverride All
    Require all granted
</Directory>
```

Finally restart Apache.
```sh
sudo systemctl restart apache2
```

### Final step

The final step is to have fun with your newly created instance, which should be up and running to `http://localhost`.

</details>

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.

I'm a developer and I know bash. Sadly when it comes to DietPi I'm a mere user and I've not looked at the source code that closely.
Right now it's unlikely that I can help (sorry)

MichaIng, 2019-01-17:
@JonasDralle 
Thank you very much for your request.

Sum-up partly from the install steps:
- Requires webserver, MariaDB and PHP, so usual web stack
  - I hope the `mcrypt` module is not really in use anymore, but a relict from old times, since it is totally outdated, deprecated and a security issue.  But with ownCloud/Nextcloud it's the same: Still in docs...
  - More importantly `php7.[123]-mcrypt` does not exist in Debian repos, only `php7.0-mcrypt`, so it's not available at all in Debian Buster. However, we still install it on Stretch with default PHP install.
- Requires PHP composer
- Setup cron job: `* * * * * sudo -u www-data php /var/www/monica/artisan schedule:run`
  - Hmm this would run every minute, but the linked background jobs are actually executed only hourly and daily: https://github.com/monicahq/monica/blob/master/app/Console/Kernel.php#L44-L50
  - [ ] No idea why they want to spam syslog and stress the system with the cron + PHP execution every minute then 🤔. Something we should directly address to the devs!
- On Apache, uses `.htaccess` with rewrite rules.
  - [ ] Check docks for Nginx/Lighttpd configs, otherwise we need to derive them from the `.htaccess` content.
- Git install suggested, but should be possible to skip and download directly the zip archive from GitHub. Git-based updater is very unlikely, since usually the MariaDB database needs to be migrated, which cannot be done by Git, but by PHP. @JonasDralle is there an internal updater available and do you know, if it depends on the installed Git client?

JonasDralle, 2019-01-18:
Those are a few very good points you raise

While very enthusiastic about the product I have not used it much yet. I don't think there's an self-updater.
According to the [update document](https://github.com/monicahq/monica/blob/master/docs/installation/update.md) it's git based. Database migartion doesn't seem to be taken into account.

wangkev, 2020-12-07:
Were there any updates here? Would definitely be interested!

exadeci, 2021-01-12:
@MichaIng Any chance to see it added anytime soon? Anything that we could help with?

MichaIng, 2021-05-27:
I added it to FeatHub, please vote to raise priority: https://feathub.com/MichaIng/DietPi/+230

MichaIng, 2021-09-24:
We decided to cleanup open issues a bit and close those which will obviously not addressed soon. The FeatHub request stays however open to allow votes and if anyone wants to start working on implementing this software, please feel free to to a start, and we'll help where we can.

Here you find instructions about how to implement a new software title to `dietpi-software`: https://github.com/MichaIng/DietPi/wiki/How-to-add-a-new-software-title

closed 2021-09-24 by MichaIng

JavaZauber, 2021-09-25:
Is FeatHub somehow broken?
I can't log in via firefox

> We're sorry, but something went wrong.
> If you are the application owner check the logs for more information.

Joulinar, 2021-09-25:
Yes SSL certificate of FeatHub is not valid anymore and usually your browser should ask to accept the risk.

JavaZauber, 2021-09-25:
It seems like GitHub Single-Sign-On is not willing to accept the risk

Joulinar, 2021-09-25:
Unfortunately FeatHub is not reliable anymore since a couple of weeks

MichaIng, 2021-09-25:
Reported already: https://github.com/feathub/feathub/issues/12

Nasty that it breaks signing in and also the badge.