MichaIng/DietPi #2426: NoIP cannot exec binary file

walker93, 2019-01-19:
## Creating a bug report/issue:

#### Required Information:
- DietPi version	| `cat /DietPi/dietpi/.version`
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=19
G_DIETPI_VERSION_RC=7
G_GITBRANCH=master
G_GITOWNER=Fourdee

- Distro version	| `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
strech

- Kernel version	| `uname -a`
Linux DietPi 4.9.0-8-amd64 #1 SMP Debian 4.9.130-2 (2018-10-27) x86_64 GNU/Linux

- SBC device		| `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
From the command: Generic Device (x86_64)  
Actually it is a LattePanda Alpha 864

- Power supply used	|  USB type-C 3A from 5V to 15V

- SDcard used		|  Sandisk Ultra 64GB UHS-I

#### Additional Information (if applicable):
- Software title	|  NoIP
- Was the software title installed freshly or updated/migrated?
Installed freshly from DietPi-config

- Can this issue be replicated on a fresh installation of DietPi?
That is what happened to me.
Installed Debian image and used PREP_SYSTEM script for DietPi conversion.

- `dietpi-bugreport` ID
ec78fbf4-1439-4dfa-b26c-75c6a3da229e

#### Steps to reproduce:

1. Install NoIP
2. Launch from DietPi-config or Terminal

#### Expected behaviour:

- NoIP should start asking for the configuration

#### Actual behaviour:

- This error appears: 
`/DietPi/dietpi/dietpi-config: line 4885: /usr/local/bin/noip2: cannot execute binary file: Exec format error`

MichaIng, 2019-01-19:
@walker93 
Many thanks for your report.

Found the reason:
- Instead of checking directly the machines architecture during install (`x86_64` in your case), we check for known x86_64 devices, which were (obviously when we touched code last time) native PC and virtual machine only.
- If the machine is no native PC or VM, we check for known ARMv8 devices next, and otherwise choose the `armhf` binary.
- Since generic device does not match anything else, it will install the `armhf` binary, which is in your case of course wrong.
- So we need to directly check the CPU architecture, which is also the usual way in all other installs. Seems No-IP install was not touched/reviewed for a long time 😅.

To fix this now:
```
wget https://dietpi.com/downloads/binaries/all/noip_x32_x64.zip
unzip noip_x32_x64.zip
rm noip_x32_x64.zip
mkdir -p /usr/local/bin
mv noip_binary /usr/local/bin/noip2
chmod +x /usr/local/bin/noip2
mkdir -p /usr/local/etc
```
Then retry using `dietpi-config` for setup.

_____________

Just found other minor quirks:
- The binary (so our systemd unit as well) requires `/usr/local/etc/no-ip2.conf` to exist. This is not the case on fresh install, so `dietpi-services` will throw an error until `noip2 -C` is called (e.g. from `dietpi-config`) to create a config file. No big issue but easy to fix by adding `ConditionPathExists=/usr/local/etc/no-ip2.conf` to the systemd unit.

More urgently:
- In my case, since I cleared `/usr/local` once, the binary `mv` failed first because `/usr/local/bin` did not exist.
- Then `noip2 -C` from `dietpi-config` failed as well,  because `/usr/local/etc` did not exist.
- We should pre-create both dirs to be safe.

walker93, 2019-01-19:
@MichaIng Thank you for the quick fix! I'm glad to say that it worked like a charm.
NoIP is configured and working now.

MichaIng, 2019-01-19:
@walker93 
Perfect, thanks for feedback 😃.

MichaIng, 2019-01-20:
Will mark as closed, since issue and fix are pretty clear, no specific testing required.

closed 2019-01-20 by MichaIng