MichaIng/DietPi #2427: allo sparky with all uxbridge don't work stable

seniorgod, 2019-01-19:
## Creating a bug report/issue:

#### Required Information:
- DietPi version    

G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=19
G_DIETPI_VERSION_RC=7
G_GITBRANCH=master
G_GITOWNER=Fourdee

- Distro version	: 9.6
- Kernel version :Linux DietPi 3.10.38 #22 SMP PREEMPT Fri Dec 22 10:45:41 IST 2017 armv7l GNU/Linux
- SBC device	: Sparky SBC (armv7l)
- Power supply used   EG: 5V 3A power supply from allo   
- SDcard used	32 GB SD card SANDISK ultra 

#### Additional Information (if applicable):
Software title: mpd
I bought a new allo usbridge, including  allo sparky and a usbridge. Because i had trouble with the first usbridge, i get a new one and set up the system yesterday. 
The whole system is "fresh", downloaded yesterday and flashed to a sd card. Currently i use sd cards, because it is quicker to exchange than the emmc card. 
I added mpd and shairport-sync. 
I also added Spotify-connect-web. After installing Spotify connect web the system hung. I was not able to play music via Spotify connect web. Never. When i use spotify-connect-web the systems sometimes reboot. I disabled and unmask Spotify-connect-web restart the system in order to be able to remove Spotify-connect-web.
 
MPD works quit well after rebooting the sparky. But sometimes it stop to work. In that case i'm not able to stream music via mpd. Typically a reboot helps. After a reboot it work at least one time correctly. Sometimes it work only one time, sometimes it work several times. 

Is it more a software or a hardware problem. I try also volumio, with more ore less the same result. 

- `dietpi-bugreport` ID  ff26ccb2-d25e-4dff-949f-b156991c4793

i found that in mpd.log

Jan 19 13:11 : exception: Failed to open "DietPi Alsa" [alsa]
Jan 19 13:11 : exception: nested: Failed to open ALSA device "hw:1,0": No such file or directory

and i found that in journalctl: 

Jan 19 12:11:10 DietPi kernel: irqvector:66, 0x42
Jan 19 12:11:10 DietPi kernel: hcd ep err ep_num:2, is_in:1
Jan 19 12:11:10 DietPi kernel: err_type :4
Jan 19 12:11:10 DietPi kernel: td->err_count:0
Jan 19 12:11:10 DietPi kernel: irqvector:66, 0x42
Jan 19 12:11:10 DietPi kernel: hcd ep err ep_num:2, is_in:1
Jan 19 12:11:10 DietPi kernel: err_type :4
Jan 19 12:11:10 DietPi kernel: td->err_count:1
Jan 19 12:11:10 DietPi kernel: usb 4-1.1: USB disconnect, device number 3
Jan 19 12:11:10 DietPi kernel: aotg_hcd aotg_hcd.1: <EP DISABLE> ep0 index 0 from ep [in]
Jan 19 12:11:10 DietPi kernel: aotg_hub_endpoint_disable
Jan 19 12:11:10 DietPi kernel: aotg_hub_endpoint_disable, ep2 dma buf free
Jan 19 12:11:10 DietPi kernel: aotg_hcd aotg_hcd.1: <EP DISABLE> ep4 index 2 from ep [in]
Jan 19 12:11:25 DietPi kernel: usb 4-1.1: new high-speed USB device number 4 using aotg_hcd
Jan 19 12:11:25 DietPi kernel: aotg_hcd aotg_hcd.1: device addr : 0x00000082
Jan 19 12:11:25 DietPi kernel: aotg_hcd aotg_hcd.1: <EP DISABLE> ep0 index 0 from ep [in]
Jan 19 12:11:25 DietPi kernel: aotg_hcd aotg_hcd.1: device addr : 0x00000082
Jan 19 12:11:25 DietPi kernel: usb 4-1.1: New USB device found, idVendor=2772, idProduct=0221
Jan 19 12:11:25 DietPi kernel: usb 4-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Jan 19 12:11:25 DietPi kernel: usb 4-1.1: Product: Project RS USB Audio 2.0
Jan 19 12:11:25 DietPi kernel: usb 4-1.1: Manufacturer: Project RS 
Jan 19 12:11:25 DietPi kernel: usb 4-1.1: SerialNumber: 0000
Jan 19 12:11:27 DietPi kernel: iso_packets:0, bInterval:1, urb_interval:1, reg_con:0x4
Jan 19 12:11:27 DietPi kernel: iso_packets:0, bInterval:4, urb_interval:8, reg_con:0x4
Jan 19 12:11:47 DietPi kernel: irqvector:67, 0x43
Jan 19 12:11:47 DietPi kernel: hcd ep err ep_num:3, is_in:1
Jan 19 12:11:47 DietPi kernel: err_type :4
Jan 19 12:11:47 DietPi kernel: td->err_count:0
Jan 19 12:11:47 DietPi kernel: irqvector:67, 0x43
Jan 19 12:11:47 DietPi kernel: hcd ep err ep_num:3, is_in:1
Jan 19 12:11:47 DietPi kernel: err_type :4
Jan 19 12:11:47 DietPi kernel: td->err_count:1
Jan 19 12:11:47 DietPi kernel: irqvector:67, 0x43
[... 168 lines cut ...]
Jan 19 12:20:42 DietPi kernel: [<c037acd4>] (bus_remove_device+0xd4/0x124) from [<c0378804>] (device_del+0xf0/0x174)
Jan 19 12:20:42 DietPi kernel: [<c0378804>] (device_del+0xf0/0x174) from [<c03ffb9c>] (usb_disable_device+0xac/0x250)
Jan 19 12:20:42 DietPi kernel: [<c03ffb9c>] (usb_disable_device+0xac/0x250) from [<c03f6c68>] (usb_disconnect+0x9c/0x1b0)
Jan 19 12:20:42 DietPi kernel: [<c03f6c68>] (usb_disconnect+0x9c/0x1b0) from [<c03f8b44>] (hub_thread+0x6f0/0x15a0)
Jan 19 12:20:42 DietPi kernel: [<c03f8b44>] (hub_thread+0x6f0/0x15a0) from [<c004d394>] (kthread+0xa4/0xb0)
Jan 19 12:20:42 DietPi kernel: [<c004d394>] (kthread+0xa4/0xb0) from [<c000f3d8>] (ret_from_fork+0x14/0x3c)
Jan 19 12:21:47 DietPi systemd[1]: Starting Cleanup of Temporary Directories...
Jan 19 12:21:47 DietPi systemd[1]: Started Cleanup of Temporary Directories.
Jan 19 12:22:42 DietPi kernel: INFO: task khubd:56 blocked for more than 120 seconds.
Jan 19 12:22:42 DietPi kernel: "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
Jan 19 12:22:42 DietPi kernel: khubd           D c07bfe48     0    56      2 0x00000000
Jan 19 12:22:42 DietPi kernel: [<c07bfe48>] (__schedule+0x390/0x7cc) from [<c03fe17c>] (usb_kill_urb+0x88/0xd4)
Jan 19 12:22:42 DietPi kernel: [<c03fe17c>] (usb_kill_urb+0x88/0xd4) from [<c03fd214>] (usb_hcd_flush_endpoint+0xa8/0x174)
Jan 19 12:22:42 DietPi kernel: [<c03fd214>] (usb_hcd_flush_endpoint+0xa8/0x174) from [<c03ffa68>] (usb_disable_endpoint+0x4c/0x80)
Jan 19 12:22:42 DietPi kernel: [<c03ffa68>] (usb_disable_endpoint+0x4c/0x80) from [<c03ffae0>] (usb_disable_interface+0x44/0x54)
Jan 19 12:22:42 DietPi kernel: [<c03ffae0>] (usb_disable_interface+0x44/0x54) from [<c0401f58>] (usb_unbind_interface+0x16c/0x170)
Jan 19 12:22:42 DietPi kernel: [<c0401f58>] (usb_unbind_interface+0x16c/0x170) from [<c037b230>] (__device_release_driver+0x70/0xc8)
Jan 19 12:22:42 DietPi kernel: [<c037b230>] (__device_release_driver+0x70/0xc8) from [<c037b2a4>] (device_release_driver+0x1c/0x28)
Jan 19 12:22:42 DietPi kernel: [<c037b2a4>] (device_release_driver+0x1c/0x28) from [<c037acd4>] (bus_remove_device+0xd4/0x124)
Jan 19 12:22:42 DietPi kernel: [<c037acd4>] (bus_remove_device+0xd4/0x124) from [<c0378804>] (device_del+0xf0/0x174)
Jan 19 12:22:42 DietPi kernel: [<c0378804>] (device_del+0xf0/0x174) from [<c03ffb9c>] (usb_disable_device+0xac/0x250)
Jan 19 12:22:42 DietPi kernel: [<c03ffb9c>] (usb_disable_device+0xac/0x250) from [<c03f6c68>] (usb_disconnect+0x9c/0x1b0)
Jan 19 12:22:42 DietPi kernel: [<c03f6c68>] (usb_disconnect+0x9c/0x1b0) from [<c03f8b44>] (hub_thread+0x6f0/0x15a0)
Jan 19 12:22:42 DietPi kernel: [<c03f8b44>] (hub_thread+0x6f0/0x15a0) from [<c004d394>] (kthread+0xa4/0xb0)
Jan 19 12:22:42 DietPi kernel: [<c004d394>] (kthread+0xa4/0xb0) from [<c000f3d8>] (ret_from_fork+0x14/0x3c)

seniorgod, 2019-01-19:
One more thing.....

Because the system was not stable with DietPi. I also try volumio. There is more or less the same result. Sometimes it work especially after reboot. Sometimes it doesn't work. 
What i do not understand: when it would be a hardware defect, it would not work at all. 
If it would be a software defect, i wonder why i don't find more problems from other usbridge users?

MichaIng, 2019-01-19:
@seniorgod 
Many thanks for your report.

I marked for investigation. Can't do it myself due to lack of device and audio system to test.

Indeed interesting that it works first and fails to connect to the audio device (obviously). Perhaps some voltage/power drop or physical connection issue?

At least the USB get's disconnects:
```
kernel: usb 4-1.1: USB disconnect, device number 3
...
kernel: usb 4-1.1: New USB device found, idVendor=2772, idProduct=0221
...
kernel: [] (usb_kill_urb+0x88/0xd4) from [] (usb_hcd_flush_endpoint+0xa8/0x174)
# Followed by some more kill/disable/disconnect messages
```

Did you try a different cable for somehow bring the wire in a fix position, so there will be surely no motion in the USB connection?

seniorgod, 2019-01-19:
Hm, there is surely no motion...... 
And as the last hifi freak under 50 years i spend 50€ for an USB Cable to connect the DAC. The same DAC with the same cable works flawlessly with my raspberry pi. 
Until now i used a raspberry pi as mpd server. I connected my Project PreAmp/Dac via usb to raspberry. I wrote an u-dev rule to observe if the preamp/dac is shut on o shut off. 
If the dac is shut off the dev rule stops mpd from playing. 
Is the dac is shut on, i start a script via u-dev which load a playlists and start playing. That works great with raspberry pi since years. But not reliable with sparky. 

What i wonder is if:
...
kernel: usb 4-1.1: USB disconnect, device number 3
...
point to the project or to usbridge from allo? 
Do you think there is a physical mechanical or electrical problem?

MichaIng, 2019-01-19:
@seniorgod 
Ah, it's the Project RS audio:
> Jan 19 12:11:10 DietPi kernel: usb 4-1.1: USB disconnect, device number 3
> Jan 19 12:11:10 DietPi kernel: aotg_hcd aotg_hcd.1: ep0 index 0 from ep [in]
> Jan 19 12:11:10 DietPi kernel: aotg_hub_endpoint_disable
> Jan 19 12:11:10 DietPi kernel: aotg_hub_endpoint_disable, ep2 dma buf free
> Jan 19 12:11:10 DietPi kernel: aotg_hcd aotg_hcd.1: ep4 index 2 from ep [in]
> Jan 19 12:11:25 DietPi kernel: usb 4-1.1: new high-speed USB device number 4 using aotg_hcd
> Jan 19 12:11:25 DietPi kernel: aotg_hcd aotg_hcd.1: device addr : 0x00000082
> Jan 19 12:11:25 DietPi kernel: aotg_hcd aotg_hcd.1: ep0 index 0 from ep [in]
> Jan 19 12:11:25 DietPi kernel: aotg_hcd aotg_hcd.1: device addr : 0x00000082
> Jan 19 12:11:25 DietPi kernel: usb 4-1.1: New USB device found, idVendor=2772, idProduct=0221
> Jan 19 12:11:25 DietPi kernel: usb 4-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> Jan 19 12:11:25 DietPi kernel: usb 4-1.1: Product: **Project RS USB Audio 2.0**
> Jan 19 12:11:25 DietPi kernel: usb 4-1.1: Manufacturer: **Project RS**
> Jan 19 12:11:25 DietPi kernel: usb 4-1.1: SerialNumber: 0000
___________
> Do you think there is a physical mechanical or electrical problem?

To be true, I have no idea. It was just a guess in the blue based on the relatively quick disconnect/connect/disable logs, which is simple to verify.
Clearly not my profession, since I don't use any media system with my DietPi.

@Fourdee 
Might you have a look into this?

seniorgod, 2019-01-19:
** ok, i do some more investigations: I shut on and off my project an log the result of journalctl 

the system is started
i start on my Preamp/DAC (which is always connected to sparky usbridge via USB cable)
the system find a new usb device 
my u-dev rule fire and call a script which clerk the mpd playlist, load a playlist with radios stations and plays the first entry 
everything is fine  - i hear music 

this ist the verbose output of journalctl during the above described steps
**

Sat 2019-01-19 19:10:02.429490 CET [s=8c1ccadf9de24df9982f86c021e95431;i=3c2;b=96074b8be7c04e49aa7a7269eaf84336;m=1ff8d0b6;t=57fd389550032;x=e24f286d6a72629a]
    _KERNEL_SUBSYSTEM=usb
    _TRANSPORT=kernel
    PRIORITY=6
    SYSLOG_FACILITY=0
    SYSLOG_IDENTIFIER=kernel
    _BOOT_ID=96074b8be7c04e49aa7a7269eaf84336
    _MACHINE_ID=a4db9ecd9b5041b3a7cc6334db4d0c18
    _HOSTNAME=DietPi
    _KERNEL_DEVICE=+usb:3-1.1
    _SOURCE_MONOTONIC_TIMESTAMP=536405026
    MESSAGE=usb 3-1.1: new high-speed USB device number 3 using aotg_hcd
Sat 2019-01-19 19:10:02.430010 CET [s=8c1ccadf9de24df9982f86c021e95431;i=3c3;b=96074b8be7c04e49aa7a7269eaf84336;m=1ff8d2bb;t=57fd38955023a;x=91f95a8a9ff1045f]
    _TRANSPORT=kernel
    PRIORITY=6
    SYSLOG_FACILITY=0
    SYSLOG_IDENTIFIER=kernel
    _BOOT_ID=96074b8be7c04e49aa7a7269eaf84336
    _MACHINE_ID=a4db9ecd9b5041b3a7cc6334db4d0c18
    _HOSTNAME=DietPi
    _KERNEL_SUBSYSTEM=platform
    _KERNEL_DEVICE=+platform:aotg_hcd.1
    _UDEV_SYSNAME=aotg_hcd.1
    _SOURCE_MONOTONIC_TIMESTAMP=536405048
    MESSAGE=aotg_hcd aotg_hcd.1: device addr : 0x00000082
Sat 2019-01-19 19:10:02.509478 CET [s=8c1ccadf9de24df9982f86c021e95431;i=3c4;b=96074b8be7c04e49aa7a7269eaf84336;m=1ffa0929;t=57fd3895638a6;x=e3351611dae4d04d]
    _TRANSPORT=kernel
    PRIORITY=6
    SYSLOG_FACILITY=0
    SYSLOG_IDENTIFIER=kernel
    _BOOT_ID=96074b8be7c04e49aa7a7269eaf84336
    _MACHINE_ID=a4db9ecd9b5041b3a7cc6334db4d0c18
    _HOSTNAME=DietPi
    _KERNEL_SUBSYSTEM=platform
    _KERNEL_DEVICE=+platform:aotg_hcd.1
    _UDEV_SYSNAME=aotg_hcd.1
    MESSAGE=aotg_hcd aotg_hcd.1: <EP DISABLE> ep0 index 0 from ep [in]
    _SOURCE_MONOTONIC_TIMESTAMP=536485081
Sat 2019-01-19 19:10:02.529584 CET [s=8c1ccadf9de24df9982f86c021e95431;i=3c5;b=96074b8be7c04e49aa7a7269eaf84336;m=1ffa57bc;t=57fd389568730;x=36277060d0ac423f]
    _TRANSPORT=kernel
    PRIORITY=6
    SYSLOG_FACILITY=0
    SYSLOG_IDENTIFIER=kernel
    _BOOT_ID=96074b8be7c04e49aa7a7269eaf84336
    _MACHINE_ID=a4db9ecd9b5041b3a7cc6334db4d0c18
    _HOSTNAME=DietPi
    _KERNEL_SUBSYSTEM=platform
    _KERNEL_DEVICE=+platform:aotg_hcd.1
    _UDEV_SYSNAME=aotg_hcd.1
    MESSAGE=aotg_hcd aotg_hcd.1: device addr : 0x00000082
    _SOURCE_MONOTONIC_TIMESTAMP=536505002
Sat 2019-01-19 19:10:02.531328 CET [s=8c1ccadf9de24df9982f86c021e95431;i=3c6;b=96074b8be7c04e49aa7a7269eaf84336;m=1ffa5e81;t=57fd389568e00;x=802ac149124bd19e]
    _KERNEL_SUBSYSTEM=usb
    _TRANSPORT=kernel
    PRIORITY=6
    SYSLOG_FACILITY=0
    SYSLOG_IDENTIFIER=kernel
    _BOOT_ID=96074b8be7c04e49aa7a7269eaf84336
    _MACHINE_ID=a4db9ecd9b5041b3a7cc6334db4d0c18
[... 1673 lines cut ...]
    _MACHINE_ID=a4db9ecd9b5041b3a7cc6334db4d0c18
    _HOSTNAME=DietPi
    PRIORITY=4
    MESSAGE=[<c03f6c68>] (usb_disconnect+0x9c/0x1b0) from [<c03f8b44>] (hub_thread+0x6f0/0x15a0)
    _SOURCE_MONOTONIC_TIMESTAMP=1201177519
Sat 2019-01-19 19:21:07.216386 CET [s=8c1ccadf9de24df9982f86c021e95431;i=462;b=96074b8be7c04e49aa7a7269eaf84336;m=4798a884;t=57fd3b0f4d802;x=5201103a327c5af7]
    _TRANSPORT=kernel
    SYSLOG_FACILITY=0
    SYSLOG_IDENTIFIER=kernel
    _BOOT_ID=96074b8be7c04e49aa7a7269eaf84336
    _MACHINE_ID=a4db9ecd9b5041b3a7cc6334db4d0c18
    _HOSTNAME=DietPi
    PRIORITY=4
    MESSAGE=[<c03f8b44>] (hub_thread+0x6f0/0x15a0) from [<c004d394>] (kthread+0xa4/0xb0)
    _SOURCE_MONOTONIC_TIMESTAMP=1201190023
Sat 2019-01-19 19:21:07.216612 CET [s=8c1ccadf9de24df9982f86c021e95431;i=463;b=96074b8be7c04e49aa7a7269eaf84336;m=4798a964;t=57fd3b0f4d8e4;x=714b0e23bf2320e6]
    _TRANSPORT=kernel
    SYSLOG_FACILITY=0
    SYSLOG_IDENTIFIER=kernel
    _BOOT_ID=96074b8be7c04e49aa7a7269eaf84336
    _MACHINE_ID=a4db9ecd9b5041b3a7cc6334db4d0c18
    _HOSTNAME=DietPi
    PRIORITY=4
    MESSAGE=[<c004d394>] (kthread+0xa4/0xb0) from [<c000f3d8>] (ret_from_fork+0x14/0x3c)
    _SOURCE_MONOTONIC_TIMESTAMP=1201201900

Fourdee, 2019-01-20:
@seniorgod 

Thanks for the report 👍 

Appears to be a hardware specific issue, possibily system instability. 

Might be worth trying the following, to see if same results:
- Different PSU
- Different USB DAC

Failing that, if problems persists, I would get in touch with Allo and reference in this ticket.

seniorgod, 2019-01-20:
@Fourdee 

thanks for replying 

i investigated further.
I use the sparky together with allo usbridge which provides better "usb sound"
So i thought thats a problem can occur at that device. The usbridge is an separate board, which is connected via the 50PIN connector to the sparky. 
Currently feed power by the usbridge to the sparky. 
I change the connection to the DAC from the usb port of the usbridge to the usb port of the sparky. It is the same result. 
I also experimented with mpc to stop playing music to the dac before i disconnect the DAC (by shutting off) .

The result is the same regardless if i use the usb port of the usbbridge or the usb port of the sparky:

seniorgod, 2019-01-20:
Dear all,

i tested different configurations today 

I detached the usbridge from sparky and used a different power supply only for the sparky. 
The result is the same. 
When i disconnect the DAC without playing music, everything is flawless. When is shut on the DAC again, he is found by the sparky. 
When i disconnect the DAC while i play music, i face the behaviour described above. The connection is broken and lost until restart of sparky.   

i set up my raspberry Pi with dietpi and do the tests again 
With raspberry and Dietpi all went fine. 
Any ideas? 
Could it be a problem of the driver for the usb hub on sparky?

# raspberry pi with dietpi 

 ────────────────────────────────────────────────
 DietPi | Sun 20/01/19 - 19:09
 ────────────────────────────────────────────────
 v6.19.7 | RPi 3 Model B (armv7l)
 ────────────────────────────────────────────────

## raspberry pi dac shut on 
udev script find the DAC, load a playlist and play music
flawless 

### udevadm output

KERNEL[412.962017] add      /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3 (usb)
KERNEL[412.964940] add      /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3:1.0 (usb)
KERNEL[412.966265] add      /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3:1.1 (usb)
KERNEL[412.967222] add      /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3:1.2 (usb)
UDEV  [412.971395] add      /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3 (usb)
UDEV  [412.978661] add      /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3:1.1 (usb)
UDEV  [412.981083] add      /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3:1.2 (usb)
KERNEL[412.988179] add      /module/snd (module)
UDEV  [412.990117] add      /module/snd (module)
KERNEL[412.993002] add      /module/snd_timer (module)
KERNEL[412.993507] add      /devices/virtual/sound/timer (sound)
UDEV  [412.995027] add      /module/snd_timer (module)
UDEV  [412.996925] add      /devices/virtual/sound/timer (sound)
KERNEL[413.004434] add      /module/snd_pcm (module)
UDEV  [413.006027] add      /module/snd_pcm (module)
KERNEL[413.007535] add      /module/snd_seq_device (module)
KERNEL[413.007908] add      /bus/snd_seq (bus)
UDEV  [413.009458] add      /module/snd_seq_device (module)
UDEV  [413.009780] add      /bus/snd_seq (bus)
KERNEL[413.012292] add      /module/snd_rawmidi (module)
UDEV  [413.013924] add      /module/snd_rawmidi (module)
KERNEL[413.016362] add      /module/snd_usbmidi_lib (module)
UDEV  [413.017958] add      /module/snd_usbmidi_lib (module)
KERNEL[413.019635] add      /module/snd_hwdep (module)
UDEV  [413.021249] add      /module/snd_hwdep (module)
KERNEL[413.036654] add      /module/snd_usb_audio (module)
UDEV  [413.038380] add      /module/snd_usb_audio (module)
KERNEL[413.091894] add      /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3:1.0/sound/card1 (sound)
KERNEL[413.092380] add      /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3:1.0/sound/card1/controlC1 (sound)
KERNEL[413.092783] add      /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3:1.0/sound/card1/pcmC1D0p (sound)
UDEV  [413.095430] add      /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3:1.0 (usb)
KERNEL[413.095590] add      /bus/usb/drivers/snd-usb-audio (drivers)
UDEV  [413.095714] add      /bus/usb/drivers/snd-usb-audio (drivers)
UDEV  [413.097135] add      /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3:1.0/sound/card1 (sound)
UDEV  [413.100373] add      /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3:1.0/sound/card1/pcmC1D0p (sound)
KERNEL[413.104142] change   /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3:1.0/sound/card1 (sound)
UDEV  [413.180728] add      /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3:1.0/sound/card1/controlC1 (sound)
UDEV  [413.187956] change   /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3:1.0/sound/card1 (sound)

### journalctl output 

Jan 20 19:02:36 DietPi kernel: usb 1-1.3: new high-speed USB device number 8 using dwc_otg
[... 176 lines cut ...]
Jan 20 19:17:01 DietPi CRON[1896]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)
Jan 20 19:17:02 DietPi CRON[1892]: pam_unix(cron:session): session closed for user root
Jan 20 19:17:12 DietPi kernel: usb 1-1.3: USB disconnect, device number 11
Jan 20 19:17:12 DietPi systemd[1]: sound.target: Unit not needed anymore. Stopping.
Jan 20 19:17:12 DietPi systemd[1]: Stopped target Sound Card.
Jan 20 19:17:34 DietPi kernel: usb 1-1.3: new high-speed USB device number 12 using dwc_otg
Jan 20 19:17:34 DietPi kernel: usb 1-1.3: New USB device found, idVendor=2772, idProduct=0221
Jan 20 19:17:34 DietPi kernel: usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Jan 20 19:17:34 DietPi kernel: usb 1-1.3: Product: Project RS USB Audio 2.0
Jan 20 19:17:34 DietPi kernel: usb 1-1.3: Manufacturer: Project RS 
Jan 20 19:17:34 DietPi kernel: usb 1-1.3: SerialNumber: 0000
Jan 20 19:17:34 DietPi systemd[1]: Reached target Sound Card.
Jan 20 19:17:44 DietPi kernel: usb 1-1.3: USB disconnect, device number 12
Jan 20 19:17:44 DietPi systemd[1]: sound.target: Unit not needed anymore. Stopping.
Jan 20 19:17:44 DietPi systemd[1]: Stopped target Sound Card.
Jan 20 19:18:26 DietPi kernel: usb 1-1.3: new high-speed USB device number 13 using dwc_otg
Jan 20 19:18:26 DietPi kernel: usb 1-1.3: New USB device found, idVendor=2772, idProduct=0221
Jan 20 19:18:26 DietPi kernel: usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Jan 20 19:18:26 DietPi kernel: usb 1-1.3: Product: Project RS USB Audio 2.0
Jan 20 19:18:26 DietPi kernel: usb 1-1.3: Manufacturer: Project RS 
Jan 20 19:18:26 DietPi kernel: usb 1-1.3: SerialNumber: 0000
Jan 20 19:18:26 DietPi systemd[1]: Reached target Sound Card.
Jan 20 19:19:58 DietPi kernel: usb 1-1.3: USB disconnect, device number 13
Jan 20 19:19:58 DietPi systemd[1]: sound.target: Unit not needed anymore. Stopping.
Jan 20 19:19:58 DietPi systemd[1]: Stopped target Sound Card.

Fourdee, 2019-01-21:
@seniorgod 

In ```dietpi-config``` have you tried setting ```usb-dac-1.1``` for the soundcard? This forces USB 1.1 mode for stability.

After applying, reboot and see if it improves.

Failing the above, I suspect hardware related issue as the issue occurs on volumio also.

seniorgod, 2019-01-21:
Hello Fourdee,

thanks for you help. 
First: The change to usb-dac-1.1 only result in a quit room. I hear nothing. Maybe i can't play music because i address the DAC in MPD via "hw:1:0". I don't know.

Meanwhile the allo support send a mail. It seems there is a "trick" in the USB driver or chip or whatever else. 
In the file 

/DietPi/uEnv.txt 

i had to change a variable from 
   aotg.urb_fix = 1
   to 
   aotg.urb_fix = 0 

and reboot the system.
Afterward it works. 

Maybe you have to adapt that value in the image, which is offered for download?

best regards and many thanks

MichaIng, 2019-01-21:
@seniorgod 
Ah this is great. AFAIK we do not touch `uEnv.txt` on Sparky SBC, so yeah this is something we should do then, when selecting a USB DAC at least.

Not sure what the setting does and if there are any downsides to not apply this by default on all devices, needs some research. I reopen the topic to not forget about this.

MichaIng, 2019-03-04:
Lol forgotten. Added to v6.22 milestone. Possibly related: https://github.com/MichaIng/DietPi/issues/2610

closed 2019-03-05 by MichaIng