MichaIng/DietPi #2429: Pinebook | initramfs-tools fails on APT upgrade

MichaIng, 2019-01-20:
Ref: https://dietpi.com/phpbb/viewtopic.php?f=9&t=5422&start=10#p16131

Suspicious log part:
```
update-initramfs: Generating /boot/initrd.img-4.20.0-sunxi64
/etc/initramfs/post-update.d//99-uboot: 3: .: Can't open /etc/armbian-release
run-parts: /etc/initramfs/post-update.d//99-uboot exited with return code 2
```

There seems to be a post-update script for the initramfs placed on the base image, which obviously is ARMbian?

Fourdee, 2019-01-21:
@MichaIng 

Yep, already fixed, however if the v6.20 patch fails, users will need to manually apply the ```/etc/armbian-release``` file before updating.
```
cat << _EOF_ > /etc/armbian-release
# PLEASE DO NOT EDIT THIS FILE
BOARD=pinebook-a64
BOARD_NAME="Pinebook A64"
BOARDFAMILY=sun50iw1
VERSION=5.67
LINUXFAMILY=sunxi64
BRANCH=next
ARCH=arm64
IMAGE_TYPE=stable
BOARD_TYPE=conf
INITRD_ARCH=arm64
KERNEL_IMAGE_TYPE=Image
_EOF_
```

>  I think we should find the root of this and remove the APT trigger that was obviously placed by ARMbian:

I'd prefer we leave it with the file left in place. Prevent any further issues down the road should that file and/or its contents be required in any form.

MichaIng, 2019-01-21:
@Fourdee 
> I'd prefer we leave it with the file left in place. Prevent any further issues down the road should that file and/or its contents be required in any form.

Good point. However perhaps we could additionally remove the offending APT drop-in trigger, if it does nothing useful. Generally good to know/learn about the custom ARMbian changes.

Fourdee, 2019-01-28:
Will update image in following ticket:
https://github.com/Fourdee/DietPi/issues/2026

closed 2019-01-28 by Fourdee