MichaIng/DietPi #2430: Add Squashfs kernel support

ThomasLeister, 2019-01-20:
#### Is your feature request related to a problem? Please describe:
Currently I'm using DietPi with my Rock64 SBC. Since it's quite powerful, I'd like to run multiple applications in linux containers on it. I'd like to use LXD. LXD is available as Canonical Snap package (this is the recommended way to install LXD). So I've tried installing the snapd package, first, but I'm facing this error message when i'm trying to install any snap package:

(snap tries to install its "core" snap package at first and fails:)

```
root@rock64:~# snap install lxd
error: cannot perform the following tasks:
- Mount snap "core" (6133) ([start snap-core-6133.mount] failed with exit status 1: Job for snap-core-6133.mount failed.
See "systemctl status snap-core-6133.mount" and "journalctl -xe" for details.
)
```

```
Jan 20 17:05:39 rock64 /usr/lib/snapd/snapd[11010]: taskrunner.go:353: DEBUG: Running task 181 on Do: Download snap "core" (6133) from channel "stable"
Jan 20 17:06:20 rock64 kernel: [UFW BLOCK] IN=eth0 OUT= MAC=01:00:5e:00:00:01:44:4e:6d:71:c9:88:08:00 SRC=192.168.178.1 DST=224.0.0.1 LEN=36 TOS=0x00 PREC=0xC0 TTL=1 ID=39670 DF PROTO=2 
Jan 20 17:06:22 rock64 /usr/lib/snapd/snapd[11010]: taskrunner.go:353: DEBUG: Running task 182 on Do: Fetch and check assertions for snap "core" (6133)
Jan 20 17:06:29 rock64 /usr/lib/snapd/snapd[11010]: taskrunner.go:353: DEBUG: Running task 183 on Do: Mount snap "core" (6133)
Jan 20 17:06:29 rock64 systemd[1]: Reloading.
Jan 20 17:06:29 rock64 systemd-fstab-generator[11296]: Ignoring "nofail" for root device
Jan 20 17:06:29 rock64 systemd[1]: Mounting Mount unit for core...
-- Subject: Unit snap-core-6133.mount has begun start-up
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- Unit snap-core-6133.mount has begun starting up.
Jan 20 17:06:29 rock64 systemd[1]: snap-core-6133.mount: Mount process exited, code=exited status=32
Jan 20 17:06:29 rock64 systemd[1]: Failed to mount Mount unit for core.
-- Subject: Unit snap-core-6133.mount has failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- Unit snap-core-6133.mount has failed.
-- 
-- The result is failed.
Jan 20 17:06:29 rock64 systemd[1]: snap-core-6133.mount: Unit entered failed state.
Jan 20 17:06:29 rock64 /usr/lib/snapd/snapd[11010]: task.go:303: DEBUG: 2019-01-20T17:06:29Z ERROR [start snap-core-6133.mount] failed with exit status 1: Job for snap-core-6133.mount failed.
```

#### Describe the solution you'd like:
After a bit of research I figured out that squashfs file system support is missing. Therefore the mount procedure fails. (I verified that by comparing my Rock64 system to another system where Snap was working. ```cat /proc/filesystems``` does not contain "squashfs" on DietPi, while it does on my other system.).

My request: **Please add squashfs file system support to kernel to enable use of Snapd.**

#### Describe alternatives you've considered:
I've looked for a DKMS module for squashfs but couldn't find any suitable.

MichaIng, 2019-01-20:
@ThomasLeister 
Many thanks for your request.

Could you try: `G_AGI squashfs-tools`

ThomasLeister, 2019-01-20:
I've installed squashfs-tools before but it didn't give me squashfs support. I guess this is just a package for some userland tools, but not for the file system support itself.

MichaIng, 2019-01-20:
@ThomasLeister 
Could you try:
`modprobe squashfs`

If it fails:
`modinfo squashfs`

Otherwise perhaps indeed it is not included within the default Rock64 kernel 🤔. Need to check how to add it then.

Halytskyi, 2019-01-21:
I can confirm that snapd works well on Rock64 with the latest (for now) kernel: [*4.4.154-1128-rockchip-ayufan-g61b4b1151f9a*](https://github.com/ayufan-rock64/linux-kernel/releases/tag/4.4.154-1128-rockchip-ayufan)
```
root@rock64:~# systemctl status snapd | grep Active
   Active: active (running) since Mon 2019-01-21 03:01:51 UTC; 4min 36s ago
root@rock64:~# uname -a
Linux rock64 4.4.154-1128-rockchip-ayufan-g61b4b1151f9a #1 SMP Tue Dec 25 14:31:31 UTC 2018 aarch64 aarch64 aarch64 GNU/Linux
root@rock64:~# systemctl status snapd | grep Active
   Active: active (running) since Mon 2019-01-21 03:01:51 UTC; 4min 46s ago
root@rock64:~# snap info lxd | grep 3.9
  stable:        3.9         (9930) 48MB -
  candidate:     3.9         (9930) 48MB -
```

MichaIng, 2019-01-21:
@Halytskyi 
Thanks for the info. Is this kernel already packed with the `linux-rock64` APT package?

Halytskyi, 2019-01-21:
> @Halytskyi
> Thanks for the info. Is this kernel already packed with the `linux-rock64` APT package?

yes, you can install it by command:
```
apt-get install linux-image-4.4.154-1128-rockchip-ayufan-g61b4b1151f9a
```

MichaIng, 2019-01-21:
@Halytskyi 
Okay great!

@ThomasLeister 
Please try `apt-get dist-upgrade` first, to stay with `linux-rock64` marked as manual only (and thus automatically switch to newer kernels, when available).
If the mentioned version is not installed by this, go with above and see if this enables `squashfs` support as expected.

MichaIng, 2019-01-23:
@ThomasLeister 
Not sure, obviously the kernel package install does not trigger a bootloader update. Did you try `apt update && apt-get dist-upgrade`, to check if the new version is already pulled with the meta package? With this for sure the bootloader would be updated as well.

Sadly I currently can't find the correct method how the update uboot the load/match the new kernel. Perhaps it's as we an ATP package upgrade required or there is a general method the rebuild uboot with the new kernel, similar to `update-grub` for grub bootloader.

You could try to identify the uboot package naming scheme:
`dpkg -l | grep uboot`

Fourdee, 2019-01-26:
I've updated our image which includes support for ```squashfs```. However, cant release image until v6.20 release, ETA day/two.

ThomasLeister, 2019-01-27:
Hi again, sorry for the delay. ```update && apt-get dist-upgrade``` did not work - there was no new kernel version available at this time. 

I'll try again this evening and maybe do some investigation how to (reliably) update the boot loader configs :-)

ThomasLeister, 2019-01-27:
Oh, just read your comment, @Fourdee . Thank you very much! It's not urgent in my case. I can wait :-)

MichaIng, 2019-01-27:
@ThomasLeister 
Okay, so the meta package does not yet pull the new kernel. And I guess if you install it (as you tried), you need to update the bootloader as well, either via separate APT package (not sure how they are called on Pine64 SBCs) or some command.

However maybe it's easier to wait for kernel package integration or our new image then.

Fourdee, 2019-01-28:
@ThomasLeister 

I'll move this to following ticket: https://github.com/Fourdee/DietPi/issues/2026#issuecomment-458294964

V6.20 (+ updated image) aiming to release today. Please keep an eye on twitter for release notice:
https://twitter.com/dietpi_

closed 2019-01-28 by Fourdee