MichaIng/DietPi #2433: Error during Install Nextcloud

WTFMaster, 2019-01-21:
Setup of Nextcloud stuck during Installation:
[image]
   
 │ DietPi-Software: cp -f /DietPi/dietpi/conf/apache.nextcloud.conf
             │ /etc/apache2/sites-available/dietpi-nextcloud.conf
             │  - Exit code: 1
             │  - DietPi version: v6.19.7 (Fourdee/master) | HW_MODEL:3 | HW_ARCH:2 | DISTRO:4
             │  - Image creator: DietPi Core Team
             │  - Pre-image: Raspbian Lite
             │
             │ Log file contents:
             │ cp: cannot stat '/DietPi/dietpi/conf/apache.nextcloud.conf': No such file or directory
             │
             │
             │
             │ Unable to continue, DietPi-Software will now terminate.
             │                                                                                        ▒
             │                Retry         : Re-run the last command that failed                     ↓
             │                DietPi-Config : Edit network, APT/NTP mirror settings etc

WTFMaster, 2019-01-21:
Awesome!

I would really appreachiate, when the php.ini will be optimized by the Installer regarding to the recommended settings of Nextcloud

opcache.enable=1
opcache.enable_cli=1
opcache.interned_strings_buffer=8
opcache.max_accelerated_files=10000
opcache.memory_consumption=128
opcache.save_comments=1
opcache.revalidate_freq=1

memory_limit = 512M

MichaIng, 2019-01-22:
@WTFMaster 
Will try to replicate.
____________
The `imagick` warning should be ignored. This is not required, just enables the create previews for some very rare image formats, used by the Theming app and disappears, if you disable theming.
There is even some discussion about that on Nextcloud forum, that imagick is an outdated app that implies some security holes, so should be really not enabled, is you not explicitly require a feature implied.
______________
PHP7.0 warning is known, we are thinking about how to solve this on Debian Stretch, which natively ships PHP7.0 only: https://github.com/Fourdee/DietPi/issues/2367
__________
The `memory_limit` value is actually served by Nextclouds own `.htaccess` file.
___________

Ahh, I know what went wrong, respectively I forgot:
```
a2ensite dietpi-nextcloud
systemctl reload apache2
```

MichaIng, 2019-01-26:
@WTFMaster 
Are the OPcache warnings resolved with the two commands above?

MichaIng, 2019-02-09:
Mark as closed, since initial error is solved and second only replicable with Lighttpd + HTTPS enabled: https://github.com/Fourdee/DietPi/issues/2489

closed 2019-02-09 by MichaIng