MichaIng/DietPi #2434: DietPi-WiFi-Monitor | Service startup not reliable on boot

MichaIng, 2019-01-22:
Ref: https://dietpi.com/phpbb/viewtopic.php?p=16169#p16169

Associated commits:
- https://github.com/Fourdee/DietPi/commit/69fbd3825a1fc09fb3aa5251b1e42de605184c07#diff-80b2c3c813fefbd33d0a7d43b5de7084
- https://github.com/Fourdee/DietPi/commit/0e7bb27b08f4b1df924d65d5b8b6a679fbccc266#diff-80b2c3c813fefbd33d0a7d43b5de7084
- https://github.com/Fourdee/DietPi/commit/0140b639e7e415963fa9ab5009b2ca3021583976#diff-80b2c3c813fefbd33d0a7d43b5de7084
- https://github.com/Fourdee/DietPi/commit/72e7c8c37c180f826f32a47a5146e32bac88b02a#diff-16e0f9bb73e19edf0a16d7bf732c7e92

Need to run some tests on my notebook, but I think the issue indeed was too early start of the service on boot, especially since `network-online.target` is not pulled in by default: https://www.freedesktop.org/software/systemd/man/systemd.special.html#network-online.target
So the service started directly with `network.target` (where network is not yet necessarily online) and possibly parallel to DietPi-Boot.

Fourdee, 2019-01-28:
@MichaIng 

Should be fine as starts after ```dietpi-boot``` which does ```Wait_For_Valid_Network_Connection```.

Ideally, user should set:
- ```dietpi-config``` > 8  : Network Options: Misc > Boot Net Wait: Infinite

This will ensure connection is valid, before rest of system boots.

MichaIng, 2019-01-28:
@Fourdee 
We could disable WiFi monitor service, if network connection times out during `dietpi-boot`. However user might still expect it to run.
But I think it's fine for now. 10s is a very long time already to connect. If connection fails in 10s, then something with the WiFi AP is wrong. And 10s wait on boot fits 10s tick rate of WiFi monitor and is a good trade off IMO.

MichaIng, 2019-03-02:
The service now pulls in `network-online.target` which further assures that it does not start after successful network connection: https://github.com/MichaIng/DietPi/commit/09a2b4ceab34fbdb32e1bb7343dd55440327dce1

In combination with the other fixes/commits this should be now bullet-prove. I mark this as closed.

closed 2019-03-02 by MichaIng