MichaIng/DietPi #2435: Bazarr not working anymore. Probably needs gevent but cannot be installed from python pip

garret, 2019-01-22:
I have DietPi v6.19.7 installed on a Raspberry Pi 3B. Here I have had [Bazarr](https://github.com/morpheus65535/bazarr) and running without any issues. It was installed by giving these commands on a fresh install: 
```
apt-get install git
cd /mnt/dietpi_userdata/
git clone https://github.com/morpheus65535/bazarr.git
python /mnt/dietpi_userdata/bazarr/bazarr.py & 
```

Recently I have noticed that Bazarr was not working anymore because I could not access to it via browser. 
So when giving: 
```
root@transmission:~# Bazarr starting...
2019-01-22 16:09:53,400 - root                             (76f7a8c0) :  ERROR (main:53) - BAZARR require gevent Python module to be installed using pip.
Bazarr exited.
^C
[1]+  Done                    python /mnt/dietpi_userdata/bazarr/bazarr.py
root@transmission:~#
```

I installed the packages _git-core_ and _python-pip_ through apt-get as suggested by [Bazarr wiki](https://github.com/morpheus65535/bazarr/wiki/Installation-Running-from-Source). Later, I gave the command to install pip but this gives me an error. Probably this is why Bazarr is not working anymore?! 
```
root@transmission:~# pip install gevent
Collecting gevent
  Using cached https://files.pythonhosted.org/packages/ed/27/6c49b70808f569b66ec7fac2e78f076e9b204db9cf5768740cff3d5a07ae/gevent-1.4.0.tar.gz
  Ignoring cffi: markers 'sys_platform == "win32" and platform_python_implementation == "CPython"' don't match your environment
Collecting greenlet>=0.4.14; platform_python_implementation == "CPython" (from gevent)
  Using cached https://files.pythonhosted.org/packages/f8/e8/b30ae23b45f69aa3f024b46064c0ac8e5fcb4f22ace0dca8d6f9c8bbe5e7/greenlet-0.4.15.tar.gz
Building wheels for collected packages: gevent, greenlet
  Running setup.py bdist_wheel for gevent ... error
  Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-xDLbDB/gevent/setup.py';f=getattr(tokeniz  e, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmpToyvuNpip-whe  el- --python-tag cp27:
  /usr/lib/python2.7/distutils/dist.py:267: UserWarning: Unknown distribution option: 'cffi_modules'
    warnings.warn(msg)
  /usr/lib/python2.7/distutils/dist.py:267: UserWarning: Unknown distribution option: 'project_urls'
    warnings.warn(msg)
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help

  error: invalid command 'bdist_wheel'

  ----------------------------------------
  Failed building wheel for gevent
  Running setup.py clean for gevent
  Running setup.py bdist_wheel for greenlet ... error
  Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-xDLbDB/greenlet/setup.py';f=getattr(token  ize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmp6iWSxvpip-w  heel- --python-tag cp27:
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help

  error: invalid command 'bdist_wheel'

  ----------------------------------------
  Failed building wheel for greenlet
  Running setup.py clean for greenlet
Failed to build gevent greenlet
Installing collected packages: greenlet, gevent
  Running setup.py install for greenlet ... error
    Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-xDLbDB/greenlet/setup.py';f=getattr(tok  enize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-DjgVyd  -record/install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    running build_ext
    building 'greenlet' extension
    creating build
    creating build/temp.linux-armv7l-2.7
    arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdeb  ug-prefix-map=/build/python2.7-Ub3vap/python2.7-2.7.13=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I/usr/include/python2.7 -c   greenlet.c -o build/temp.linux-armv7l-2.7/greenlet.o
    In file included from greenlet.c:5:0:
    greenlet.h:8:20: fatal error: Python.h: No such file or directory
     #include <Python.h>
                        ^
    compilation terminated.
    error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

    ----------------------------------------
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-xDLbDB/greenlet/setup.py';f=getattr(tokenize, 'open', open)(__f  ile__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-DjgVyd-record/install-record.t  xt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-xDLbDB/greenlet/
root@transmission:~#
```

Do you have any idea why I get this error? 

I recently reinstalled Home Assistant from the optimized software because it got an update. I don't know if that screwed something in the python installation that affects Bazarr (Home Assistant works ok instead).

MichaIng, 2019-01-22:
I guess it is due to two different Python versions, 2.7 and 3.

Just for a quick test, could you try to run the bazaar script with python3 command?
Otherwise we need to make a deeper look. Generally Python2 and 3 should work fine beside each other as long as you start each script with the correct version.

MichaIng, 2019-01-22:
Hmm difference in install output/steps is:
```
Ignoring cffi: markers 'sys_platform == "win32" and platform_python_implementation == "CPython"' don't match your environment
...
Building wheels for collected packages: gevent, greenlet
  Running setup.py bdist_wheel for gevent ... error
```
- The `cffi` thing should be not related, but the `wheels building` is the one that fails first. Not sure what it is, but this is what needs to be checked.

garret, 2019-01-22:
Thank you MichaIng for the support. 

Trying the first command you told me still gives me an error: 
```
root@transmission:~# pip --no-cache-dir install greenlet
Collecting greenlet
  Downloading https://files.pythonhosted.org/packages/f8/e8/b30ae23b45f69aa3f024b46064c0ac8e5fcb4f22ace0dca8d6f9c8bbe5e7/greenlet-0.4.15.tar.gz (59kB)
    100% |████████████████████████████████| 61kB 1.5MB/s
Installing collected packages: greenlet
  Running setup.py install for greenlet ... error
    Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-RGAYZa/greenlet/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-fGtLRN-record/install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    running build_ext
    building 'greenlet' extension
    creating build
    creating build/temp.linux-armv7l-2.7
    arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-Ub3vap/python2.7-2.7.13=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I/usr/include/python2.7 -c greenlet.c -o build/temp.linux-armv7l-2.7/greenlet.o
    In file included from greenlet.c:5:0:
    greenlet.h:8:20: fatal error: Python.h: No such file or directory
     #include <Python.h>
                        ^
    compilation terminated.
    error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

    ----------------------------------------
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-RGAYZa/greenlet/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-fGtLRN-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-RGAYZa/greenlet/
root@transmission:~#
```

The same is valid for the second command: 

```
root@transmission:~# pip --no-cache-dir install gevent
Collecting gevent
  Downloading https://files.pythonhosted.org/packages/ed/27/6c49b70808f569b66ec7fac2e78f076e9b204db9cf5768740cff3d5a07ae/gevent-1.4.0.tar.gz (5.2MB)
    100% |████████████████████████████████| 5.2MB 8.0MB/s
  Ignoring cffi: markers 'sys_platform == "win32" and platform_python_implementation == "CPython"' don't match your environment
Collecting greenlet>=0.4.14; platform_python_implementation == "CPython" (from gevent)
  Downloading https://files.pythonhosted.org/packages/f8/e8/b30ae23b45f69aa3f024b46064c0ac8e5fcb4f22ace0dca8d6f9c8bbe5e7/greenlet-0.4.15.tar.gz (59kB)
    100% |████████████████████████████████| 61kB 7.7MB/s
Installing collected packages: greenlet, gevent
  Running setup.py install for greenlet ... error
    Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-nA8aYN/greenlet/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-HHwonb-record/install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    running build_ext
    building 'greenlet' extension
    creating build
    creating build/temp.linux-armv7l-2.7
    arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-Ub3vap/python2.7-2.7.13=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I/usr/include/python2.7 -c greenlet.c -o build/temp.linux-armv7l-2.7/greenlet.o
    In file included from greenlet.c:5:0:
    greenlet.h:8:20: fatal error: Python.h: No such file or directory
     #include <Python.h>
                        ^
    compilation terminated.
    error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

    ----------------------------------------
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-nA8aYN/greenlet/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-HHwonb-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-nA8aYN/greenlet/
root@transmission:~#
```

This is the output of the commands that you suggested earlier instead:
```
root@transmission:~# which -a python && readlink -f $(which python)
/usr/bin/python
/usr/bin/python2.7
root@transmission:~# which -a pip && readlink -f $(which pip)
/usr/bin/pip
/usr/bin/pip
root@transmission:~# 
```

I tried to use python3 as command to start Bazarr but still unsuccessful:
```
root@transmission:~# python3 /mnt/dietpi_userdata/bazarr/bazarr.py &
[1] 10953
root@transmission:~#   File "/mnt/dietpi_userdata/bazarr/bazarr.py", line 15
    print 'bazarr.py -h --no-update --config <config_directory>'
                                                               ^
SyntaxError: Missing parentheses in call to 'print'
^C
[1]+  Exit 1                  python3 /mnt/dietpi_userdata/bazarr/bazarr.py
root@transmission:~#
```

Finally, I tired to clone again Bazarr in another new folder and start it but got the same issues as above. Thus it seems related to the python installation. 

On my dietpi, apart from home assistant, I also have installed through the optimized software: docker, sonarr, radarr, jackett and certbot.