MichaIng/DietPi #2439: shairport-sync not working

seniorgod, 2019-01-25:
## Creating a bug report/issue:
on a fresh install on DietPi on Sparky/USBridge shairport-sync is running but the streams break down after some seconds or minutes. 
My Apple devices find the shairport-sync-device. The devices are able to stream to shairport-sync. The stream is received and i can hear music via my Hifi. After some seconds or minutes the audio abort. On the iPhone it look like the music is still streaming, but i hear nothing. 

I found no error in configuration.
I had no other idea then to do my own compilation of shairport-sync.
After loading shairport-sync from git (an all libraries it depending on), do the described steps from
https://github.com/mikebrady/shairport-sync
everything work flawlessly.
  
the only miracle i found in journalctl is: 
Jan 22 03:14:14 DietPi shairport-sync[2181]: ALSA lib pcm_hw.c:1713:(_snd_pcm_hw_open) Invalid value for card

Because i found a workaround (compile shairport-sync by myself) you can close the issue. I want only publish it you. 

thanks

#### Required Information:
- DietPi version	| `cat /DietPi/dietpi/.version`
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=19
G_DIETPI_VERSION_RC=7
G_GITBRANCH=master
G_GITOWNER=Fourdee
- Distro version	| `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
9.6
- Kernel version	| `uname -a`
Linux DietPi 3.10.38 #22 SMP PREEMPT Fri Dec 22 10:45:41 IST 2017 armv7l GNU/Linux
- SBC device		| `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
allo sparky with usbridge
- Power supply used	| (EG: 5V 1A RAVpower)
5V 3A
- SDcard used		| (EG: SanDisk ultra)
sandig ultra 

#### Additional Information (if applicable):
- Software title	| shairport-sync
- Was the software title installed freshly or updated/migrated?
- freshly installed from dietPi minimal image
- Can this issue be replicated on a fresh installation of DietPi?
the installation was a fresh one
- `dietpi-bugreport` ID

Fourdee, 2019-01-26:
@seniorgod 

Many thanks for the report + solution. 👍 

Our current version binaries are ```shairport-sync_3.1.7```. I'll rebuild latest and release for v6.20.

seniorgod, 2019-01-26:
I test also spotifyd 

It’s a spotify daemon based on rust language.

Advantage: 
  - simpler user/password login - you do not need spotify_app_key 
  - automation with scripts which run at start or stop music 
  - auto discover without additional avahi daemon

Disadvantage:
  - password is readable in config on disk 
  - can’t be controlled from outside 

Matthias

Fourdee, 2019-01-26:
Install tests:
- 🈯️ armv6
- 🈯️ armv7
- 🈯️ armv7
- 🈯️ x86_64

Completed, patched in v6.20

closed 2019-01-26 by Fourdee

seniorgod, 2019-01-26:
did you include libalac  for shairport-sync ?

other question regarding mpd. 
Do you update to 0.21.x ? in dietPi 6.20 ? 

best regards

seniorgod, 2019-01-26:
> @seniorgod <https://github.com/seniorgod>
> Do you update to 0.21.x ? in dietPi 6.20 ?
>
> MPD is updated to 0.20.23
> 0.21.3 failed to build: #2377 (comment) <https://github.com/Fourdee/DietPi/issues/2377#issuecomment-450977281>

short version: is compiled mpd 21.4 on DietPi and it works! 

long version: 

i asked you because following point from the 0.21.x mpd changelog sound very interesting: 
* output
  - alsa: non-blocking mode
Today the problem with running shairport, spotifyd and mpd in parallel is, that mpd locks the alsa output. You have to stop streaming mpd before starting apple airplay or spotify 

after your answer i start trying to compile mpd following the instructions on: 
   https://musicpd.org/doc/html/user.html

First i run into an error because the used build env named „meson“ (i never heard of this before; is it the typical me too project?, i’m simply too old)
So i have to download meson from stretch-backports and follow the instructions 

apt-get -t stretch-backports install meson

apt install g++ \
  libpcre3-dev \
  libmad0-dev libmpg123-dev libid3tag0-dev \
  libflac-dev libvorbis-dev libopus-dev \
  libadplug-dev libaudiofile-dev libsndfile1-dev libfaad-dev \
  libfluidsynth-dev libgme-dev libmikmod2-dev libmodplug-dev \
  libmpcdec-dev libwavpack-dev libwildmidi-dev \
  libsidplay2-dev libsidutils-dev libresid-builder-dev \
  libavcodec-dev libavformat-dev \
  libmp3lame-dev libtwolame-dev libshine-dev \
  libsamplerate0-dev libsoxr-dev \
  libbz2-dev libcdio-paranoia-dev libiso9660-dev libmms-dev \
  libzzip-dev \
  libcurl4-gnutls-dev libyajl-dev libexpat-dev \
  libasound2-dev libao-dev libjack-jackd2-dev libopenal-dev \
  libpulse-dev libshout3-dev \
  libsndio-dev \
  libmpdclient-dev \
  libnfs-dev libsmbclient-dev \
  libupnp-dev \
  libavahi-client-dev \
  libsqlite3-dev \
  libsystemd-dev libwrap0-dev \
  libgtest-dev \
  libboost-dev \
  libicu-dev

# source code  download 

wget http://www.musicpd.org/download/mpd/0.21/mpd-0.21.4.tar.xz
tar -xf mpd-0.21.4.tar.xz 
cd mpd-0.21.4
meson . output/release --buildtype=debugoptimized -Db_ndebug=true

meson configure output/release

ninja -C output/release

ninja -C output/release install

i get a big bunch of notes like this : 

/usr/include/c++/6/bits/stl_vector.h:1178:71: note: parameter passing for argument of type ‘__gnu_cxx::__normal_iterator<DetachedSong*, std::vector<DetachedSong> >’ will change in GCC 7.1
  return _M_erase(__beg + (__first - __cbeg), __beg + (__last - __cbeg));
                                                                       ^
/usr/include/c++/6/bits/stl_vector.h:1178:71: note: parameter passing for argument of type ‘__gnu_cxx::__normal_iterator<DetachedSong*, std::vector<DetachedSong> >’ will change in GCC 7.1
/usr/include/c++/6/bits/stl_vector.h:1178:71: note: parameter passing for argument of type ‘__gnu_cxx::__normal_iterator<DetachedSong*, std::vector<DetachedSong> >’ will change in GCC 7.1
/usr/include/c++/6/bits/stl_vector.h:1178:71: note: parameter passing for argument of type ‘__gnu_cxx::__normal_iterator<DetachedSong*, std::vector<DetachedSong> >’ will change in GCC 7.1

but mpd is build in version 21.4. 

And YEP. MPD now uses  a non blocking also mode. I can start airplay while streaming from mpd and it simply change.
So it would be a positive effect to change to mpd 21.4 for DietPi 

best regards

seniorgod, 2019-01-26:
one additional point: 

i found that the build script "installed" /usr/local as location for mpd. therefore mpd expects its config file under /usr/local/etc .... So i linked it from /etc to /usr/local/etc

no idea to change it

Fourdee, 2019-01-28:
@seniorgod 

>  that mpd locks the alsa output.

Yep. I actually prefer this as the audio hardware is dedicated to the active stream. Also ensures programs like Roon (which attaches directly to the hardware) play nice.

MichaIng, 2019-01-28:
@seniorgod 
> i found that the build script "installed" /usr/local as location for mpd. therefore mpd expects its config file under /usr/local/etc .... So i linked it from /etc to /usr/local/etc

Jep, we also found this and did the same for v6.20: https://github.com/Fourdee/DietPi/issues/2378#issuecomment-451315809

seniorgod, 2019-02-01:
> Dear all,
>
> i open a issue at mpd in Github reading this point.
> I give you notice if i get an answer

I get an answer from the mpd maintainer: 

„...
If you want MPD to look for mpd.conf in /etc/ instead of /usr/local/etc/ (=Meson's default) configure with -Dsysconfdir=/etc.
…"

best regards