MichaIng/DietPi #2445: ROCK Pi 4

tenrek, 2019-01-26:
So, any plans for support / make an DietPi for [ROCK Pi 4](http://rockpi.org/) ?

[Debian Desktop download](https://wiki.radxa.com/Rockpi4/downloads)

MichaIng, 2019-01-26:
@tenrek 
If we get a developer sample or @Fourdee is willing to buy one, we can create an image.

Official Debian Stretch desktop image available: https://wiki.radxa.com/Rockpi4/downloads
_____________
Hmm what I don't like when reading the website:
- They claim 4th gen RPi, but it has nothing to do with "Raspberry Pi" and there are no earlier ROCK Pi versions, AFAIK.
- Their graphics and website style even mimic the once from Raspberry Pi as well.
- Board style as well very similar, although this is often the case with model B SBCs.
- So overall I get the impression it is tried to build/benefit from the popularity of the Raspberry Pi by subtly mimic it and giving the impression it would be an upgraded RPi version.
- But after all it's a new "Radxa Lmited" SBC brand from China and we will see how well they maintain the product with up-to-date OS images 🤔.

Fourdee, 2019-01-27:
@MichaIng 

> subtly mimic it and giving the impression it would be an upgraded RPi version.

Yep, although ASUS TB has the same form factor as a RPi 3. However, seems the Chinese companies can go much further to "copy and promote a copy" without their copyright laws lol.

MichaIng, 2019-01-27:
@Fourdee 
Jep nothing against identical form factor. I am fans of standards, so e.g. SBC cases can fit for many different SBCs. But ROCK Pi goes a bid too far here. Calling it RPi 4 and also `v1.3` on top of Raspberry Pi 3B being `v1.2`. This is not about standards only, at least all is pointing in this direction to use a bid more careful wording 😉.

MichaIng, 2019-01-27:
I was reading a bid the forum: https://forum.radxa.com/c/community
- Feedback from users (many from UK) looks good.
- Currently only sold at AllNet, however many requests for having this on Amazon or other shops to make it easily available in Europe.
- It is still in early stage: Official heatsink in development
- Mainline kernel 5.1 integration will be done, so we can expect up-to-date kernels after 5.1 release.
- Currently they are active in development, up-to-date images and support forum. If this stays, we would have a great Chinese manufacturer compared to the experiences with OPi/BPi.

@Fourdee 
Perhaps they will send you a developer sample, when asking? 😃

MichaIng, 2019-06-23:
PR up to do initial definitions: https://github.com/MichaIng/DietPi/pull/2933
User image incoming for testing and further implementing into DietPi: https://dietpi.com/phpbb/viewtopic.php?p=18445#p18445

canabino, 2019-06-24:
@MichaIng 
Hi there so, please walk me through creating this image. I have the [script](https://raw.githubusercontent.com/MichaIng/DietPi/dev/PREP_SYSTEM_FOR_DIETPI.sh) you sent.
I have an eMMC module and microSD card. Which one should I use? After I installed the Armbian image with the 5.1 kernel on the RockPi. Should I run the [DietPi-PREP](https://github.com/MichaIng/DietPi/issues/1285)? Then what? Take out the eMMC/microSD place it in my windows pc running what OS on a virtual machine? Then run the script?
Thanks

canabino, 2019-06-24:
I installed the latest Armbian with the 5.1 kernel and ran the DietPi-PREP script.
It came up with the following errors.
```
#### Details:
- Date           | Thu  6 Jun 15:39:03 UTC 2019
- Bug report     | N/A
- DietPi version | v6.24.1 (MichaIng/master)
- Img creator    | n/a
- Pre-image      | n/a
- SBC device     | NULL (index=22)
- Kernel version | #5.88 SMP Thu Jun 6 17:19:49 CEST 2019
- Distro         | stretch (index=4)
- Command        | G_AGUP
- Exit code      | 100
- Software title | DietPi-PREP

#### Extra details:
<!-- Please post any extra details that might help solve the issue -->
- ...

#### Additional logs:
Log file contents:
Ign:1 https://apt.armbian.com stretch InRelease
Ign:2 https://cdn-aws.deb.debian.org/debian stretch InRelease
Hit:3 https://cdn-aws.deb.debian.org/debian stretch-updates InRelease
Hit:4 https://cdn-aws.deb.debian.org/debian-security stretch/updates InRelease
Hit:5 https://cdn-aws.deb.debian.org/debian stretch-backports InRelease
Hit:6 https://cdn-aws.deb.debian.org/debian stretch Release
Err:7 https://apt.armbian.com stretch Release
  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
Reading package lists...
E: The repository 'http://apt.armbian.com stretch Release' does no longer have a Release file.

```

MichaIng, 2019-06-24:
Please try:
```
curl -sSL https://apt.armbian.com/armbian.key | apt-get add -
apt update
```

canabino, 2019-06-24:
I changed the repository in /etc/apt/sources.list.d/armbian.list
And it is working now.

MichaIng, 2019-06-24:
@canabino 
Ah, what was the error with this repo? "https://apt.armbian.com stretch Release" above at least looks like it should.

I am a bid too tired now to write down to go through the DietPi-PREP steps, need to sleep, will check back tomorrow.

Just quick note about above:
Since I just added the ROCK Pi 4 definition into dev branch, you might want to run the script from that branch + install dev DietPi:
```
cd /tmp
wget https://raw.githubusercontent.com/MichaIng/DietPi/dev/PREP_SYSTEM_FOR_DIETPI.sh
chmod +x PREP_SYSTEM_FOR_DIETPI.sh
./PREP_SYSTEM_FOR_DIETPI.sh
```
- When it asks for the branch, select "dev". After v6.25 release, when we want to publish this for testing+feedback, we can manually revert to master target.

canabino, 2019-06-24:
I changed it from "https://apt.armbian.com to https://mirrors.tuna.tsinghua.edu.cn/armbian following a google search.

I will run the code you sent.
Sleep well and chat tomorrow.

canabino, 2019-06-24:
I changed back to https://apt.armbian.com and `apt-get update`  is working fine now. Maybe the server was down at the time.
I ran your script and chose "dev"
I was able to choose RockPi4 from the list.
It ran and then came up with the following errors.
```
DietPi Error Handler: 
DietPi-PREP: systemctl enable dietpi-firstboot
 - Exit code: 1
 - DietPi version: v6.24.1 (MichaIng/master) | HW_MODEL:72 | HW_ARCH:3 | DISTRO:4
 - Image creator: Canabino
 - Pre-image: Armbian
 
 Log file contents:
 Failed to enable unit: File dietpi-firstboot.service: No such file or directory
 Unable to continue, DietPi-PREP will now terminate.
```

MichaIng, 2019-06-24:
@canabino 
Found a bug in our new env vars implementation. The chosen Git info was overwritten by master branch. Fixed: https://github.com/MichaIng/DietPi/commit/e3f97a0d40ca01f97e894257bf90621fb2831bbf

Ah you second try seems to have been shortly after I fixed it, great.

Okay, next is to shutdown the system (of course) and plug SDcard or eMMC onto an external system. If you want to spin up a VM, take our VM images 🙂: https://dietpi.com/#download

There:
```
cd /tmp
wget https://github.com/MichaIng/DietPi/blob/dev/.meta/dietpi-imager
chmod +x dietpi-imager
./dietpi-imager
```

Hmm, I see now that `dd` runs as first step. I added this as last step on the minimised partition/file system as last step, but @Fourdee switched this. Ups and downs:
- The current way the disk/SDcard will stay untouched, so one can always redo if something fails.
- Also since the HDD most likely is faster than the SDcard, some steps will be faster.
- But `dd` takes much longer since it copies the whole SDcard size.

canabino, 2019-06-24:
Okay I dowloaded and installed VirtualBox.
Then I downloaded your `DietPi_VirtualBox-x86_64-Stretch.ova`
It booted up and and I did all the installations for DietPi.
I added the usb device with microSD from the RockPi  to the VM.
It shows as `/dev/sdb/` when I ran `fdisk -l` Can I run your script like this?
[image]

MichaIng, 2019-06-24:
@canabino 
Jep, the script asks you which drive to pick.

canabino, 2019-06-24:
ok 👍

canabino, 2019-06-24:
It came up with errors
[image]

MichaIng, 2019-06-24:
@canabino 
Ah me dumb, sorry wrong download link.
This one: `wget https://raw.githubusercontent.com/MichaIng/DietPi/dev/.meta/dietpi-imager`

It did download the html page when opening the file on GitHub 😉.

canabino, 2019-06-24:
I received the same error
 
[image]

MichaIng, 2019-06-24:
@canabino 
It was saved to "dietpi-imager.1" since wget does not automatically overwrite files:
```
mv dietpi-imager.1 dietpi-imager
chmod +x dietpi-imager
./dietpi-imager
```

canabino, 2019-06-24:
ok

canabino, 2019-06-24:
I'm in 👍

canabino, 2019-06-24:
[image]

MichaIng, 2019-06-24:
@canabino 
Would be nice to have disk size in this menu, at least for me this is always the best indicator to know which drive is which.

canabino, 2019-06-24:
You are right it would be nice.

canabino, 2019-06-24:
Just a quick mention I loaded the Armbian with the 5.1 kernel on my system using the same script and everything was working fine. NordVPN and Jackett.

canabino, 2019-06-24:
Where will it store the image?

MichaIng, 2019-06-24:
@canabino 
To `/root/DietPi_XXXX_XXX.img` and `/root/DietPi_XXXX_XXX.7z` (the archive).

canabino, 2019-06-24:
Oh, thanks

canabino, 2019-06-24:
It's done.
[image]

MichaIng, 2019-06-24:
@canabino 
You ran out of disk space sadly. This was the initial `dd` to create a raw image which will be resized then. It should require as much space as the SDcard size, so 15 GiB.

canabino, 2019-06-24:
Oh should I make the VM's drive bigger?

canabino, 2019-06-24:
[image]

MichaIng, 2019-06-24:
@canabino 
Ah of course did not think about that. Before doing so, do:
```
rm /root/myimage.img
systemctl enable dietpi-fs_partition_resize
```

This enables the partition resize service for next boot.

canabino, 2019-06-24:
[image]

canabino, 2019-06-24:
I rebooted. But the partition is still 8GB

canabino, 2019-06-25:
I adjusted it in the VirtualBox
[image]

canabino, 2019-06-25:
Now it doesn't want to boot.
Give me a moment to set it up again.

canabino, 2019-06-25:
I have adjusted it to 32GB  and will start again.
```
root@DietPi:~# fdisk -l
Disk /dev/sda: 32.2 GiB, 34611757056 bytes, 67601088 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x13ced30f

Device     Boot Start      End  Sectors Size Id Type
/dev/sda1  *     2048 16775167 16773120   8G 83 Linux

Disk /dev/sdb: 14.9 GiB, 15931539456 bytes, 31116288 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x9df16f21

Device     Boot Start      End  Sectors  Size Id Type
/dev/sdb1       32768 30805119 30772352 14.7G 83 Linux

```

MichaIng, 2019-06-25:
@canabino 
Note that the partition still needs to be extended, hence the `systemctl enable dietpi-fs_partition_resize` + reboot.

canabino, 2019-06-25:
ok

canabino, 2019-06-25:
I ran the partition resize code and rebooted.
I started the DietPi-Imager script again but it still ran out of space.
```
root@DietPi:/tmp# systemctl enable dietpi-fs_partition_resize
Created symlink /etc/systemd/system/local-fs.target.wants/dietpi-fs_partition_resize.service → /etc/systemd/system/dietpi-fs_partition_resize.service.
root@DietPi:/tmp# reboot
```
```
root@DietPi:/tmp# wget https://raw.githubusercontent.com/MichaIng/DietPi/dev/.meta/dietpi-imager
--2019-06-25 01:56:08--  https://raw.githubusercontent.com/MichaIng/DietPi/dev/.meta/dietpi-imager
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.4.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.4.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 7312 (7.1K) [text/plain]
Saving to: ‘dietpi-imager’

dietpi-imager                       100%[=================================================================>]   7.14K  --.-KB/s    in 0s

2019-06-25 01:56:08 (219 MB/s) - ‘dietpi-imager’ saved [7312/7312]

root@DietPi:/tmp# chmod +x dietpi-imager
root@DietPi:/tmp# ./dietpi-imager
[  OK  ] DietPi-Imager | Root access verified.
[  OK  ] DietPi-Imager | RootFS R/W access verified.
6553194496 bytes (6.6 GB, 6.1 GiB) copied, 1764.03 s, 3.7 MB/s
dd: error writing '/root/myimage.img': No space left on device
1600498+0 records in
1600497+0 records out
6555635712 bytes (6.6 GB, 6.1 GiB) copied, 1764.72 s, 3.7 MB/s
root@DietPi:/tmp#
```

MichaIng, 2019-06-25:
@canabino 
Strange, as it stops at the same point, the expansion seems to not have worked:
```
parted /dev/sdb print
cat /var/tmp/dietpi/logs/fs_partition_resize.log
```

canabino, 2019-06-25:
```
root@DietPi:~# parted /dev/sda print
Model: ATA VBOX HARDDISK (scsi)
Disk /dev/sda: 34.6GB
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags:

Number  Start   End     Size    Type     File system  Flags
 1      1049kB  8589MB  8588MB  primary  ext4         boot

root@DietPi:~# cat /var/tmp/dietpi/logs/fs_partition_resize.log
Removed /etc/systemd/system/local-fs.target.wants/dietpi-fs_partition_resize.service.
/var/lib/dietpi/services/fs_partition_resize.sh: line 45: cannot create temp file for here-document: No space left on device
resize2fs 1.43.4 (31-Jan-2017)
The filesystem is already 2096640 (4k) blocks long.  Nothing to do!
```

I used the DietPi-Drivemanager to resize to maximum.
[image]

Now it shows the right size.
```
root@DietPi:~# parted /dev/sda print
Model: ATA VBOX HARDDISK (scsi)
Disk /dev/sda: 34.6GB
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags:

Number  Start   End     Size    Type     File system  Flags
 1      1049kB  34.6GB  34.6GB  primary  ext4         boot

```
I am running the Imager now.

MichaIng, 2019-06-25:
@canabino 
Lol was a snake biting its tails. Some part of the resizing script failed because there was no space left 😄. Okay great it worked now.

canabino, 2019-06-25:
🐍 🤣 
It is almost finished, do you want the archive or the img?
Where should I upload to?

MichaIng, 2019-06-25:
@canabino 
The archive, please, lets not stress bandwidth more then required 😄.
Sounds funny but misusing the survey/bugreport user to upload to the DietPi server directly is easiest:
```
curl -T <archive_name>.7z sftp://dietpi-survey:upload2dietpi@ssh.dietpi.com/bugreport/
```

And would be great if you could then flash this image onto the SDcard again to test boot and firstrun setup. I am thinking if it's easier to use dd again from within the VM (backwards) or if I should provide you the link to the online archive, once I put it into an appropriate download directory.

The shared folder function of VirtualBox is actually helpful here to easily transfer files between host and guest, however requires guest extensions etc 😉.

canabino, 2019-06-25:
It's asking: `Is /root/myimage.img a GPT filesystem?`

MichaIng, 2019-06-25:
@canabino 
Ähhm good question, let me see.

MichaIng, 2019-06-25:
@canabino 
> Disklabel type: dos

No GPT, select "Cancel" => Should be called "No".

canabino, 2019-06-25:
ok

canabino, 2019-06-25:
[image]

MichaIng, 2019-06-25:
@canabino 
`1`

canabino, 2019-06-25:
`DietPi_<device>-<arch>-<distro>___`
DietPi_ROCK Pi 4-aarch64-??????

MichaIng, 2019-06-25:
`DietPi_ROCKPi4-ARMv8-Stretch`

canabino, 2019-06-25:
```
[  OK  ] DietPi-Imager | G_AGI gdisk dosfstools zerofree
e2fsck 1.43.4 (31-Jan-2017)
/dev/loop0p1: recovering journal
Clearing orphaned inode 1822 (uid=0, gid=0, mode=0100644, size=35568)
Clearing orphaned inode 13342 (uid=0, gid=0, mode=0100755, size=704920)
Clearing orphaned inode 34265 (uid=0, gid=0, mode=040755, size=4096)
Clearing orphaned inode 36839 (uid=0, gid=0, mode=0100644, size=758)
Clearing orphaned inode 1856 (uid=0, gid=0, mode=0100644, size=266992)
Clearing orphaned inode 9756 (uid=0, gid=0, mode=0100644, size=1679776)
Pass 1: Checking inodes, blocks, and sizes
Pass 2: Checking directory structure
Pass 3: Checking directory connectivity
Pass 4: Checking reference counts
Pass 5: Checking group summary information
Free blocks count wrong (3499616, counted=3639092).
Fix<y>?
```

MichaIng, 2019-06-25:
@canabino 
Hmm not good that there are already orphaned inodes and wrong inode count, however "yes".

canabino, 2019-06-25:
[image]

MichaIng, 2019-06-25:
@canabino 
Jep.

canabino, 2019-06-25:
Success!!!!!!
[image]

[116 more replies]

closed 2020-01-15 by MichaIng