MichaIng/DietPi #2447: Adding PiCase safe shutdown script causes freeze on next reboot

midwan, 2019-01-26:
## Creating a bug report/issue:

#### Required Information:
- DietPi version	| 6.19
- Distro version	| Stretch
- Kernel version	| Linux DietPi 4.14.93-v7+
- SBC device		| RPI3
- Power supply used	| 5V 2.5A
- SDcard used		| SanDisk ultra

#### Additional Information (if applicable):
- Can this issue be replicated on a fresh installation of DietPi?
- Yes

#### Steps to reproduce:

1. Install a fresh DietPi image
2. Upgrade system normally (apt update && apt upgrade)
3. Install the Safe Shutdown python script for the RetroFlag PiCase (https://github.com/midwan/retroflag-picase/blob/master/install.sh)
4. Reboot

#### Expected behaviour:

- The next boot should not be any different
- Additionally, the hardware buttons on the case should trigger the Shutdown and Reboot events as handled by the python script

#### Actual behaviour:

- The system fails to reach the login prompt after rebooting or powering up from scratch.

MichaIng, 2019-01-26:
@midwan 
Thanks for report.

Does `dmesg` or `journalctl` show something? Hmm hard to check if login on console is not possible. Is it possible to login via SSH instead or at least on serial console?

Hmm, on v6.19 `rc.local` will be called after `dietpi-postboot.service` prompts the login banner: `Before=rc-local.service rc.local.service`
Perhaps the login banner is just overridden by `rc.local` output and pressing return still allows login?
This is something we changed with v6.20: DietPi login prompt will be shown after `multi-user.target` reached, which is after all gettys initiated which is after `rc.local` has been called: https://github.com/Fourdee/DietPi/commit/b518b01fc879f147299e2a582ce4cf39fe244e14#diff-cfb2b2283867ce5b7b51bf3cedbaefba

A termporary dev state caused some dependency/order loop: `rc.local` after `postboot` after `getty` after `rc.local`, but that should not be the case in v6.19 nor v6.20 current state. But just to sort out any incompatibility could you try to remove `rc.local` from postboot order:
```
sed -i '/Before=/d' /etc/systemd/system/dietpi-postboot.service
systemctl daemon-reload
```

And another point:
- Actually `rc.local` is an obsolete/outdated feature on modern Debian systems, disabled by default and only implemented via compatibility systemd unit: `/lib/systemd/system/rc-local.service`
- Consider to apply the script via dedicated systemd unit outside of `rc.local`.

Aside from that, not sure how exactly python binds functions to those GPIO pins but that should not interfere with any DietPi boot script or config.

Finally just to sort out, does boot + login work well, if the `install.sh` is applied just without adding the python script to `rc.local`?

MichaIng, 2019-02-28:
@midwan 
To coming back to this with possibly related issues:
- As you are on RPi, it seems to current image we ship still ships an old DietPi touched version of the `rc-local.service`, which is incompatible with the new method how systemd loads rc.local. See: https://github.com/MichaIng/DietPi/issues/2566
- Not sure how RPi handles doubled `config.txt` entries, but on DietPi you might find `enable_uart=0`, which means that the installer doubles this entry: https://github.com/midwan/retroflag-picase/blob/master/install.sh#L14-L20
Perhaps use something like this, to have a failsafe replacement if this setting is present with any value:
```
if grep -q '^[[:blank:]]*enable_uart=' "$File"; then
    sed -i '/^[[:blank:]]*enable_uart=/c\enable_uart=1' "$File"
else
    echo 'enable_uart=1' >> $File
fi
```
- And I am not sure what the python shutdown script does and what it uses, but note that on DietPi `dbus` and `systemd-logind` are masked by default. So you might want to try to unmask them, as boot/shutdown/login/logout things somethings use this, especially dbus:
```
systemctl unmask dbus
systemctl unmask systemd-logind
```

midwan, 2019-03-01:
@MichaIng 
Thank you for the extra details. I'm planning to give this another try this weekend, so I'll report back after that.

MichaIng, 2019-04-27:
@midwan 
Does the issue still exist? I just rechecked PiCase installer and shutdown script and that should work without `systemd-logind` or `dbus`. (Btw dbus is not masked but not installed on fresh DietPi, so `apt install dbus` should have been the correct try.)
Especially there should no effect survive until next boot

If boot still hangs, then boot messages would be helpful, serial console paste or picture from attached screen or such, if possible.

MichaIng, 2019-05-26:
Closing due to missing user reply. Feel free to reopen if required.

closed 2019-05-26 by MichaIng