MichaIng/DietPi #2456: USB Installer | Native PC (BIOS)

louisefindlay23, 2019-01-28:
#### Is your feature request related to a problem? Please describe:

- ... I would love if you could implement a USB installer for the Native BIOS PC image as I can never get it to install properly to my hard drive. It always has a small root partition which I can't expand. 

It's been mentioned in #1007 that it could be worked on if the image was popular.

#### Additional context:

- ...
A similar implementation is Libreelec's USB-SD creator which creates a bootable usb which you can then select the drive you'd like to install to [https://libreelec.tv/downloads/](https://libreelec.tv/downloads)

MichaIng, 2019-01-28:
@louisefindlay23 
Thanks for your request.

Hmm, the partition and file system should be able to expand, AFAIK, and this should be done automatically on first boot.

Do you face some error message when trying it manually?
Also please paste the log or our expansion script: `cat /var/tmp/dietpi/logs/fs_partition_resize.log`

louisefindlay23, 2019-01-30:
Finally managed to installed using UEFI Rufus/Clonezilla but in my /dev/sda drive (223.6GB total) I have two partitions /dev/sda1 that is 512M EFI filesystem and /dev/sda2 which is DietPi's root directory at 28.4 GB. 

It says warning not all space available to /dev/sda appears to be used, you can fix the GPT to use all the available space (an extra 408241328 blocks) or continue with current setting?

How do I fix this?

louisefindlay23, 2019-01-30:
Managed to fix it using resizepart and got around 200 GB in my /dev/sda2 partition. A tool would be great to make it easier though.

MichaIng, 2019-02-28:
@louisefindlay23 
Sorry for the very late reply.

So as said, the root partition file system should be resized automatically on first boot of the DietPi image. 
`cat /var/tmp/dietpi/logs/fs_partition_resize.log` should give some hint, if this failed.

In case of BIOS image, GTP partitions are not required.
In case of UEFI image, GTP+EFI partition is required and this cannot be done directly with flashing an laid out image, but instead we ship this as CloneZilla bundled installer: https://github.com/MichaIng/DietPi/issues/1171
But as well on first boot, file system resize should be done automatically.

So you might want to check this, so we can investigate why this didn't work in the first place. However of course optional, otherwise enjoy the now working image 😃.

I will mark this issue as solved, however if you want to debug the initial issue, feel free to reopen.

closed 2019-02-28 by MichaIng