MichaIng/DietPi #2461: DietPi-JustBoom + MPD settings improvement

MichaIng, 2019-01-29:
Ref: https://dietpi.com/phpbb/viewtopic.php?f=12&t=5270

MichaIng, 2021-06-02:
@maartenlangeveld
It's an old request, but actually not difficult to implement and reasonable. Basically your suggestion is to make all format settings optional, to not force conversion, right? We can add an "auto" option for frequency bit rate and also buffer size. This would then replace the respective format value with an asterisk (keep it untouched), or comment the whole setting if all is auto.

We currently also force two channels, which may be correct in usual cases (stereo music), but not in all cases. So I'd add a channel choice as well, to have 1 (mono), 2 (stereo), 6 (5.1), 8 (7.1) and auto.

Finally, I found a few other things that can be enhanced in DietPi-JustBoom, and actually it make sense to merge the DietPi-Config audio options inside and turn this into a "DietPi-Audio" program, being opened when selecting "Audio Option" from within DietPi-Config. What do you think? Means I found the little larger task for v7.3 development cycle 😄.

maartenlangeveld, 2021-06-02:
@MichaIng 
Would be great!

I have included mpd.conf below.
_note: (1) I use Upmpdcli as renderer and minimserver as music sever so I have disabled mpd music database, (2) I do not use volume control since I use volume control of my pre-amp/dac (3) have left all conversion out_

`2021-06-02 13:50:00 root@M-MUSIC-PLAYER:~# cat /etc/mpd.conf`
```
log_file                 "/var/log/mpd/mpd.log"
pid_file                 "/run/mpd/pid"
state_file               "/mnt/dietpi_userdata/.mpd_cache/state"

#user "mpd" # Set via systemd unit to preserve supplementary group permissions i.e. "dietpi" group

bind_to_address          "/run/mpd/socket"
bind_to_address          "localhost"

log_level "default"
#metadata_to_use "artist,composer,album,title,track,name,genre,date"

#disable zeroconf since upmpdcli is running as media renderer
zeroconf_enabled          "no"

connection_timeout        "30"
max_connections           "8"

filesystem_charset        "UTF-8"
audio_buffer_size         "2048"

input                     {
                           plugin "curl"
                          }

audio_output              {
                           type        "alsa"
                           name        "Justboom Digi Zero"
                           device      "default"
                           mixer_type  "none"
                           dop         "yes"
                          }

# to avoid mpd 'error' message that plugins are not available
input                     {
                           enabled    "no"
                           plugin     "qobuz"
                          }

input                     {
                           enabled      "no"
                           plugin       "tidal"
                          }
```

StephanStS, 2021-06-12:
I also propose to set "Auto conversion" to "On" by default.
In a facebook group I had the case that on a PI 400 (I reproduced it on a PI 4B as well) the HDMI sound output did not work with the setting set to "Off". Default "On" should make life easier.
[image]

MichaIng, 2021-06-12:
I'm still wondering if/what the downsides of this are. We didn't even have this option until quite recently. I added it when doing the last Mycroft AI implementation update where different sound sources are used for input and output so that it was required to have it running reliably on my test machines. It is included with the equalizer option, so users might have fixed it before by enabling the equalizer, but I'm a bit puzzled why we didn't receive much more reports already years ago. I don't think that the RPi onboard audio reduced format compatibility.

I'll read a bit more into this, and if the conversion plugin really is a no-op as long as the audio device supports playing the input stream without conversion, then it makes sense to have it enabled by default. Disabling it would then be a step for enthusiasts, to assure that the input stream is played natively, or otherwise adjust the input format so that it can be played without conversion (which may imply audio quality loss and raised CPU usage/noise). E.g. when using AirPlay, LMS or such, the stream is compressed or converted often server-side into a choosable format for transmission. It wouldn't make sense to convert it server-side into an incompatible format, just to convert it client-side into a compatible format when this can be done server-side directly. So what I want to say, it is not always bad when you run into such issue first, so you know that conversion is required, or, as alternative input format needs to be changed server-side, when possible.

closed 2021-10-20 by MichaIng