MichaIng/DietPi #2463: E: dpkg was interrupted, you must manually run 'dpkg --configure -a' to correct the problem.

luttman, 2019-01-29:
#### Details:
- Date           | Tue 29 Jan 03:49:14 CET 2019
- Bug report     | N/A
- DietPi version | v6.19.7 (Fourdee/master)
- Img creator    | n/a
- Pre-image      | n/a
- SBC device     | RPi 3 Model B+ (armv7l) (index=3)
- Kernel version | #1159 SMP Sun Nov 4 17:50:20 GMT 2018
- Distro         | stretch (index=4)
- Command        | G_AGUG
- Exit code      | 100
- Software title | DietPi-Update

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
E: dpkg was interrupted, you must manually run 'dpkg --configure -a' to correct the problem.
```

luttman, 2019-01-29:
root@Sonarr-Rpi:~# dpkg --configure -a
Setting up mono-roslyn (5.18.0.225-0xamarin1+raspbian9b1) ...
Mono precompiling /usr/lib/mono/4.5/csc.exe for arm...
Mono precompiling /usr/lib/mono/4.5/vbc.exe for arm...
Mono precompiling /usr/lib/mono/4.5/VBCSCompiler.exe for arm...
Mono precompiling /usr/lib/mono/4.5/Microsoft.CodeAnalysis.CSharp.dll for arm...
Mono precompiling /usr/lib/mono/4.5/Microsoft.CodeAnalysis.VisualBasic.dll for arm...
Mono precompiling /usr/lib/mono/4.5/Microsoft.CodeAnalysis.dll for arm...

────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────

Session stopped
    - Press <return> to exit tab
    - Press R to restart session
    - Press S to save terminal output to file

Network error: Software caused connection abort

MichaIng, 2019-01-29:
> Microsoft.CodeAnalysis.dll

Of course... 🤣 
No joke, so it's a mono update, compiling the new binaries that fails.

How long did the last printed compiling step take until it stopped?
Are you running this on local console or via SSH? Due to last lines I guess SSH?
_________
Based on my guess, if you can confirm the above:
- You are running the steps via SSH.
- For some reason, during mono compilation, the network connection breaks.
- This stops the SSH session and kills all related commands, which somehow fits to the initial log: `E: dpkg was interrupted, you must manually run 'dpkg --configure -a' to correct the problem.`
AFAIK when closing an SSH session, all running commands inside will be interrupted.

To solve (based on above guesses):
- Assure your network connection client and server side is stable, so your SSH session does not randomly close.
- At best retry on local console (via HDMI and keyboard on the Pi), so we can sort out SSH connection/interruption issues.

________________

If really the above is true and SSH disconnects is an issue, I would recommend you run SSH sessions in `screen`. This is a tool which opens a subshell allows to manage multiple "windows" in one SSH session. The great benefit is, that running commands inside are not tied to the SSH session. If if breaks, the commands will still run as children of the `screen` session. You can simply reconnect SSH and attach to the exact screen windows where the command was or still is running. Quite handy, actually something I wanted to natively integrate into DietPi for a long time.

Just for reference, here is the install/config steps from my own little HowTo document:
```
G_AGI screen

cat << _EOF_ >> /etc/screenrc
# ------------------------------------------------------------------------------
# DIETPI ADJUSTMENTS
# ------------------------------------------------------------------------------

# Disable startup message screen
startup_message off
# Make screen windows scrollable
termcapinfo xterm|xterms|xs|rxvt ti@:te@
# Add information status line to screen windows
caption always "%{Gk}DietPi Screen | \$LOGNAME@%H | %LD %d.%m.%Y %c | Load: %l | %-Lw%{kG}%n%f*%50> %t%{-}%+Lw%<"
_EOF_

cat << _EOF_ > /etc/bashrc.d/dietpi-screen.sh
# Autostart screen and auto logout:
[[ \$TERM != screen && \$SSH_CONNECTION ]] && screen -S sshscreen -d -R && logout
_EOF_
```

It configures screen to show an informational line at the bottom of the terminal window. The last command will auto login into a screen session when login via SSH. Local terminal sessions are not affected. When logging out/exit from SSH and re-login the same screen session will be recovered.

Here are some commands/shortcuts to manage screen sessions and windows:
```
	screen (-r)		//open (resume) session
	ctrl+a; c		//create window
	ctrl+a; space		//switch window
	ctrl+a; k		//close window
	ctrl+a; d		//detach session
	ctrl+d			//close session
```

A "session" is a screen process, which can contain multiple (tapped) windows, each comparable to a single TTY/console.

MichaIng, 2019-03-03:
Marking as closed due to missing user reply. Feel free to reopen if issue persists.

closed 2019-03-03 by MichaIng