MichaIng/DietPi #2465: Allo GUI | Users reporting "bricked" after update

Fourdee, 2019-01-29:


gahabana, 2019-01-29:
hi just did via SSH (as i always do it that way) ... update from 6.19.7 to 6.20.5
... after running it i get msg that it needs to reboot to get to 6.20.5 ... i reboot and upon login see it's still on 6.19.7 and msg in the terminal window:
`
/DietPi/dietpi/login: line 160: ((: > 0 : syntax error: operand expected (error token is "> 0 ")
`
started 'dietpi-update' again ... finished - rebooted - it launches window to install the sw for the firstime (have only roonbridge and github installation of latest shair-port) ... saying am i sure i want to continue with /minimal installation of diet-pi' ... answered yes, ... reboots again ...
after that reboot (3rd one) ... everything seems to be ok.n except that Shair-port seems to have disappeared - i may have installed it manually/maybe from github but dietpi update process has probably removed it.

but it is working :)

MichaIng, 2019-01-29:
Verified ramdisk issue in one case at least, solution: https://github.com/Fourdee/DietPi/issues/2466#issuecomment-458594631

MichaIng, 2019-01-29:
@gahabana 
I just pushed v6.20.6 which should resolve the issue. Please apply this update and report back if this resolves the issues.

MichaIng, 2019-01-29:
I will mark this as closed, however please ask here if there are any further issues with the update, so we will quickly re-investigate.

closed 2019-01-29 by MichaIng

boschr, 2019-01-30:
I'm new to DietPI. Installed it an hour ago on a Odroid C2 and facing exactly the same issue. Can't complete the fresh install. After the completed installation it will automatically reboot (seems normal to me). After reconnection due SSH it start the installation from the beginning.

**cat /DietPi/dietpi/.version**
```
6
14
```

**cat /var/tmp/dietpi/logs/dietpi-ramdisk.log**
```
/boot/dietpi/func/dietpi-ramdisk: line 27: cd: HOME not set
Mon 10 Sep 11:05:43 BST 2018 | DietPi-Ramdisk: Starting
Mon 10 Sep 11:05:43 BST 2018 | DietPi-Ramdisk: Completed
/boot/dietpi/func/dietpi-ramdisk: line 27: cd: HOME not set
Wed 30 Jan 20:03:11 GMT 2019 | DietPi-Ramdisk: Starting
Wed 30 Jan 20:03:11 GMT 2019 | DietPi-Ramdisk: Completed
/DietPi/dietpi/func/dietpi-ramdisk: line 27: cd: HOME not set
Wed 30 Jan 20:07:50 GMT 2019 | DietPi-Ramdisk: Stopping
cp: failed to access '/boot/': Host is down
Wed 30 Jan 20:07:50 GMT 2019 | DietPi-Ramdisk: Completed
/boot/dietpi/func/dietpi-ramdisk: line 27: cd: HOME not set
Wed 30 Jan 20:07:51 GMT 2019 | DietPi-Ramdisk: Starting
Wed 30 Jan 20:07:51 GMT 2019 | DietPi-Ramdisk: Completed
/DietPi/dietpi/func/dietpi-ramdisk: line 27: cd: HOME not set
Wed 30 Jan 20:10:06 GMT 2019 | DietPi-Ramdisk: Stopping
cp: failed to access '/boot/': Host is down
Wed 30 Jan 20:10:06 GMT 2019 | DietPi-Ramdisk: Completed
/boot/dietpi/func/dietpi-ramdisk: line 27: cd: HOME not set
Wed 30 Jan 20:10:07 GMT 2019 | DietPi-Ramdisk: Starting
Wed 30 Jan 20:10:07 GMT 2019 | DietPi-Ramdisk: Completed
/DietPi/dietpi/func/dietpi-ramdisk: line 27: cd: HOME not set
Wed 30 Jan 20:27:06 GMT 2019 | DietPi-Ramdisk: Stopping
cp: failed to access '/boot/': Host is down
Wed 30 Jan 20:27:06 GMT 2019 | DietPi-Ramdisk: Completed
/boot/dietpi/func/dietpi-ramdisk: line 27: cd: HOME not set
Wed 30 Jan 20:27:07 GMT 2019 | DietPi-Ramdisk: Starting
Wed 30 Jan 20:27:07 GMT 2019 | DietPi-Ramdisk: Completed
/DietPi/dietpi/func/dietpi-ramdisk: line 27: cd: HOME not set
Wed 30 Jan 20:30:40 GMT 2019 | DietPi-Ramdisk: Stopping
cp: failed to access '/boot/': Host is down
Wed 30 Jan 20:30:40 GMT 2019 | DietPi-Ramdisk: Completed
/boot/dietpi/func/dietpi-ramdisk: line 27: cd: HOME not set
Wed 30 Jan 20:30:42 GMT 2019 | DietPi-Ramdisk: Starting
Wed 30 Jan 20:30:43 GMT 2019 | DietPi-Ramdisk: Completed
Wed 30 Jan 20:55:18 GMT 2019 | DietPi-RAMdisk: Stopping...
cp: failed to access '/boot/': Host is down
Wed 30 Jan 20:55:18 GMT 2019 | DietPi-RAMdisk: Completed
/boot/dietpi/func/dietpi-ramdisk: line 27: cd: HOME not set
Wed 30 Jan 20:55:19 GMT 2019 | DietPi-Ramdisk: Starting
Wed 30 Jan 20:55:19 GMT 2019 | DietPi-Ramdisk: Completed
```