MichaIng/DietPi #2471: DietPi-Update | chmod: cannot access '/etc/profile.d/dietpi-*.sh' | v6.20.6

tenrek, 2019-01-29:
#### Details:
- Date           | Tue 29 Jan 22:46:37 CET 2019
- Bug report     | N/A
- DietPi version | v6.19.7 (Fourdee/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 2 Model B (armv7l) (index=2)
- Kernel version | #1159 SMP Sun Nov 4 17:50:20 GMT 2018
- Distro         | stretch (index=4)
- Command        | chmod -R +x /DietPi /var/lib/dietpi/services /etc/cron.daily/dietpi /etc/cron.hourly/dietpi /etc/profile.d/dietpi-*.sh /etc/bashrc.d/dietpi-bash_completion.sh /etc/bashrc.d/dietpi-login.sh /etc/bashrc.d/dietpi-posix-set-locale.sh /etc/bashrc.d/dietpi-ps1_kill_animation.sh /etc/bashrc.d/dietpi-unsupported_terminal.sh
- Exit code      | 1
- Software title | DietPi-Update

#### Steps to reproduce:
from v6.19.7 to v6.20.6

#### Additional logs:
```
Log file contents:
chmod: cannot access '/etc/profile.d/dietpi-*.sh': No such file or directory
```

MichaIng, 2019-01-29:
@tenrek 
Thanks for the report.

Does it work when simply rerunning `dietpi-update 1`?
Otherwise: `> /etc/profile.d/dietpi-unsupported_terminal.sh`

But the second solution is actually not good. This error means that for some strange reason, even that new DietPi code is laid out on the RAMdisk, even forced to `sync` (if disk async hold it back), still somehow the old DietPi-Update code is loaded.

If `dietpi-update 1` results in the same issue, can you paste output of:
`grep 'chmod -R +x' /DietPi/dietpi/dietpi-update`
So we can check that really the old code is still on disk. If this is the case, run `sync`, wait for some seconds and recheck the `grep` command above.

MichaIng, 2019-01-29:
@tenrek 
Okay that is good. Thanks for reporting back.

MichaIng, 2019-01-30:
I reopened, since we should fix this as well shortly.

Currently running `bash -x /DietPi/dietpi/dietpi-update 1` to follow the exact update path from v6.17.20 to v6.20.6...

MichaIng, 2019-01-30:
Next run:
> Run now to update DietPi (from v6.17.20 to v6.20.6).

`sed -i 's|/DietPi/dietpi/patch_file|bash -x /DietPi/dietpi/patch_file|g' /DietPi/dietpi/dietpi-update`

```
[ INFO ] DietPi-Update | Patching 6.17.20 to 6.18
+ INPUT=17
+ /DietPi/dietpi/func/dietpi-obtain_hw_model
+ . /DietPi/dietpi/func/dietpi-globals
++ [[ -z xterm ]]
++ [[ xterm == \u\n\k\n\o\w\n ]]
++ G_PROGRAM_NAME=
++ G_DEBUG=0
++ [[ -z '' ]]
++ G_USER_INPUTS=0
++ [[ -t 0 ]]
++ G_USER_INPUTS=1
++ G_DIETPI_INSTALL_STAGE=-2
++ [[ -f /DietPi/dietpi/.install_stage ]]
++ G_DIETPI_INSTALL_STAGE=1
++ G_GITBRANCH=master
++ G_GITOWNER=Fourdee
++ G_DIETPI_VERSION_CORE=6
++ G_DIETPI_VERSION_SUB=20
++ G_DIETPI_VERSION_RC=0
++ [[ -f /DietPi/dietpi/.version ]]
+++ sed -n 1p /DietPi/dietpi/.version
++ [[ #!/bin/bash == \#\!\/\b\i\n\/\b\a\s\h ]]
++ . /DietPi/dietpi/.version
+++ G_DIETPI_VERSION_CORE=6
+++ G_DIETPI_VERSION_SUB=17
+++ G_DIETPI_VERSION_RC=20
+++ G_GITBRANCH=master
+++ G_GITOWNER=Fourdee
++ G_FP_DIETPI_USERDATA=/mnt/dietpi_userdata
++ G_DIETPI_SERVICES_DISABLE=0
++ G_HW_MODEL=0
++ G_HW_MODEL_DESCRIPTION=NULL
++ G_HW_ARCH=0
+++ uname -m
++ G_HW_ARCH_DESCRIPTION=x86_64
++ G_HW_CPUID=0
+++ nproc --all
++ G_HW_CPU_CORES=4
++ G_DISTRO=0
++ G_DISTRO_NAME=NULL
++ [[ -f /DietPi/dietpi/.hw_model ]]
+++ sed -n 1p /DietPi/dietpi/.hw_model
++ G_HW_MODEL=20
+++ sed -n 2p /DietPi/dietpi/.hw_model
++ G_HW_MODEL_DESCRIPTION='Virtual Machine (x86_64)'
+++ sed -n 6p /DietPi/dietpi/.hw_model
++ G_HW_ARCH=10
+++ sed -n 9p /DietPi/dietpi/.hw_model
++ G_HW_CPUID=0
+++ sed -n 3p /DietPi/dietpi/.hw_model
++ G_DISTRO=4
++ ((  4 == 3  ))
++ ((  4 == 4  ))
++ G_DISTRO_NAME=stretch
++ G_INIT_ALLOW_CONCURRENT=0
++ G_INIT_WAIT_CONCURRENT=5
++ G_CHECK_ROOT_USER_VERIFIED=1
++ G_CHECK_ROOTFS_RW_VERIFIED=1
++ alias 'sudo=sudo '
++ alias dietpi-process_tool=/DietPi/dietpi/dietpi-process_tool
++ alias dietpi-letsencrypt=/DietPi/dietpi/dietpi-letsencrypt
++ alias dietpi-autostart=/DietPi/dietpi/dietpi-autostart
++ alias dietpi-cron=/DietPi/dietpi/dietpi-cron
[... 297 lines cut ...]
Created symlink /etc/systemd/system/reboot.target.wants/dietpi-kill_ssh.service → /etc/systemd/system/dietpi-kill_ssh.service.
[  OK  ] DietPi-Patchfile | Desired setting in /DietPi/dietpi/.installed was already set: aSOFTWARE_INSTALL_STATE[116]=0
[  OK  ] DietPi-Patchfile | APT installation for: procps, please wait...

[  OK  ] DietPi-Patchfile | G_AGI: procps
[  OK  ] DietPi-Update | Patching to v6.20.6 completed

[ INFO ] DietPi-Survey | Lines required for Whiptail box and its contents (y=38, z=2), exceeds screen dimensions (y=20).
[ SUB2 ] DietPi-Update > Completed
[ INFO ] DietPi-Update | Current version : v6.20.6
[ INFO ] DietPi-Update | Latest version  : v6.20.6
[ SUB3 ] DietPi-Services > restart
[  OK  ] DietPi-Services | restart : cron
[ SUB4 ] DietPi-Process_tool > Apply
[  OK  ] DietPi-Process_tool | Completed
+ kill 1444
+ exit
+ G_EXIT
+ cd /tmp
+ ((  0 == 1  ))
+ ((  ! 0  ))
+ [[ -d /tmp/DietPi-Patchfile ]]
+ declare -F G_EXIT_CUSTOM
Terminated
```

MichaIng, 2019-01-30:
Everything exactly as intended:
- Patch finds `.update_stage` and `.install_stage == 1`, thus restarts update
- Nested update contains **new** code, can be seen from some tiny changed wording.
- Nested update starts incremental patches, which then go though.
- `[ INFO ] DietPi-Update | Patching 6.19.6 to 6.20` After this, `/etc/profile.d/dietpi-unsupported_terminal.sh` is removed
- Update terminates after nested update + incremental patches have finished.
- No chance to run into above error, even if old update code would be executed, since termination prevents another update start after the related file is removed 🤔...

Everything as it should be:
```
root@DietPi:~# cat /DietPi/dietpi/.update_stage /DietPi/dietpi/.install_stage
cat: /DietPi/dietpi/.update_stage: No such file or directory
2
```

Akito13, 2019-01-30:
> Sorry to bother you guys.

It's the opposite of a bother, to help a dedicated developer solve a problem in his work.

> Does anyone of you have/had one of these software titles installed, and were you asked to create a backup before reinstalling them? Or did this even happen twice?

I didn't install any of those manually. I can't guarantee for any automatic install, but I doubt these are remotely dependencies at all, for anything I have installed.

> And for completeness, could you paste the output of cat /tmp/dietpi-update/dietpi-update.log when the error prompt is active, e.g. from a second SSH session? Of course only e.g. if you reverted to v6.19.7 and did not yet finish the update via workaround.

I can try to re-run the reversion, as well as the following upgrade in a simulated environment and post the results.

MichaIng, 2019-01-30:
@Akito13 
> I can try to re-run the reversion, as well as the following upgrade in a simulated environment and post the results.

That would be perfect, although I know you already invested much time with this update trouble, so would be totally acceptable it you simply want to enjoy a running system now.

Okay lets move all conversation here, especially since the issue header matches the error message, so other users might find this faster.

Akito13, 2019-01-31:
@MichaIng I restored a complete system backup into a docker container, but the update worked flawlessly. The only thing that came to my mind when thinking about what warnings and errors I got on the original system is that DietPi thought I have the full LLSP stack, but it somehow failed to install/upgrade and detect the PHP-7 packages, as far as I recall. In the docker experiment, this part worked flawlessly as opposed to the original scenario.

Akito13, 2019-01-31:
@MichaIng I have some news.

I performed an update from v6.19.7 to v6.20.6 on a Raspberry Pi 3B, that has more or less the same content like the Raspberry Pi 0W I was talking about. The update was processed without any errors. When it prompted to reboot the system, I confirmed the restart. After the restart, I could see the Raspi being connected to my router but was constantly getting `No router to host` ssh error, when trying to connect. Using the local IP or ssh instead of mosh, didn't change anything.
I then turned off the power supply and turned it on again, basically doing a hard reboot. I could ssh into the Pi to discover something very interesting:
[image]
I was welcomed with this screen after _updating_ from v6.19.7 to v6.20.6 .

Now I ran `dietpi-update` once again, encountering the initial error from the OP. I ssh'ed and copied the `dietpi-update.log` file to my computer. `nano` said, that the file has to be converted from Mac format, probably because of the line length excess. These are the contents of the log file:

```

[0m[90m[[0m[33m .... [0m[90m][0m [90mDietPi-Update | [0mwget https://raw.githubusercontent.com/Fourdee/DietPi/master/dietpi/pre-patch_file -O pre-patch_file[0m[1A
[0m[90m[[0m[1;33m.     [0m[90m][0m 
[0m[90m[[0m[0;33m.[1;33m.    [0m[90m][0m [J
[K[0m[90m[[0m[32m  OK  [0m[90m][0m [90mDietPi-Update | [0mwget https://raw.githubusercontent.com/Fourdee/DietPi/master/dietpi/pre-patch_file -O pre-patch_file
[0m
 [1mDietPi-Pre-patch[0m
[90m─────────────────────────────────────────────────────[0m
[ [1mINFO[0m ] Applying critical pre-patches

 [1mDietPi-Pre-patch[0m
[90m─────────────────────────────────────────────────────[0m
[ [1mINFO[0m ] Completed pre-patches with exit code: [33m0[0m

[K[0m[90m[[0m[32m  OK  [0m[90m][0m [90mDietPi-Update | [0mAPT update, please wait...
[0m[90mHit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Hit:2 https://archive.raspberrypi.org/debian stretch InRelease
Reading package lists...
[0m

[K[0m[90m[[0m[32m  OK  [0m[90m][0m [90mDietPi-Update | [0mG_AGUP
[0m
[K[0m[90m[[0m[32m  OK  [0m[90m][0m [90mDietPi-Update | [0mAPT upgrade, please wait...
[0m[90mReading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
[0m

[K[0m[90m[[0m[32m  OK  [0m[90m][0m [90mDietPi-Update | [0mG_AGUG
[0m  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100   119    0   119    0     0    320      0 --:--:-- --:--:-- --:--:--   321

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
 30 2493k   30  759k    0     0   436k      0  0:00:05  0:00:01  0:00:04  793k
100 2493k  100 2493k    0     0  1188k      0  0:00:02  0:00:02 --:--:-- 1897k

[0m[90m[[0m[33m .... [0m[90m][0m [90mDietPi-Update | [0mUnpack update archive[0m
[0m[90m[[0m[1;33m.     [0m[90m][0m 
[0m[90m[[0m[0;33m.[1;33m.    [0m[90m][0m 
[0m[90m[[0m [0;33m.[1;33m.   [0m[90m][0m 
[0m[90m[[0m  [0;33m.[1;33m.  [0m[90m][0m [J
[K[0m[90m[[0m[32m  OK  [0m[90m][0m [90mDietPi-Update | [0mUnpack update archive
[0m
[0m[90m[[0m[33m .... [0m[90m][0m [90mDietPi-Update | [0mCopy DietPi core files to RAMdisk[0m
[0m[90m[[0m[1;33m.     [0m[90m][0m [J
[K[0m[90m[[0m[32m  OK  [0m[90m][0m [90mDietPi-Update | [0mCopy DietPi core files to RAMdisk
[0m
[0m[90m[[0m[33m .... [0m[90m][0m [90mDietPi-Update | [0mCopy rootfs files in place[0m
[0m[90m[[0m[1;33m.     [0m[90m][0m 
[0m[90m[[0m[0;33m.[1;33m.    [0m[90m][0m 
[0m[90m[[0m [0;33m.[1;33m.   [0m[90m][0m 
[0m[90m[[0m  [0;33m.[1;33m.  [0m[90m][0m 
[0m[90m[[0m   [0;33m.[1;33m. [0m[90m][0m 
[0m[90m[[0m    [0;33m.[1;33m.[0m[90m][0m [J
[K[0m[90m[[0m[32m  OK  [0m[90m][0m [90mDietPi-Update | [0mCopy rootfs files in place
[0m
[0m[90m[[0m[33m .... [0m[90m][0m [90mDietPi-Update | [0mSet execute permissions for DietPi scripts[0m
[0m[90m[[0m[1;33m.     [0m[90m][0m [J
[K[0m[90m[[0m[31mFAILED[0m[90m][0m [90mDietPi-Update | [0mchmod -R +x /DietPi /var/lib/dietpi/services /etc/cron.daily/dietpi /etc/cron.hourly/dietpi /etc/profile.d/dietpi-*.sh /etc/bashrc.d/dietpi-bash_completion.sh /etc/bashrc.d/dietpi-login.sh /etc/bashrc.d/dietpi-posix-set-locale.sh /etc/bashrc.d/dietpi-ps1_kill_animation.sh /etc/bashrc.d/dietpi-unsupported_terminal.sh
[0m[?1049h[1;30r[4l[?25l(B[m[37m[40m[1;30r[H[2J[1;1H[1m[37m[44m                                                                                                                        [2;1H                                                                                                                        [3;1H                                                                                                                        [4;1H                                                                                                                        [5;1H                                                                                                                        [6;1H                                                                           [... cut ...] 40m [25;16H(B[m[30m[47m│                        <Ok>                            <Exit>                           [1m[37m[40m [26;16H(B[m[30m[47m│                                                                                        │[1m[37m[40m [27;16H(B[m[30m[47m└────────────────────────────────────────────────────────────────────────────────────────┘[1m[37m[40m [28;17H                                                                                          [5;18H
```

rtwx, 2019-02-01:
Workaround in OP worked for me, thank you.

axwax, 2019-02-01:
I also ran into the OP's symptoms when I tried dietpi-update for the second time after my first update from v6.17.12 resulted in the error 
`/DietPi/dietpi/login: line 150: ((: > 0 : syntax error: operand expected (error token is "> 0 ")`
after reboot.
The workaround in OP worked for me too!

MichaIng, 2019-02-04:
@Fourdee 
I was reading the `x-systemd.automount` issue again: https://github.com/Fourdee/DietPi/pull/2357#issuecomment-449177715

Actually already there I recognized that (re)adding `After=boot.mount` alone did not resolve the issue with RAMdisk stopping too early since `boot.automount` is the active one.

But actually when checking systemd unit order/dependencies, automounts are started even before the actual mounts, so stopped afterwards. And all are bound `Before=local-fs.target`:
```
2019-02-04 00:59:40 root@micha:~# systemctl show -p Requires,Wants,WantedBy,Requisite,Before,After mnt-sda.automount
Requires=-.mount
Requisite=
Wants=
WantedBy=local-fs.target
Before=local-fs.target umount.target mnt-sda.mount
After=-.mount
2019-02-04 01:00:00 root@micha:~# systemctl show -p Requires,Wants,WantedBy,Requisite,Before,After mnt-sda.mount
Requires=-.mount system.slice
Requisite=
Wants=
WantedBy=
Before=local-fs.target umount.target mnt-sda-.swapfile.swap
After=-.mount dev-disk-by\x2duuid-192978f5\x2d3451\x2d45ce\x2db00b\x2de5e96738ddeb.device systemd-journald.socket dev-sda.device system.slice mnt-sda.automount local-fs-pre.target
```
_________
🈴 Ah yep on Stretch indeed this is different (now):
```
root@VM-Stretch:~# systemctl show -p Requires,Wants,WantedBy,Before,After mnt-sdb.mount
Requires=system.slice -.mount
Wants=
WantedBy=
Before=umount.target
After=mnt-sdb.automount system.slice local-fs-pre.target -.mount dev-disk-by\x5cx2duuid-1AFB\x5cx2d90C2.device
root@VM-Stretch:~# systemctl show -p Requires,Wants,WantedBy,Before,After mnt-sdb.automount
Requires=-.mount
Wants=
WantedBy=local-fs.target
Before=umount.target local-fs.target mnt-sdb.mount
After=-.mount
```
- This MUST have changed with recent systemd update, otherwise DietPi-RAMdisk would have never worked.
- Only the `automount` is pulled in on boot (`WantedBy=local-fs.target`), the mount itself is not pulled by anything, only triggered by first access.
- Since `boot.mount` for this reason on pre-v6.19 images is not part of the boot order, `Before/After=boot.mount` is completely without any effect (on boot and shutdown order). Only `boot.automount` would have.

Actually these automounts do not make any sense if we anyway set `auto` option, or the other way round, automount implies `noauto` and overrides `auto`, so removes the mounts from boot order, as can be seen above. I think we should remove `x-systemd.automount` completely from all `dietpi-drive_manager` mounts. I anyway have no idea why this should be a benefit. If the drive is mounted anyway on first touch, having it mounted before or on trigger makes no difference. Energy savings/spin up/down is triggered by I/O, not mount, since mount is a UNIX system thing only, AFAIK.
Automounts complicate and double the systemd generator effort and makes things more complicated, obviously, especially when rules change, being different on Stretch and Buster now etc.

If required/requested we could add an optional `noauto,x-systemd.automount` choice to drive manager later, for non boot/root/userdata/swap drives only.
___________
@Fourdee 
I am still not able to explain all user reports, but worth to try?
Also we should remove `.update_stage` and check/adjust `.install_stage` separately, e.g. `[[ -f /DietPi/dietpi/.installed ]] && echo 2 > /DietPi/dietpi/.install_stage` independently from `.update_stage` existence, just to be failsafe and perhaps fix some systems where the update failed or was not 100% finished.
And before restarting update or reboot, do `systemctl restart dietpi-ramdisk` as well.
Perhaps do this as v6.21 release this time, so users, where v6.20 was successful, don't need to redo reinstalls, which can take long (e.g. Homeassistant) or require config file recovery (MPD)?

MichaIng, 2019-02-06:
@Akito13 
Another question:
- Do you remember, if the `chmod: cannot access '/etc/profile.d/dietpi-*.sh` error appeared after a reboot the first time, or directly after starting the update the first time (without reboot in between)?

Akito13, 2019-02-06:
> 
> 
> @Akito13
> Another question:
> 
>     * Do you remember, if the `chmod: cannot access '/etc/profile.d/dietpi-*.sh` error appeared after a reboot the first time, or directly after starting the update the first time (without reboot in between)?

I didn't use the test-pi that encountered the error since then, to use it for further testing. I ssh'ed into it now seeing this:
```
[FAILED] DietPi-Software | Root privileges required. Please run the command with "G_SUDO", or, "sudo -i".
mv: failed to access '/var/tmp/dietpi/logs/dietpi-firstrun-setup.log': Permission denied
```
Which means, that the error occured before rebooting.

Fourdee, 2019-02-07:
Resolved with v6.21 update, thanks @MichaIng 👍

closed 2019-02-07 by Fourdee

Akito13, 2019-02-07:
Thank you, everyone, good job.

toporchelo, 2019-03-27:
### **HELP!**

I can not upgrade to version from version 6.21.1 to version 6.22.3 version
[image]

[image]

MichaIng, 2019-03-27:
@toporchelo 
Please paste the full output of:
`apt -y upgrade`

Akito13, 2019-03-27:
My certainly necessary 2 cents:
DietPi update worked for me, even from an older version.

MichaIng, 2019-03-27:
@Akito13 
Jep, only users which tried to update to v6.20 one time (and failed) could run into original issue from first post.

toporchelo has a different issue related to the APT package upgrades, but the important last output lines are missing.

Akito13, 2019-03-27:
Well, seems like I didn't read the error message correctly, at first. But now I did.

@toporchelo Do you have apache2 installed or maybe another web server running on Port 80? If there is, then nginx-full will always fail, as it can't allocate Port 80 for itself.

MichaIng, 2019-03-27:
@Akito13 
Since it is not install but only an APT upgrade, nginx was already installed before. Lets wait for the full error output before guessing 😉.