MichaIng/DietPi #2472: Please do not overwrite mpd.conf with every update!

maartenlangeveld, 2019-01-30:
Just updated from v6.20.5 to v6.20,6 my rpi-zero-w music streamer with Justboom Zero HAT. The faulty mpd.conf (that I edited manually in order to work correctly) gets again overwritten.... argh

MichaIng, 2019-01-30:
@maartenlangeveld 
But there was a backup of your manually adjusted `mpd.conf`, wasn't it?

MichaIng, 2019-01-30:
@maartenlangeveld 
Okay good to know. I am a bid wondering how this is solved with the default Debian package, where MPD runs as non-root as well. But perhaps is is there an issue as well.

Yeah about the config file overwriting, in many cases we handle this already different now, preserve existing settings and only inject single settings that are required. However in case of `mpd.conf` it is a bid difficult, due to the complexity of the file and it's syntax. It is impossible (or high effort with high failure chance) to apply all required settings just via `sed` and such. If we leave it completely untouched, on fresh install users face issues as well. so we apply the default config, which should work well in most cases and instead create the backup, in case users made manual adjustments. But yeah the root requirement needs to be investigated, I already opened another issue about this from earlier report: https://github.com/Fourdee/DietPi/issues/2462

Fourdee, 2019-02-02:
```G_BACKUP_FP``` rolled out for MPD. Will mark this as closed.

closed 2019-02-02 by Fourdee