MichaIng/DietPi #2474: WireGuard not running on RPi 1

WilburWalsh, 2019-01-30:
#### Required Information
- DietPi version | 6.20.5
- Distro version | stretch
- SBC device | RPi B (armv6l)

#### Additional Information (if applicable)
- Software title | WireGuard
- Was the software title installed freshly or updated/migrated? Fresh Install
- Can this issue be replicated on a fresh installation of DietPi? Probably

#### Steps to reproduce
1. Select WireGuard to install
2. Start installation

#### Expected behaviour
- Wireguard should be installed

#### Actual behaviour
- Errors during Setup process due to failing 'wg' command (wg throws Segmentation Fault)

#### Extra details
- The Debian Wireguard binaries are apparently not running on armv6l, see for example https://github.com/adrianmihalko/raspberrypiwireguard/wiki/Install-WireGuard-on-Raspberry-Pi-1,-2-(not-v1.2),-Zero,-Zero-W

I compiled it from source, and could then repeat the failed steps. Could not test the running server however, because I faced some problems with my network and switched back to 6.19.7 in the meantime. So you might consider removing WireGuard from the software list for affected Pi models or change the setup to compile from source (if you support this kind of installation).
I also noticed a little bug in the creation of wg0-client.conf, where you use the standard port of 51820 instead of the port number specified in the dialog.

On another note, I will probably soon upgrade to a newer Pi for a couple of reasons and would then gladly provide the RPi1 and some of my time for testing or any other support.

Fourdee, 2019-01-30:
@WilburWalsh 

> The Debian Wireguard binaries are apparently not running on armv6l, see for example 

Many thanks for the report, yep, appears ARMv7+ only.

> I also noticed a little bug in the creation of wg0-client.conf, where you use the standard port of 51820 instead of the port number specified in the dialog.

Thanks, i'll check this.

MichaIng, 2019-01-30:
@WilburWalsh @Fourdee 
Thanks for the hint with the port, jep that line needs to be `Endpoint = $domain:$port` of course: https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-software#L10068

No config, when selecting client mode is expected. In very most cases the VPN provider will ship a config, if it is another DietPi, it creates `wg0-client.conf` there. There is no way we can create a reasonable client config client side without knowing anything about the server of course, so leave this step to user. The prompts and online docs should contain all general info we can give.

Fourdee, 2019-01-31:
@MichaIng 

> Ah @Fourdee that fix in wrong place

Whoopsie, thanks Micha 😄 👍

Fourdee, 2019-01-31:
@MichaIng 

> No config, when selecting client mode is expected. In very most cases the VPN provider will ship a config, if it is another DietPi, it creates wg0-client.conf

👍 Excellent, as ARMv6 is now disabled due to no official binaries, and, server port applied correctly to ```wg0-client.conf``` I believe we can now mark this as resolved?

closed 2019-01-31 by Fourdee

MichaIng, 2019-01-31:
Jep should be fine.