MichaIng/DietPi #2475: error prompt: No space left on device

msongz, 2019-01-30:
during and after update from 6.17 to 6.20 , the pi always prompt me this:

Failed to add /run/systemd/ask-password to directory watch: No space left on device

but it is 32gb storage left.

but the update is a success after two reboots finally.

MichaIng, 2019-01-30:
@msongz 
Yeah sorry there is an issue with the current RPi images. Those will be updated shortly. But you can easily fix this now:
```
systemctl enable dietpi-fs_partition_resize
reboot
```

msongz, 2019-01-30:
@MichaIng thank you for your support!

msongz, 2019-01-30:
`systemctl restart node-red.service`                                                                          
`Failed to add /run/systemd/ask-password to directory watch: No space left on device`

@MichaIng  sorry, reopen this. follow your solution the command above will also prompt the same thing.

MichaIng, 2019-01-30:
Can you check the scripts log: `cat /var/tmp/dietpi/logs/fs_partition_resize.log`

msongz, 2019-01-31:
`Removed /etc/systemd/system/local-fs.target.wants/dietpi-fs_partition_resize.service.
Disk /dev/mmcblk0: 59.6 GiB, 64021856256 bytes, 125042688 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x27504eef

Old situation:

Device         Boot Start       End   Sectors  Size Id Type
/dev/mmcblk0p1       8192     93802     85611 41.8M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      98304 125042687 124944384 59.6G 83 Linux

/dev/mmcblk0p2:
New situation:

Device         Boot Start       End   Sectors  Size Id Type
/dev/mmcblk0p1       8192     93802     85611 41.8M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      98304 125042687 124944384 59.6G 83 Linux

The partition table has been altered.
Calling ioctl() to re-read partition table.
Re-reading the partition table failed.: Device or resource busy
The kernel still uses the old table. The new table will be used at the next reboot or after you run partprobe(8) or kpartx(8).
Syncing disks.
resize2fs 1.43.4 (31-Jan-2017)
The filesystem is already 15618048 (4k) blocks long.  Nothing to do!`

Fourdee, 2019-01-31:
REF: update RPi image: https://github.com/Fourdee/DietPi/issues/2026

Fourdee, 2019-01-31:
@msongz 

> /run

Is mounted to tmpfs. Seems you might be running out of memory on the device, or, ```/run``` is filling up from a program.

### 1: Check used space in ```run```
```
G_TREESIZE /run
```

### 2: Also try increasing the swapfile:
```
dietpi-services stop
/DietPi/dietpi/func/dietpi-set_dphys-swapfile /var/swap 2000
dietpi-services restart
```

msongz, 2019-01-31:
G_TREESIZE /run                                                                                                                     ✔  ⚡  412  21:18:30
```
44.8 MB /run
43.8 MB /run/log
420.0 KB /run/samba-cache
332.0 KB /run/samba
148.0 KB /run/udev
68.0 KB /run/systemd
12.0 KB /run/network
8.0 KB /run/resolvconf
4.0 KB /run/tmpfiles.d
4.0 KB /run/redis
4.0 KB /run/php
4.0 KB /run/mysqld
4.0 KB /run/blkid
4.0 KB /run/avahi-daemon
0.0 KB /run/user
0.0 KB /run/sshd
0.0 KB /run/sendsigs.omit.d
0.0 KB /run/screen
0.0 KB /run/mount
0.0 KB /run/lock
0.0 KB /run/lighttpd
0.0 KB /run/dbus
0.0 KB /run/console-setup
```
```
  Mem[|||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||349M/876M]
  Swp[|                                                                 512K/1.05G]
```

```dietpi-services stop

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: stop

[  OK  ] DietPi-Services | ncc maintenance:mode --on
[.     ] Failed to add /run/systemd/ask-password to directory watch: No space left on device
[  OK  ] DietPi-Services | stop : cron
[.     ] Failed to add /run/systemd/ask-password to directory watch: No space left on device
[  OK  ] DietPi-Services | stop : node-red
[.     ] Failed to add /run/systemd/ask-password to directory watch: No space left on device
[  OK  ] DietPi-Services | stop : gitea
[.     ] Failed to add /run/systemd/ask-password to directory watch: No space left on device
[  OK  ] DietPi-Services | stop : aria2
[.     ] Failed to add /run/systemd/ask-password to directory watch: No space left on device
[  OK  ] DietPi-Services | stop : lighttpd
[.     ] Failed to add /run/systemd/ask-password to directory watch: No space left on device
[  OK  ] DietPi-Services | stop : php7.0-fpm
[.     ] Failed to add /run/systemd/ask-password to directory watch: No space left on device
[  OK  ] DietPi-Services | stop : mariadb
[.     ] Failed to add /run/systemd/ask-password to directory watch: No space left on device
[  OK  ] DietPi-Services | stop : redis-server
[.     ] Failed to add /run/systemd/ask-password to directory watch: No space left on device
[  OK  ] DietPi-Services | stop : smbd
[.     ] Failed to add /run/systemd/ask-password to directory watch: No space left on device
[  OK  ] DietPi-Services | stop : nmbd
[.     ] Failed to add /run/systemd/ask-password to directory watch: No space left on device
[  OK  ] DietPi-Services | stop : avahi-daemon
 root@pi2  ~  /DietPi/dietpi/func/dietpi-set_dphys-swapfile /var/swap 2000                                                                        ✔  ⚡  415  21:20:49

 DietPi-Swapfile
─────────────────────────────────────────────────────
 Mode: Apply

[  OK  ] DietPi-Swapfile | Disable swapfile
[  OK  ] DietPi-Swapfile | swapoff -a
[ INFO ] DietPi-Swapfile | Deleting existing swapfile (/var/swap)
[  OK  ] DietPi-Swapfile | Setting in /DietPi/dietpi.txt adjusted: AUTO_SETUP_SWAPFILE_SIZE=0
[  OK  ] DietPi-Swapfile | Desired setting in /DietPi/dietpi.txt was already set: AUTO_SETUP_SWAPFILE_LOCATION=/var/swap
[ INFO ] DietPi-Swapfile | Setting /tmp tmpfs size: 437M
[  OK  ] DietPi-Swapfile | mount -o remount,size=437M tmpfs /tmp
[FAILED] DietPi-Swapfile | Free space check: path=2000 | available=437 MB | required=1072 MB
[FAILED] DietPi-Swapfile | Unable to generate swapfile due to insufficient disk space. Swap has be disabled.
```
@Fourdee

Fourdee, 2019-02-06:
@msongz 

Apologies for the delay.
Please check max size of ```run``` by using:
- ```df -h```

> Free space check: path=2000

Appears incorrect, also check ```dmesg``` for any undervolt/filesystem issues:
- ```dmesg | grep volt```
- ```dmesg | grep ext```

I suspect system instability/hardware issue has caused the system to become unstable. Will wait for the above results.
Also please confirm SD card and PSU used on system.

msongz, 2019-02-12:
```
root@pi2  ~  df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        59G   28G   32G  47% /
devtmpfs        434M     0  434M   0% /dev
tmpfs           438M  4.0K  438M   1% /dev/shm
tmpfs           438M   45M  393M  11% /run
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           438M     0  438M   0% /sys/fs/cgroup
tmpfs           437M     0  437M   0% /tmp
tmpfs            10M  1.4M  8.7M  14% /DietPi
tmpfs            50M   12K   50M   1% /var/log
/dev/mmcblk0p1   42M   24M   18M  57% /boot
```

``` 
root@pi2  ~  dmesg | grep volt
[    2.152401] Under-voltage detected! (0x00050005)
[   10.472354] Under-voltage detected! (0x00050005)
[   85.352293] Under-voltage detected! (0x00050005)
[  628.232273] Under-voltage detected! (0x00050005)
[  642.792261] Under-voltage detected! (0x00050005)
[ 1241.832286] Under-voltage detected! (0x00050005)
[ 2909.992283] Under-voltage detected! (0x00050005)
[12405.192339] Under-voltage detected! (0x00050005)
[21652.872391] Under-voltage detected! (0x00050005)
[27485.192358] Under-voltage detected! (0x00050005)
[33417.352390] Under-voltage detected! (0x00050005)
[36462.472513] Under-voltage detected! (0x00050005)
[36587.272530] Under-voltage detected! (0x00050005)
[36884.712555] Under-voltage detected! (0x00050005)
[37030.312395] Under-voltage detected! (0x00050005)
[37061.512386] Under-voltage detected! (0x00050005)
[37186.312468] Under-voltage detected! (0x00050005)
[37286.152423] Under-voltage detected! (0x00050005)
[37361.032384] Under-voltage detected! (0x00050005)
[37487.912415] Under-voltage detected! (0x00050005)
[37650.152401] Under-voltage detected! (0x00050005)
[37660.552405] Under-voltage detected! (0x00050005)
[39761.352494] Under-voltage detected! (0x00050005)
[40060.872386] Under-voltage detected! (0x00050005)
[40487.272414] Under-voltage detected! (0x00050005)
[40661.992406] Under-voltage detected! (0x00050005)
[40741.032558] Under-voltage detected! (0x00050005)
[40961.512419] Under-voltage detected! (0x00050005)
[40969.832440] Under-voltage detected! (0x00050005)
[40984.392459] Under-voltage detected! (0x00050005)
[43686.322468] Under-voltage detected! (0x00050005)
[43698.792710] Under-voltage detected! (0x00050005)
[43709.193924] Under-voltage detected! (0x00050005)
[44046.152447] Under-voltage detected! (0x00050005)
[44064.872496] Under-voltage detected! (0x00050005)
[44260.392902] Under-voltage detected! (0x00050005)
[45462.632416] Under-voltage detected! (0x00050005)
[46488.072420] Under-voltage detected! (0x00050005)
[46787.592416] Under-voltage detected! (0x00050005)
[47736.072477] Under-voltage detected! (0x00050005)
[47860.872444] Under-voltage detected! (0x00050005)
[48108.392449] Under-voltage detected! (0x00050005)
[48407.912430] Under-voltage detected! (0x00050005)
[48707.432434] Under-voltage detected! (0x00050005)
[48757.352447] Under-voltage detected! (0x00050005)
[48882.152493] Under-voltage detected! (0x00050005)
```

```
 root@pi2  ~  dmesg | grep ext
[    0.000000] Kernel command line: bcm2708_fb.fbwidth=800 bcm2708_fb.fbheight=480 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=PARTUUID=27504eef-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait plymouth.enable=0
                     .text : 0x80008000 - 0x80800000   (8160 kB)
[    0.009493] CPU: Virtualization extensions available.
[    1.103648] VFS: Mounted root (ext4 filesystem) readonly on device 179:2.
```

MichaIng, 2019-02-12:
@msongz 
Although not 100% sure if it's the reason, at least as you can see you need a more stable power supply. You could try to run some quick check, e.g. use a different power supply or a micro USB - USB adapter and plug the Pi to some notebook/desktops USB3.0 port. Check if the `Under-voltage` messages go away and hopefully as well the "no space left" issue.

Fourdee, 2019-02-12:
@msongz 

As suspected, system instability due to PSU is most likely the cause of the issues you are experiencing. 

I'd highly recommend purchasing an official RPi PSU 5v/2A, then do a fresh image installation of DietPi, to ensure a stable experience. 

I'll mark this as closed due to hardware related issue which is outside our support.

closed 2019-02-12 by Fourdee