MichaIng/DietPi #2477: Diet-Pi

rjjrdadev, 2019-01-30:
#### Details:
- Date           | Wed 30 Jan 11:24:47 EST 2019
- Bug report     | N/A
- DietPi version | v6.19.7 (Fourdee/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 2 Model B (armv7l) (index=2)
- Kernel version | #1159 SMP Sun Nov 4 17:50:20 GMT 2018
- Distro         | stretch (index=4)
- Command        | chmod -R +x /DietPi /var/lib/dietpi/services /etc/cron.daily/dietpi /etc/cron.hourly/dietpi /etc/profile.d/dietpi-*.sh /etc/bashrc.d/dietpi-bash_completion.sh /etc/bashrc.d/dietpi-login.sh /etc/bashrc.d/dietpi-posix-set-locale.sh /etc/bashrc.d/dietpi-ps1_kill_animation.sh /etc/bashrc.d/dietpi-unsupported_terminal.sh
- Exit code      | 1
- Software title | DietPi-Update

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
chmod: cannot access '/etc/profile.d/dietpi-*.sh': No such file or directory
```

Akito13, 2019-01-30:
I just wanted to post this about this error right now...

Friend, all you have to do is the following:

```
sudo touch /etc/profile.d/dietpi-placeholder.sh
sudo dietpi-update
```
I solved the issue like that!

rjjrdadev, 2019-01-30:
Thank you! That solved the issue.

Akito13, 2019-01-30:
> 
> 
> Thank you! That solved the issue.

It's a workaround. 😁

MichaIng, 2019-01-30:
@rjjrdadev @Akito13 @Fourdee 
Thanks for your report.

Jep this is a known, although uncritical issue with the update to v6.20. We are not yet sure about how this can occur, but at least it is an indicator that the old (v6.19) updater code is executed, even that the new (v6.20) should be used.
The reason must be related to disk `async`, a default Linux feature that does not directly write every file change to disk, but keeps it in cache for a while, until a certain overall write size is reached. This reduces disk writes (due to fix block size) quite much, if the written files are small, but it means that when you edit/overwrite a file and immediately read it back, you might still get the old version.
This is the only possibility how the old dietpi-update code can be called, after the now code was already downloaded, unpacked and copied to disk.

While the solution above works well, it is a bid dangerous. Actually there is good reason why we explicitly  restart the updater with new code. Allowing to run the old code by applying some workaround produces some possible follow-up errors.

To be sure you do not run into further issues, please exit the update, when facing this error. Run `sync` (which forces syncing file changes really to disk), wait for 1 second or two and simply restart the update: `dietpi-update 1`

Akito13, 2019-01-30:
> @rjjrdadev @Akito13 @Fourdee
> Thanks for your report.
>
> Jep this is a known, although uncritical issue with the update to v6.20. We are not yet sure about how this can occur, but at least it is an [...]

Indeed, I received a big issue with that. But every update warns, that something might happen, obviously, so I make a full system backup before every update, anyway. For the sake of convenience, I will restore the backup and run `dietpi-update`, as you recommended it.

MichaIng, 2019-01-30:
@Akito13 
Restoring the backup should not be required, as long as you have console access, rerunning the backup (with new updater code) should bring everything on desired stage.

And jep, with warning or not, doing a backup before every update is anyway always recommended.

Akito13, 2019-01-30:
> 
> 
> @Akito13
> Restoring the backup should not be required, as long as you have console access, rerunning the backup (with new updater code) should bring everything on desired stage.
> 
> And jep, with warning or not, doing a backup before every update is anyway always recommended.

It didn't work quite well, it said that I am already running the newest version. I didn't want to risk anything, so I just went the safe way, restoring my backup.

MichaIng, 2019-01-30:
@Akito13 
If you updated via placeholder file workaround, then simply assure that `/DietPi/dietpi/.update_stage` does **not** exist and `cat /DietPi/dietpi/.install_stage` shows `2`, just to be sure. This is done by the new update code, but not by the old, no other differences.

If you chose my way, exiting the update, if this error shows up, `sync` and `dietpi-update 1` it should go through as desired and apply the the changes above directly.

Both methods are as safe as they can be,

MichaIng, 2019-01-30:
Actually I want to fix that damn thing, so no one runs into that error again. Trying to replicate but on my VM it never happens: https://github.com/Fourdee/DietPi/issues/2471#issuecomment-459092596

Akito13, 2019-01-30:
> 
> 
> Actually I want to fix that damn thing, so no one runs into that error again. Trying to replicate but on my VM it never happens and everything works exactly as intended...: [#2471 (comment)](https://github.com/Fourdee/DietPi/issues/2471#issuecomment-459092596)

I don't know if that's the solution, but I updated my Raspi 0W to the newest version and everything went OK. Then I updated my Raspi 3B+ and this happened. The difference between both systems is that the Raspi 0W had already the previous `v6.20.5` update, whereas my Raspi 3B+ was still on `v6.19.x`. Looking at the OP, it is stated:

> * DietPi version | v6.19.7 (Fourdee/master)

So maybe upgrading from a "too low" version might be the issue.

Akito13, 2019-01-30:
@MichaIng @rjjrdadev I know restored my backup, was back on `v6.19.x`, ran `sync` and `sudo sync` to be safe. Ran `sudo dietpi-update` again. Now everything went smoothly. 👍

MichaIng, 2019-01-30:
Okay I will close this issue here in favour of: https://github.com/Fourdee/DietPi/issues/2471#issuecomment-459123285

Please go on with all conversation and possible questions over there 😉.

closed 2019-01-30 by MichaIng