MichaIng/DietPi #2479: Certbot install fails with pkgProblemResolver::Resolve generated breaks, this may be caused by held packages

JeffDanJ, 2019-01-31:
#### Details:
- Date           | Thu 31 Jan 05:32:45 GMT 2019
- Bug report     | N/A
- DietPi version | v6.20.6 (Fourdee/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B (armv7l) (index=3)
- Kernel version | #1159 SMP Sun Nov 4 17:50:20 GMT 2018
- Distro         | stretch (index=4)
- Command        | G_AGI: certbot python-certbot-nginx
- Exit code      | 100
- Software title | DietPi-Software

Log file contents:
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.
```

MichaIng, 2019-01-31:
@JeffDanJ 
Thanks for your report.

Hmm held packages (as of error message) actually can't be it, since G_AGI overrides held state. Regardless can you paste output of:
`apt-mark showhold`

Please also try to clean the APT cache:
```
apt clean
rm -R /var/cache/apt
G_AGUP
```

JeffDanJ, 2019-01-31:
Thanks for the response.

root@DietPi:~# apt-mark showhold
root@DietPi:~#

cleaning APT cache also didnt work and got the same error on retry.

#### Details:
- Date           | Thu 31 Jan 15:03:12 GMT 2019
- Bug report     | N/A
- DietPi version | v6.20.6 (Fourdee/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B (armv7l) (index=3)
- Kernel version | #1159 SMP Sun Nov 4 17:50:20 GMT 2018
- Distro         | stretch (index=4)
- Command        | G_AGI: certbot python-certbot-nginx
- Exit code      | 100
- Software title | DietPi-Software

MichaIng, 2019-02-05:
@JeffDanJ 
Sorry for the late reply.

Did you check output of: `apt-mark showhold`?
Also please try to install those two packages manually:
```
G_AGI certbot
G_AGI python-certbot-nginx
```

MichaIng, 2019-02-28:
I will close due to missing user reply and general cleanup. Feel free to reopen, if issue persists.

closed 2019-02-28 by MichaIng

ice41, 2019-07-11:
```
apt-get install certbot python-certbot-apache -t stretch-backports
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libc6-dev : Breaks: binutils (< 2.26) but 2.25-5+deb8u1 is to be installed
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.
```

what i can make to solvit ?

MichaIng, 2019-07-11:
@ice41 
I see, your on RPi1 or RPi Zero, right?

What do you require `libc6-dev` for? Please run `apt purge libc6-dev`, but do not confirm yet (if asked), instead paste which packages would be removed together with this.

There is no point in installing CertBot from stretch-backports btw, the version is exactly the same as in normal Stretch repo. However the issue appears in both cases.

ice41, 2019-07-16:
i just use what say on official website

MichaIng, 2019-07-16:
@ice41 
> i just use what say on official website

For Certbot? Using the APT packages does not require any dev package. In this case you could simply purge it. You can actually purge in in mostly any case. Even if it was required, this was most likely only the case for a source build, but not to run any software.

Ah or you mean regarding backports? This is then simply an outdated info, however it does not really matter.

ice41, 2019-07-16:
they resquest backports to install right?

> Add backports to your sources.list
deb http://deb.debian.org/debian stretch-backports main
apt-get -t stretch-backports install "package"