MichaIng/DietPi #2482: Wireguard config should set DNS to 10.8.0.1 instead of 127.0.0.1 if running local DNS

swrobel, 2019-01-31:
#### Required Information
- DietPi version | 6.20.6
- Distro version | stretch
- Kernel version | Linux DietPi 4.14.79-v7+ #1159
- SBC device | RPi 3 Model B+ (armv7l)

#### Additional Information (if applicable)
- Software title | Wireguard

#### Steps to reproduce

1. Install Wireguard on a system running Pi-Hole (ex: `/etc/resolv.conf` contains `nameserver 127.0.0.1`)

#### Expected behaviour

`/etc/wireguard/wg0-client.conf` contains `DNS = 10.8.0.1`

#### Actual behaviour

`/etc/wireguard/wg0-client.conf` contains `DNS = 127.0.0.1`

#### Extra details

See this line: https://github.com/Fourdee/DietPi/blob/dev/dietpi/dietpi-software#L10055

I would submit a PR but I'm not very good at shellscripting and believe this conditional should be a quick fix for someone knowledgeable.

MichaIng, 2019-02-07:
@swrobel
Thanks for the hint.

Indeed we get the DNS address from the nameserver entry of the server machine.

If it resolves DNS requests itself, e.g. Pi-hole installed or dnsmasq, the entry is the loop back address with of course is wrong for clients.

Solution code side is to check, if the DNS entry is loopback and in case replace it with the VPN sever IP as you did. I guess with the iptables rules in place, the local server IP (from local network, not WireGuard interface) should work as well?

swrobel, 2019-02-08:
You are indeed correct, it works with the local server ip

MichaIng, 2019-02-08:
@swrobel 
Okay not sure which one is best. I guess using `10.8.0.1` is a more direct connection, since at least forwarding to local network interface should be skipped. As well it works even without the `iptables` forward rules in place, which allow local network and in case www access through the VPN.

MichaIng, 2019-02-09:
PR up to solve this: https://github.com/Fourdee/DietPi/pull/2526

closed 2019-02-24 by MichaIng

niwre2019, 2019-04-14:
Hello,
since the latest upgrade to DietPi 6.22, my local DNS 192.168.xxx.xxx specified in wg0-client.conf has stopped working. After recovering the last version 6.21 everything works fine again. Does this have to do with these changes and how can I get WireGuard up and running again?
Thank you for your help

MichaIng, 2019-04-14:
@niwre2019 
I believe this is due to a kernel upgrade but missing WireGuard package (kernel module) upgrade. Please see for solution: https://github.com/MichaIng/DietPi/issues/2671

niwre2019, 2019-04-15:
I've done the steps, but the problem persists.  A correct connection to WireGuard is established, but the DNS resolution no longer exists.  I'm back to version 6.21.1 and everything works fine again.