MichaIng/DietPi #2483: Revert to old values - gpu_mem_1024=320

GulyFMG, 2019-02-01:
Would it be possible to not rewrite this option on every re / install?
[  OK  ] DietPi-Set_Hardware | Desired setting in /DietPi/config.txt was already set: gpu_mem_1024=320
[  OK  ] DietPi-Set_Hardware | Desired setting in /DietPi/config.txt was already set: gpu_mem_512=320
[  OK  ] DietPi-Set_Hardware | Desired setting in /DietPi/config.txt was already set: gpu_mem_256=192

My reasoning is simple i have some services running on my PI and allocating 320 Mb is a little to much just for kodi, for me 192 is more than enough. 320 with torrents downloading and sonarr / radarr searching for movies and shows i start having breaks on kodi.

Best regards.

Fourdee, 2019-02-01:
@GulyFMG 

Hi,

Many thanks for the report 👍 

I'll see if we can add an exclude option. Currently the value is only set if the current value is lower than what ```dietpi-software``` requires to be set during installation.

```dietpi-config``` can be used to set a new value. Which will not change unless you install software again in ```dietpi-software```

GulyFMG, 2019-02-01:
Thank you for this an exclude would be great, im always testing new software and with every reinstall i need to revert to my old value, this is just a little bit of a hassle... maybe an yes or no question.
If you need any help im available.

Fourdee, 2019-02-06:
Completed for v6.21, yes/no prompt will now appear if a change is required.

closed 2019-02-06 by Fourdee