MichaIng/DietPi #2487: Unable to PREP_SYSTEM from Armbian

holda29, 2019-02-01:
#### Details:
- Date           | Fri  1 Feb 17:04:01 UTC 2019
- Bug report     | N/A
- DietPi version | v6.20.0 (Fourdee/master)
- Img creator    | n/a
- Pre-image      | n/a
- SBC device     | NULL (index=22)
- Kernel version | #5.69 SMP Wed Jan 9 18:10:00 CET 2019
- Distro         | stretch (index=4)
- Command        | chmod -R +x /DietPi/dietpi /var/lib/dietpi/services /etc/cron.daily/dietpi /etc/cron.hourly/dietpi
- Exit code      | 1
- Software title | DietPi-PREP

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:
I am trying to install DietPi to OrangePi Zero + H5 on fresh latest Armbian install. I am following steps from here - https://github.com/Fourdee/DietPi/issues/1285

#### Actual behaviour:
But after submitting username and image name I choose stable build, but it ends in error chmod: cannot access '/DietPi/dietpi': No such file or directory#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
chmod: cannot access '/DietPi/dietpi': No such file or directory
```

Fourdee, 2019-02-01:
@holda29 

Hi,

Thanks for the report, this is fixed in v6.21. You can use the updated PREP script with:
```
wget https://raw.githubusercontent.com/Fourdee/DietPi/dev/PREP_SYSTEM_FOR_DIETPI.sh -O PREP_SYSTEM_FOR_DIETPI.sh
chmod +x PREP_SYSTEM_FOR_DIETPI.sh
./PREP_SYSTEM_FOR_DIETPI.sh
```

Fourdee, 2019-02-02:
Verified local testing by using the above fixed script:
https://github.com/Fourdee/DietPi/issues/2487#issuecomment-459809920

holda29, 2019-02-02:
Great, it worked! Thank you.