MichaIng/DietPi #2492: DietPi-Drive_Manager | Fresh drive format fails

MichaIng, 2019-02-03:
#### Details:
- Date           | Sun  3 Feb 17:16:09 CET 2019
- Bug report     | N/A
- DietPi version | v6.20.6 (Fourdee/dev)
- Img creator    | DietPi Core Team
- Pre-image      | Debian netboot
- SBC device     | Virtual Machine (x86_64) (index=20)
- Kernel version | #1 SMP Debian 4.9.130-2 (2018-10-27)
- Distro         | stretch (index=4)
- Command        | parted -a optimal /dev/sdb mkpart primary 0% 100%
- Exit code      | 143
- Software title | DietPi-Drive_Manager

#### Steps to reproduce:

1. Plug empty drive
2. Select MBR partition table (file system doesn't matter, since script doesn't point where it matters)
3. Format

#### Expected behaviour:

- Partition table + partition is created, file system written

#### Actual behaviour:

- Script hangs, waiting for interactive input, blocked by G_RUN_CMD until interrupted.
- Input requested due to non optimal alignment. `-a optimal` seems to not work as intended.

#### Extra details:

root@VM-Stretch:~# fdisk -l /dev/sdb
Disk /dev/sdb: 128 MiB, 134217728 bytes, 262144 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
root@VM-Stretch:~# dd if=/dev/zero of=/dev/sdb bs=4K count=1337
1337+0 records in
1337+0 records out
5476352 bytes (5.5 MB, 5.2 MiB) copied, 0.0912108 s, 60.0 MB/s
root@VM-Stretch:~# parted -s /dev/sdb mklabel msdos
root@VM-Stretch:~# fdisk -l /dev/sdb
Disk /dev/sdb: 128 MiB, 134217728 bytes, 262144 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x77b059a2
root@VM-Stretch:~# parted -a optimal /dev/sdb mkpart primary 0% 100%
Warning: The resulting partition is not properly aligned for best performance.
Ignore/Cancel?

#### Additional logs:
```
Log file contents:
Warning: The resulting partition is not properly aligned for best performance.
Ignore/Cancel?
```

MichaIng, 2019-02-03:
When ignoring:
```
Ignore/Cancel? i
Information: You may need to update /etc/fstab.

root@VM-Stretch:~# partprobe /dev/sdb
root@VM-Stretch:~# fdisk -l /dev/sdb
Disk /dev/sdb: 128 MiB, 134217728 bytes, 262144 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xa3847b0f

Device     Boot Start    End Sectors  Size Id Type
/dev/sdb1           1 262143  262143  128M 83 Linux
root@VM-Stretch:~# parted /dev/sdb align-check
alignment type(min/opt)  [optimal]/minimal?
Partition number? 1
1 not aligned
```

MichaIng, 2019-02-03:
Issue is: No optimal block size available:
```
root@VM-Stretch:~# cat /sys/block/sdb/queue/optimal_io_size
0
root@VM-Stretch:~# cat /sys/block/sdb/queue/minimum_io_size
512
root@VM-Stretch:~# cat /sys/block/sdb/alignment_offset
0
root@VM-Stretch:~# cat /sys/block/sdb/queue/physical_block_size
512
```

Fourdee, 2019-02-06:
@MichaIng 

Yep, if optimal is 0/null, set to ```mimimal``` 👍 

If ```mimimal``` does also not exist, set to a failsafe value of ```512```?

MichaIng, 2019-02-06:
@Fourdee 
Setting a custom value doesn't seem to be possible: https://manpages.debian.org/stretch/manpages-de/parted.8.de.html
`none` should work in any case, no idea what `cylinder` means 🤔.

Fourdee, 2019-02-08:
@MichaIng 

> Setting a custom value doesn't seem to be possible

😢 

> cylinder

Aligns to physical cylinder location I believe, used on old platter drives in past:
http://cisn.metu.edu.tr/97-2/images/Image9.gif
http://gparted-forum.surf4.info/viewtopic.php?pid=31628#p31628

MichaIng, 2019-02-19:
@Fourdee 
> Use Cylinder alignment to maintain compatibility with operating systems released before the year 2000, such as DOS. This setting aligns partitions to start and end on disk cylinder boundaries.

Okey should be not required.
_______
So go with optimal > minimal > none. Verify /sys file scheme:
#### VM Stretch
```
root@VM-Stretch:~# l /sys/block/sda/queue/
total 0
-rw-r--r-- 1 root root 4096 Feb 19 14:15 add_random
-r--r--r-- 1 root root 4096 Feb 19 14:15 dax
-r--r--r-- 1 root root 4096 Feb 19 14:15 discard_granularity
-rw-r--r-- 1 root root 4096 Feb 19 14:15 discard_max_bytes
-r--r--r-- 1 root root 4096 Feb 19 14:15 discard_max_hw_bytes
-r--r--r-- 1 root root 4096 Feb 19 14:15 discard_zeroes_data
-r--r--r-- 1 root root 4096 Feb 19 14:15 hw_sector_size
-rw-r--r-- 1 root root 4096 Feb 19 14:15 io_poll
drwxr-xr-x 2 root root    0 Feb 19 14:15 iosched
-rw-r--r-- 1 root root 4096 Feb 19 14:15 iostats
-r--r--r-- 1 root root 4096 Feb 19 14:15 logical_block_size
-r--r--r-- 1 root root 4096 Feb 19 14:15 max_hw_sectors_kb
-r--r--r-- 1 root root 4096 Feb 19 14:15 max_integrity_segments
-rw-r--r-- 1 root root 4096 Feb 19 14:15 max_sectors_kb
-r--r--r-- 1 root root 4096 Feb 19 14:15 max_segments
-r--r--r-- 1 root root 4096 Feb 19 14:15 max_segment_size
-r--r--r-- 1 root root 4096 Feb 19 14:15 minimum_io_size
-rw-r--r-- 1 root root 4096 Feb 19 14:15 nomerges
-rw-r--r-- 1 root root 4096 Feb 19 14:15 nr_requests
-r--r--r-- 1 root root 4096 Feb 19 14:15 optimal_io_size
-r--r--r-- 1 root root 4096 Feb 19 14:15 physical_block_size
-rw-r--r-- 1 root root 4096 Feb 19 14:15 read_ahead_kb
-rw-r--r-- 1 root root 4096 Feb 19 14:15 rotational
-rw-r--r-- 1 root root 4096 Feb 19 14:15 rq_affinity
-rw-r--r-- 1 root root 4096 Feb 19 14:15 scheduler
-rw-r--r-- 1 root root 4096 Feb 19 14:15 write_cache
-r--r--r-- 1 root root 4096 Feb 19 14:15 write_same_max_bytes
root@VM-Stretch:~# cat /sys/block/sda/queue/logical_block_size
512
root@VM-Stretch:~# cat /sys/block/sda/queue/optimal_io_size
0
root@VM-Stretch:~# cat /sys/block/sda/queue/minimum_io_size
512
```
#### RPI2
```
2019-02-19 13:46:16 root@micha:/tmp# l /sys/block/mmcblk0/queue/
total 0
-rw-r--r-- 1 root root 4.0K Feb 19 14:16 add_random
-r--r--r-- 1 root root 4.0K Feb 19 14:16 chunk_sectors
-r--r--r-- 1 root root 4.0K Feb 19 14:16 dax
-r--r--r-- 1 root root 4.0K Feb 19 14:16 discard_granularity
-rw-r--r-- 1 root root 4.0K Feb 19 14:16 discard_max_bytes
-r--r--r-- 1 root root 4.0K Feb 19 14:16 discard_max_hw_bytes
-r--r--r-- 1 root root 4.0K Feb 19 14:16 discard_zeroes_data
-r--r--r-- 1 root root 4.0K Feb 19 14:16 hw_sector_size
-rw-r--r-- 1 root root 4.0K Feb 19 14:16 io_poll
-rw-r--r-- 1 root root 4.0K Feb 19 14:16 io_poll_delay
drwxr-xr-x 2 root root    0 Feb 19 14:16 iosched
-rw-r--r-- 1 root root 4.0K Feb 19 14:16 iostats
-r--r--r-- 1 root root 4.0K Feb 19 14:16 logical_block_size
-r--r--r-- 1 root root 4.0K Feb 19 14:16 max_discard_segments
-r--r--r-- 1 root root 4.0K Feb 19 14:16 max_hw_sectors_kb
-r--r--r-- 1 root root 4.0K Feb 19 14:16 max_integrity_segments
-rw-r--r-- 1 root root 4.0K Feb 19 14:16 max_sectors_kb
-r--r--r-- 1 root root 4.0K Feb 19 14:16 max_segments
-r--r--r-- 1 root root 4.0K Feb 19 14:16 max_segment_size
-r--r--r-- 1 root root 4.0K Feb 19 14:16 minimum_io_size
-rw-r--r-- 1 root root 4.0K Feb 19 14:16 nomerges
-rw-r--r-- 1 root root 4.0K Feb 19 14:16 nr_requests
-r--r--r-- 1 root root 4.0K Feb 19 14:16 optimal_io_size
-r--r--r-- 1 root root 4.0K Feb 19 14:16 physical_block_size
-rw-r--r-- 1 root root 4.0K Feb 19 14:16 read_ahead_kb
-rw-r--r-- 1 root root 4.0K Feb 16 22:21 rotational
-rw-r--r-- 1 root root 4.0K Feb 19 14:16 rq_affinity
-rw-r--r-- 1 root root 4.0K Feb 19 14:16 scheduler
-rw-r--r-- 1 root root 4.0K Feb 19 14:16 wbt_lat_usec
-rw-r--r-- 1 root root 4.0K Feb 19 14:16 write_cache
-r--r--r-- 1 root root 4.0K Feb 19 14:16 write_same_max_bytes
-r--r--r-- 1 root root 4.0K Feb 19 14:16 write_zeroes_max_bytes
-r--r--r-- 1 root root 4.0K Feb 19 14:16 zoned
2019-02-19 14:16:42 root@micha:/tmp# cat /sys/block/mmcblk0/queue/logical_block_size
512
2019-02-19 14:16:54 root@micha:/tmp# cat /sys/block/mmcblk0/queue/optimal_io_size
0
2019-02-19 14:16:58 root@micha:/tmp# cat /sys/block/mmcblk0/queue/minimum_io_size
512
```
- Öuuuhhh, `optimal_io_size` == 0/zero as well, but `parted -a optimal` succeeded here? Although some months ago I installed this machine.
______
What is the default alignment `parted` attempts? Doesn't it already do best possible alignment automatically, so we can skip the `-a` option?

MichaIng, 2019-02-19:
Found: https://blog.hqcodeshop.fi/archives/273-GNU-Parted-Solving-the-dreaded-The-resulting-partition-is-not-properly-aligned-for-best-performance.html

Interesting part of the script:
```
# Do the calc
# optimal_io_size: Storage devices may report an optimal I/O size, which is
#                   the device's preferred unit for sustained I/O [bytes]
optimal_io_size=$(cat /sys/block/$BLOCK_DEVICE/queue/optimal_io_size)
stat=$?
# physical_block_size: This is the smallest unit a physical storage device
#                       can write atomically [bytes]
physical_block_size=$(cat /sys/block/$BLOCK_DEVICE/queue/physical_block_size)
stat+=$?
if [ "$stat" != "00" ]; then
    echo Internal error: Failed to extract block device information!
    exit 1
fi

if [ $optimal_io_size == 0 ]; then
    echo Using default 1 MiB default alignment in calc
    default_io_size_calc align_to_sector $BLOCK_DEVICE
else
    echo Using optimal_io_size: $optimal_io_size in calc
    optimal_io_size_calc align_to_sector $BLOCK_DEVICE $optimal_io_size
fi
```

So if optimal value is `0` it uses 1 MiB as default which leads to the widely seen `2048s` start. However in the articular 0% 100% was later found and used without any `-a` option. 

Found many other sources that state the 1 MiB default works in very most cases and that using percentage boundaries 0% 100% auto align the drive, e.g. quite clear: https://wiki.archlinux.org/index.php/Parted#Alignment

So let's just skip `-a`, leave this to parted and add `-s` instead where we call it with G_RUN_CMD or other non-interactive way?

MichaIng, 2019-03-05:
Fixed: https://github.com/MichaIng/DietPi/commit/e070adf71a03844b588a186ac513b56bd2c148cf

closed 2019-03-23 by MichaIng