MichaIng/DietPi #2494: Wireguard for Nanopi (Neo 2)

jvteleco, 2019-02-03:
### Creating a feature request
Wireguard was recently made available to install, but only 
"NB: Currently only works on x86_64 and RPi's. Since kernel headers are required, we need to add those for other devices, before the WireGuard kernel module can be compiled."
Reference:  https://github.com/Fourdee/DietPi/pull/2398

Not sure what requirements are needed for the Nano Pi  Neo2. Seems it is a kernel related issue ?

#### Describe the solution you'd like:
Hopefully will be added in future releases.

#### Additional context
Seems that there also a request "Enable dm-crypt support in kernel #2464" that will require kernel modification.
Reference: https://github.com/Fourdee/DietPi/issues/2464

Thanks

MichaIng, 2019-02-04:
@jvteleco 
Thanks for your request.

- WireGuard is a kernel-based VPN solution, which has (compared to OpenVPN) many advantages, but requires the `wireguard` kernel module, which is not by default integrated in the Linux kernel tree.
- Generally to build kernel modules, the kernel headers (sources) are required. For the devices were we currently offer WireGuard, those are either available via APT repositories or we found another reliable source to install and update them from.
- For NanoPi's we did not yet found, or, simply did not yet have the time collect the correct header sources for each NanoPi device and implement the WireGuard install for those. It is simply a research and testing effort for each individual device. Let's see if we find more users requesting WireGuard for NanoPi NEO2 to raise priority of this request.

`dm-crypt` is a similar but not identical issue:
- It is as well a kernel module, but one that is built into the kernel, so no "external" module. To enable it, some compile-time options are required when building the kernel itself.
- So this is something that needs to be done by the NanoPi kernel developers. Since we do no kernel development, we need to forward this request accordingly.

olegbc1, 2019-02-19:
Also want to run wireguard on my nanoPi Neo2. Where do I +1? :-)

MichaIng, 2019-02-20:
@olegbc1 
You may want to add the request to our FeatHub page to collect votes 🙂: https://feathub.com/Fourdee/DietPi

olegbc1, 2019-02-20:
@MichaIng Thanks - done! https://feathub.com/Fourdee/DietPi/+43

MichaIng, 2020-06-25:
Generic support has been added (coming with DietPi v6.31) for all boards where the WireGuard kernel module is shipped with the kernel package: #3620
This is true for most current Armbian-based images (with Linux 5.X kernel) and is expected to become true for more boards, since Linux 5.6 has integrated WireGuard natively.

closed 2020-07-02 by MichaIng