MichaIng/DietPi #2495: Odroid XU3/XU4/HC1/HC2 image still 6.14 and it causes installation loop

rox-wolf, 2019-02-03:
### Creating a bug report/issue

#### Steps to reproduce
1. Download Odroid XU3/XU4/HC1/HC2 image, it's currently DietPi_v6.14_OdroidXU4-ARMv7-Stretch.img  So not updated to 6.20, for example RaspberryPi image is 6.20. 
2. Install the image, go through installation steps.

#### Expected behaviour
- Dietpi should install

#### Actual behaviour
- After installing the distro dietpi-software asks for reboot, after reboot installation starts again asking to change passwords and so on. So installation never stops. 

To get out of the loop you need to click cancel on software install dialog and get to raw DietPi. Then you need to run dietpi-update and update it to 6.20. After update installer runs again asking to change passwords, but this time it will finish. 

So looks like having 6.14 image and installing the distro causes issues.

MichaIng, 2019-02-04:
@rox-wolf 
Thanks for your report.

Jep this matches other reports, that `dietpi-ramdisk` is still failing to sync the update to disk on reboot in some cases. Also your solution matches the assumption. I did some deeper review/research here: https://github.com/Fourdee/DietPi/issues/2471#issuecomment-460104856

It is a pain that I was never able replicate any sync issue on testing systems, so I am not able to test/verify any fixes, since nothing changes on my particular boot/shutdown order. It seems to be a bid random when systemd stops which service and mount, but at least I found a gap which can be the reason. If this is true, it must have changed with the latest `systemd` APT updates, since this would have failed on older systems before. This also fits to the fact, that on Debian Jessie, Stretch and Buster, the mounts are implemented slightly different (according to my tests).

I will prepare a v6.21 pull request later to implement the thoughts within the link. I vote for a new subversion release (instead of v6.20.7 or such), to avoid possible long taking dietpi-software reinstalls for users, where the update succeeded. And I vote for a very soon release, since for effected users the update issue is very annoying and shocking when after update one is presented with a first run setup that looks like if all existing software is lost (which is not of course).

rox-wolf, 2019-02-04:
I have successfully installed 6.20 image on Raspberry Pi 3 and Zero W, no install loops on these devices. But on my Odroid HC1 it loops. If you need some early tests on Odroid HC1, I can do them :)

symbios24, 2019-02-06:
On my Raspberry pi 3 it loops

MichaIng, 2019-02-06:
@rox-wolf @symbios24
PR is up to solve the issue: https://github.com/Fourdee/DietPi/pull/2509
We will release v6.21 as hotfix with this and some other smaller fixes very soon.

Until then please try the following:
When `dietpi-software` prompts on login, simply exit it.
```
echo 1 > /DietPi/dietpi/.install_stage
dietpi-update 1 # Do NOT reboot, when asked, reboot later
echo 1 > /DietPi/dietpi/.install_stage
systemctl restart dietpi-ramdisk
sync
sleep 5
reboot
```

Fourdee, 2019-02-07:
Testing, current XU4 image:
- Master
- Dev

Fourdee, 2019-02-07:
Completed, v6.21 released.

closed 2019-02-07 by Fourdee

openoms, 2019-03-27:
Hi! I am still running into the same bootloop problem during the automated setup on the Odroid XU4.
The solution could be to upload a new image  so the setup does not need to start at below v6.21.
The image here https://dietpi.com/downloads/images/DietPi_OdroidXU4-ARMv7-Stretch.7z  is still the v6.14 however the issue has been fixed afterwards.

ezraholm50, 2019-10-23:
Also have this on an RPI Zero

```
[ INFO ] DietPi-Update | Current version : v6.26.3
[ INFO ] DietPi-Update | Latest version  : v6.26.3
```

Suggested fixes don't work. Please advise.

MichaIng, 2019-10-23:
@ezraholm50 
Could you explain a bid more in detail and at best paste some more logs about the issue you face? At best open a new issue, since it is not related to the old one, which has been fixed by the way the update is applied.

ezraholm50, 2019-10-23:
Will do, thanks.